stereolabs/zed-ros2-wrapper #113: Build error on Jetson Orin 5.0.2

MrOCW, 2022-10-27:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Build fails in a humble container for Jetpack 5.0.2

### Steps to Reproduce

1. Dockerfile

`ARG BASE_IMAGE=dustynv/ros:humble-ros-base-l4t-r35.1.0
FROM ${BASE_IMAGE}

SHELL ["/bin/bash", "-c"]
ENV SHELL /bin/bash

ENV DEBIAN_FRONTEND=noninteractive
ARG MAKEFLAGS=-j$(nproc)
ENV LANG=en_US.UTF-8
ENV PYTHONIOENCODING=utf-8
RUN locale-gen en_US en_US.UTF-8 && update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8

ARG ROS_ENVIRONMENT=${ROS_ROOT}/install/setup.bash

# Required packages
RUN apt-get update && \
    apt-get install -y --no-install-recommends \
    software-properties-common \
    apt-transport-https \
    ca-certificates \
    gnupg \
    && rosdep update && \
    apt-get -y upgrade && \
    apt-get install zstd && \
    rm -rf /var/lib/apt/lists/* \
    && apt-get clean

ARG ZED_SDK_URL="https://download.stereolabs.com/zedsdk/3.8/l4t35.1/jetsons"
ARG ZED_SDK_RUN="ZED_SDK_Linux_JP.run"

RUN cd /tmp && \
    pip3 install protobuf && \
    echo "# R35 (release), REVISION: 1.0" > /etc/nv_tegra_release ; \
    wget --quiet --show-progress --progress=bar:force:noscroll --no-check-certificate ${ZED_SDK_URL} -O ${ZED_SDK_RUN} && \
    chmod +x ${ZED_SDK_RUN} && \
    ./${ZED_SDK_RUN} silent skip_tools && \
    rm -rf /usr/local/zed/resources/* && \
    rm -rf ${ZED_SDK_RUN} && \
    rm -rf /var/lib/apt/lists/* && \
    apt-get clean

RUN source ${ROS_ENVIRONMENT} && \
    cd ${ROS_ROOT} && \
    apt-get update && \
    apt-get upgrade && \
    mkdir -p src/zed && \
    cd src/zed && \
    git clone -b 3.0.0 https://github.com/ros-perception/image_common.git &&\
    git clone -b ros2-devel https://github.com/ros/diagnostics.git && \
    git clone -b ros2 https://github.com/ros/xacro.git && \
    cd ../.. &&\
    rosdep install --from-paths src/zed --ignore-src --rosdistro ${ROS_DISTRO} -y --skip-keys "libopencv-dev libopencv-contrib-dev libopencv-imgproc-dev python-opencv python3-opencv" && \
    colcon build --symlink-install --merge-install --base-paths src/zed && \
    rm -rf /var/lib/apt/lists/* && \
    apt-get clean
    
COPY ./ros_entrypoint.sh /
ENTRYPOINT ["/ros_entrypoint.sh"]
CMD ["bash"]`
...

### Expected Result

colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release 
suceeds

### Actual Result

```
Finished <<< zed_interfaces [18.1s]
Starting >>> zed_components
--- stderr: zed_components                               
* Found ROS2 humble
* ROS2 humble is officially supported by this package.
/home/robot/external/zed/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp: In member function ‘bool stereolabs::ZedCamera::startCamera()’:
/home/robot/external/zed/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:2689:31: error: ‘class sl::Camera’ has no member named ‘setRegionOfInterest’
 2689 |     sl::ERROR_CODE err = mZed.setRegionOfInterest(roi_mask);
      |                               ^~~~~~~~~~~~~~~~~~~
/home/robot/external/zed/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp: In member function ‘bool stereolabs::ZedCamera::startPosTracking()’:
/home/robot/external/zed/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:2922:15: error: ‘struct sl::PositionalTrackingParameters’ has no member named ‘depth_min_range’
 2922 |   trackParams.depth_min_range = mPosTrackDepthMinRange;
      |               ^~~~~~~~~~~~~~~
/home/robot/external/zed/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:2923:15: error: ‘struct sl::PositionalTrackingParameters’ has no member named ‘set_gravity_as_origin’; did you mean ‘set_floor_as_origin’?
 2923 |   trackParams.set_gravity_as_origin = mSetGravityAsOrigin;
      |               ^~~~~~~~~~~~~~~~~~~~~
      |               set_floor_as_origin
/home/robot/external/zed/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp: In member function ‘bool stereolabs::ZedCamera::startObjDetect()’:
/home/robot/external/zed/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:3102:8: error: ‘struct sl::ObjectDetectionParameters’ has no member named ‘prediction_timeout_s’
 3102 |   od_p.prediction_timeout_s = mObjDetPredTimeout;
      |        ^~~~~~~~~~~~~~~~~~~~
/home/robot/external/zed/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp: In member function ‘void stereolabs::ZedCamera::callback_setRoi(std::shared_ptr<rmw_request_id_s>, std::shared_ptr<zed_interfaces::srv::SetROI_Request_<std::allocator<void> > >, std::shared_ptr<zed_interfaces::srv::SetROI_Response_<std::allocator<void> > >)’:
/home/robot/external/zed/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:6854:31: error: ‘class sl::Camera’ has no member named ‘setRegionOfInterest’
 6854 |     sl::ERROR_CODE err = mZed.setRegionOfInterest(roi_mask);
      |                               ^~~~~~~~~~~~~~~~~~~
/home/robot/external/zed/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp: In member function ‘void stereolabs::ZedCamera::callback_resetRoi(std::shared_ptr<rmw_request_id_s>, std::shared_ptr<std_srvs::srv::Trigger_Request_<std::allocator<void> > >, std::shared_ptr<std_srvs::srv::Trigger_Response_<std::allocator<void> > >)’:
/home/robot/external/zed/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:6885:29: error: ‘class sl::Camera’ has no member named ‘setRegionOfInterest’
 6885 |   sl::ERROR_CODE err = mZed.setRegionOfInterest(empty_roi);
      |                             ^~~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/zed_camera_component.dir/build.make:90: CMakeFiles/zed_camera_component.dir/src/zed_camera/src/zed_camera_component.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:164: CMakeFiles/zed_camera_component.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
---
Failed   <<< zed_components [23.7s, exited with code 2]
```

### ZED Camera model

ZED2i

### Environment

```shell
Jetpack 5.0.2
Nvidia Jetson Orin AGX
```

Myzhar, 2022-10-27:
Hi @MrOCW 
the master branch of the repository requires the latest ZED SDK v3.8.x to compile correctly.

Myzhar, 2022-10-27:
This line is not required for ROS2 Humble:
`git clone -b 3.0.0 https://github.com/ros-perception/image_common.git &&\`

Myzhar, 2022-10-27:
I tried to build the image using the fixed Dockerfile and the problem that you reported is not present, maybe you have not correctly downloaded the SDK v3.8. Try to clean and rebuild the image.

There is instead a bigger problem: the base image that you are using `dustynv/ros:humble-ros-base-l4t-r35.1.0` seems derived by the `l4t-r35.1.0` image that does not contain CUDA, so it is not possible to use the ZED SDK with it:

```
--- stderr: zed_wrapper
* Found ROS2 humble
* ROS2 humble is officially supported by this package.
/usr/bin/ld: warning: libcuda.so.1, needed by /usr/local/zed/lib/libsl_zed.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libnvbuf_utils.so.1.0.0, needed by /usr/local/zed/lib/libsl_zed.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: /usr/local/zed/lib/libsl_zed.so: undefined reference to `cuCtxPushCurrent_v2'
/usr/bin/ld: /usr/local/zed/lib/libsl_zed.so: undefined reference to `NvBufferMemMap'
/usr/bin/ld: /usr/local/zed/lib/libsl_zed.so: undefined reference to `cuStreamSynchronize'
/usr/bin/ld: /usr/local/zed/lib/libsl_zed.so: undefined reference to `NvBufferMemUnMap'
/usr/bin/ld: /usr/local/zed/lib/libsl_zed.so: undefined reference to `cuCtxPopCurrent_v2'
/usr/bin/ld: /usr/local/zed/lib/libsl_zed.so: undefined reference to `NvBufferMemSyncForDevice'
/usr/bin/ld: /usr/local/zed/lib/libsl_zed.so: undefined reference to `cuCtxSetCurrent'
/usr/bin/ld: /usr/local/zed/lib/libsl_zed.so: undefined reference to `cuCtxCreate_v2'
/usr/bin/ld: /usr/local/zed/lib/libsl_zed.so: undefined reference to `NvBufferGetParams'
/usr/bin/ld: /usr/local/zed/lib/libsl_zed.so: undefined reference to `NvBufferDestroy'
/usr/bin/ld: /usr/local/zed/lib/libsl_zed.so: undefined reference to `NvBufferCreateEx'
/usr/bin/ld: /opt/ros/humble/install/lib/libzed_camera_component.so: undefined reference to `cuCtxGetDevice'
/usr/bin/ld: /usr/local/zed/lib/libsl_zed.so: undefined reference to `cuGetErrorName'
/usr/bin/ld: /usr/local/zed/lib/libsl_zed.so: undefined reference to `NvBufferTransform'
/usr/bin/ld: /usr/local/zed/lib/libsl_zed.so: undefined reference to `cuMemcpy2DAsync_v2'
/usr/bin/ld: /usr/local/zed/lib/libsl_zed.so: undefined reference to `cuCtxDestroy_v2'
/usr/bin/ld: /usr/local/zed/lib/libsl_zed.so: undefined reference to `NvBufferMemSyncForCpu'
/usr/bin/ld: /usr/local/zed/lib/libsl_zed.so: undefined reference to `cuCtxGetCurrent'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/zed_wrapper.dir/build.make:363: zed_wrapper] Error 1
make[1]: *** [CMakeFiles/Makefile2:164: CMakeFiles/zed_wrapper.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
---
Failed   <<< zed_wrapper [1min 20s, exited with code 2]
```

maxpolzin, 2022-11-07:
@Myzhar Nvidida Jetson docker images contain stubs of the cuda library. You can compile the ZED SDK in it with the following colcon command:     
`colcon build --symlink-install --event-handlers console_direct+ --base-paths src/zed-ros2-wrapper --cmake-args ' -DCMAKE_BUILD_TYPE=Release' ' -DCMAKE_LIBRARY_PATH=/usr/local/cuda/lib64/stubs' ' -DCMAKE_CXX_FLAGS="-Wl,--allow-shlib-undefined"'`

maxpolzin, 2022-11-07:
```
FROM --platform=arm64  alpine:latest as unzipper

RUN apk add unzip wget curl
WORKDIR /opt

ARG XACRO_VERSION=2.0.8
RUN wget https://github.com/ros/xacro/archive/refs/tags/${XACRO_VERSION}.tar.gz -O - | tar -xvz && mv xacro-${XACRO_VERSION} xacro
ARG DIAGNOSTICS_VERSION=3.0.0
RUN wget https://github.com/ros/diagnostics/archive/refs/tags/${DIAGNOSTICS_VERSION}.tar.gz -O - | tar -xvz && mv diagnostics-${DIAGNOSTICS_VERSION} diagnostics

FROM --platform=arm64 dustynv/ros:humble-ros-base-l4t-r35.1.0

ARG ZED_SDK_URL="https://download.stereolabs.com/zedsdk/3.8/l4t35.1/jetsons"
ARG ZED_SDK_RUN="ZED_SDK_Linux_JP.run"

RUN cd /tmp && \
    apt update && apt install zstd -y --no-install-recommends && \
    wget --quiet --show-progress --progress=bar:force:noscroll --no-check-certificate ${ZED_SDK_URL} -O ${ZED_SDK_RUN} && \
    chmod +x ${ZED_SDK_RUN} && \
    ./${ZED_SDK_RUN} silent skip_tools && \
    rm -rf /usr/local/zed/resources/* && \
    rm -rf ${ZED_SDK_RUN} && \
    rm -rf /var/lib/apt/lists/* && \
    apt-get clean

WORKDIR /colcon_ws

COPY  --from=unzipper /opt/xacro src/xacro    
COPY  --from=unzipper /opt/diagnostics src/diagnostics    

RUN git clone --recursive https://github.com/stereolabs/zed-ros2-wrapper src/zed-ros2-wrapper && \
    . /opt/ros/${ROS_DISTRO}/install/setup.sh && \
    colcon build --symlink-install --event-handlers console_direct+ --base-paths src --cmake-args ' -DCMAKE_BUILD_TYPE=Release' ' -DCMAKE_LIBRARY_PATH=/usr/local/cuda/lib64/stubs' ' -DCMAKE_CXX_FLAGS="-Wl,--allow-shlib-undefined"'

WORKDIR /

COPY ros_entrypoint.sh .

```

Here is a link to the repository with the Dockerfile which I am using to build it: https://github.com/rosblox/ros-zed-ros2-wrapper
It builds and runs successfully for L4T 35.1 with ROS2 Humble on my Jetson Xavier NX.

Myzhar, 2022-11-07:
Thank you for sharing :+1:

maxpolzin, 2022-11-07:
You are welcome.

As soon as the CI jobs are run, you can pull the image:
`docker pull ghcr.io/rosblox/ros-zed-ros2-wrapper:humble`

Myzhar, 2022-11-07:
Do you plan to add nav2 support?

maxpolzin, 2022-11-07:
Nope, this image will remain standalone, i.e. pure zed-ros2-wrapper. I'd recommend implementing nav2 support in another container.

Myzhar, 2022-11-07:
Yes, I agree with this.
I pinned this issue as a future reference for everyone.

Myzhar, 2022-12-06:
@maxpolzin 

> You can pull the image: docker pull ghcr.io/rosblox/ros-zed-ros2-wrapper:humble

is the image publicly available? We can't pull it:
`Head "https://ghcr.io/v2/rosblox/ros-zed-ros2-wrapper/manifests/humble": unauthorized`

maxpolzin, 2022-12-06:
My bad... It should be publicly available now. Let me know if it works.

closed 2022-12-15 by MrOCW