stereolabs/zed-ros2-wrapper #114: Calibration file can not be downloaded.

peredwardsson, 2022-10-27:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

SDK downloaded from https://download.stereolabs.com/zedsdk/3.5/jp45/jetsons
zed_ros2 built from source, commit: de1dfc4cde9ada3173074a05e14631d4d56e442c

Starting the zed_wrapper node with a Zed2 connected to my Jetson AGX Xavier does not start the camera. I removed all parameter changes so they should be standard. Camera is connected via USB PCI-E extension. Has worked beautifully fine until today. It seems to get stuck at `Downloading calibration`. See full starting output:

```
# LC_ALL="C" ros2 run zed_wrapper zed_wrapper
[INFO] [1666869636.400344029] [zed_node]: ********************************
[INFO] [1666869636.400692946] [zed_node]:       ZED Camera Component
[INFO] [1666869636.400770422] [zed_node]: ********************************
[INFO] [1666869636.400844123] [zed_node]:  * namespace: /
[INFO] [1666869636.400949121] [zed_node]:  * node name: zed_node
[INFO] [1666869636.401029286] [zed_node]: ********************************
[INFO] [1666869636.401372986] [zed_node]: *** DEBUG parameters ***
[INFO] [1666869636.401480544] [zed_node]: *** GENERAL parameters ***
[INFO] [1666869636.401839413] [zed_node]:  * Camera model: zed - ZED
[INFO] [1666869636.402127078] [zed_node]:  * SDK Verbose: 1
[INFO] [1666869636.402503420] [zed_node]:  * SVO:
[INFO] [1666869636.402790189] [zed_node]:  * SVO Loop: FALSE
[INFO] [1666869636.403072798] [zed_node]:  * SVO Realtime: FALSE
[INFO] [1666869636.403329805] [zed_node]:  * Camera name: zed
[INFO] [1666869636.403672545] [zed_node]:  * Camera ID: 0
[INFO] [1666869636.403920560] [zed_node]:  * Camera SN: 0
[INFO] [1666869636.404220514] [zed_node]:  * Camera timeout [sec]: 5
[INFO] [1666869636.404475537] [zed_node]:  * Camera reconnection temptatives: 5
[INFO] [1666869636.404762433] [zed_node]:  * Camera framerate: 15
[INFO] [1666869636.405020401] [zed_node]:  * GPU ID: -1
[INFO] [1666869636.405302401] [zed_node]:  * Camera resolution: 2 - HD720
[INFO] [1666869636.405562929] [zed_node]:  * Camera self calibration: TRUE
[INFO] [1666869636.405824032] [zed_node]:  * Camera flip: FALSE
[INFO] [1666869636.406050061] [zed_node]:  * [DYN] Publish framerate [Hz]: 15
[INFO] [1666869636.406172340] [zed_node]: *** VIDEO parameters ***
[INFO] [1666869636.406484103] [zed_node]:  * Use old extrinsic parameters: 0
[INFO] [1666869636.406774104] [zed_node]:  * [DYN] Image downsample factor: 1
[INFO] [1666869636.407043784] [zed_node]:  * [DYN] Brightness: 4
[INFO] [1666869636.407328889] [zed_node]:  * [DYN] Contrast: 4
[INFO] [1666869636.407607337] [zed_node]:  * [DYN] Hue: 0
[INFO] [1666869636.407842999] [zed_node]:  * [DYN] Saturation: 4
[INFO] [1666869636.408155721] [zed_node]:  * [DYN] Sharpness: 4
[INFO] [1666869636.408401336] [zed_node]:  * [DYN] Gamma: 8
[INFO] [1666869636.408683240] [zed_node]:  * [DYN] Auto Exposure/Gain: TRUE
[INFO] [1666869636.408914614] [zed_node]:  * [DYN] Exposure: 80
[INFO] [1666869636.409226088] [zed_node]:  * [DYN] Gain: 80
[INFO] [1666869636.409453430] [zed_node]:  * [DYN] Auto White Balance: TRUE
[INFO] [1666869636.409757160] [zed_node]:  * [DYN] White Balance Temperature: 42
[INFO] [1666869636.410023287] [zed_node]:  * Video QoS History: KEEP_LAST
[INFO] [1666869636.410284295] [zed_node]:  * Video QoS History depth: 1
[INFO] [1666869636.410642268] [zed_node]:  * Video QoS Reliability: RELIABLE
[INFO] [1666869636.410918156] [zed_node]:  * Video QoS Durability: VOLATILE
[INFO] [1666869636.411001521] [zed_node]: *** DEPTH parameters ***
[INFO] [1666869636.411239615] [zed_node]:  * Depth quality: 1 - PERFORMANCE
[INFO] [1666869636.411497998] [zed_node]:  * Depth downsample factor: 1
[INFO] [1666869636.411810336] [zed_node]:  * Min depth [m]: 0.2
[INFO] [1666869636.412057967] [zed_node]:  * Max depth [m]: 10
[INFO] [1666869636.412299741] [zed_node]:  * Depth Sensing Mode: 0 - STANDARD
[INFO] [1666869636.412565869] [zed_node]:  * Depth Stabilization: TRUE
[INFO] [1666869636.412809115] [zed_node]:  * OpenNI mode (16bit point cloud): FALSE
[INFO] [1666869636.413093068] [zed_node]:  * [DYN] Point cloud rate [Hz]: 15
[INFO] [1666869636.413353531] [zed_node]:  * [DYN] Depth Confidence: 50
[INFO] [1666869636.413637868] [zed_node]:  * [DYN] Depth Texture Confidence: 100
[INFO] [1666869636.413869818] [zed_node]:  * Depth QoS History: KEEP_LAST
[INFO] [1666869636.414134121] [zed_node]:  * Depth QoS History depth: 1
[INFO] [1666869636.414449436] [zed_node]:  * Depth QoS Reliability: RELIABLE
[... 411 lines cut ...]
[INFO] [1666871508.679879804] [zed_node]: Camera-IMU Rotation:
 7FED8F0F18
0.999964 0.007649 -0.003558
-0.007636 0.999965 0.003526
0.003585 -0.003499 0.999987

[INFO] [1666871508.680223148] [zed_node]: Published static TF: 'zed_imu_link' -> 'zed_left_camera_frame'
terminate called after throwing an instance of 'std::length_error'
  what():  vector::_M_default_append
#
```
and crashes.

### ZED Camera model

ZED2

### Environment

```shell
OS: 4.9.253-tegra
CPU/GPU: Jetson AGX Xavier
Zed SDK: 3.5 I think, see link to download above
ROS2 Foxy
```

peredwardsson, 2022-10-27:
```
# locale
LANG=en_US.UTF-8
LANGUAGE=
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_PAPER="en_US.UTF-8"
LC_NAME="en_US.UTF-8"
LC_ADDRESS="en_US.UTF-8"
LC_TELEPHONE="en_US.UTF-8"
LC_MEASUREMENT="en_US.UTF-8"
LC_IDENTIFICATION="en_US.UTF-8"
LC_ALL=
```

Myzhar, 2022-10-27:
I'm not sure that it's related to your problem, but you can fix the missing locale by following this procedure:
```
sudo apt update && sudo apt install locales
sudo locale-gen en_US en_US.UTF-8
sudo update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8
export LANG=en_US.UTF-8
```

peredwardsson, 2022-10-27:
Indeed, exact same problem occurs after running your steps. Any other ideas?

Myzhar, 2022-10-27:
I just noticed that you are running the node with 
`ros2 run zed_wrapper zed_wrapper`
but this is not the correct way because it does not start all the requirements.
You must use the `launch` file, that also loads the URDF and creates the correct TF tree:
`ros2 launch zed_wrapper zed2.launch.py`

peredwardsson, 2022-10-27:
Fantastic, back on track. Must have been the locale stuff I guess.

As a minor note, why have that run file if it can produce these fail states? I have spent some time trying to use only your package to launch Zed Cameras so that my robot_bringup package does not need its own wrapper to a ZED node. But since I can't specify serial numbers as input arguments to your launch files, I have to write my own (I have two cameras). Perhaps something for the future if you agree with the design sentiment. Thanks for a very quick response.

closed 2022-10-27 by peredwardsson