stereolabs/zed-ros2-wrapper #115: Lock timeout

swiz23, 2022-10-28:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

When launching the zed_wrapper node (latest master with all default settings) in docker with debug mode on, there seems to be many times that a 'Lock timeout' message appears. Also, there is quite a deal of lag in the published images. Instead of being published at 60 fps, they are being published around 20 fps.

### Steps to Reproduce

1. Used the same dockerfile as mentioned at https://github.com/stereolabs/zed-ros2-wrapper/issues/112#issuecomment-1292673561
3. Inside the container, I ran `ros2 launch zed_wrapper zed2i.launch.py` (first, I changed the 'debug_mode' parameter to be 'true')
4. Did a ros2 topic hz on the `/zed2i/zed_node/rgb/image_rect_color` topic to see how often it was being published.

### Expected Result

Images published at 60 fps, maybe a 'Lock timeout' message once every 10 seconds or so.

### Actual Result

Images published at 20 fps (or thereabouts), and the 'Lock timeout' message appearing multiple times a second.

### ZED Camera model

ZED2i

### Environment

```shell
OS: Ubuntu 22.04 in Docker (Host is Ubuntu 18.04.6)
CPU: Intel® Core™ i7-10875H
GPU: NVIDIA GeForce GTX 1650 Ti/PCIe/SSE2
ZED SDK version: 3.8
ROS Distro: Humble
```

Myzhar, 2022-10-28:
Hi @swiz23 
I just tested the Dockerfile in #112 and it works as expected. 
I used the new `zed_topic_benchmark` tool to get a reliable estimation of the topic frequency.

After building the image I started it like this:
`docker run -i -t --gpus all --runtime nvidia --privileged -e DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix <image_tag>`
in the docker shell I started the node:
`root@14b9fff61631:/opt/colcon_ws# ros2 launch zed_wrapper zed2i.launch.py`

In a new console, outside the docker container I started the benchmark (you must clone and build the zed-ros2-examples repository):
`$ ros2 run zed_topic_benchmark zed_topic_benchmark --ros-args -p topic_name:=/zed2i/zed_node/rgb/image_rect_color`
and I got
```
[INFO] [1666951770.631084020] [topic_benchmark]: ***** Benchmark parameters *****
[INFO] [1666951770.631223234] [topic_benchmark]: * Topic name: /zed2i/zed_node/rgb/image_rect_color
[INFO] [1666951770.631297334] [topic_benchmark]: Average window size: 500
[INFO] [1666951770.631310841] [topic_benchmark]: *** START BENCHMARK ***
[INFO] [1666951770.632021445] [topic_benchmark]: Advertised on topic: /zed2i/zed_node/rgb/image_rect_color_stats
[INFO] [1666951771.131560747] [topic_benchmark]: Found topic: '/zed2i/zed_node/rgb/image_rect_color' of type: 'sensor_msgs/msg/Image'
#462 - Freq: 60.47 Hz (Avg: 58.55 Hz) - Bandwidth: 1700.77 Mbps (Avg: 1646.78 Mbps) - Msg size: 3.52 MB```

The PC I'm using is a an Intel i5 with Nvidia GTX 1080

swiz23, 2022-10-28:
Thanks for taking a look at this.
When I run the benchmarking node, it looks like I'm getting a much lower average frequency. Not sure if this could be causing the issue, but I currently have the camera hooked up to a usb 3.0 hub, which is in turn connected to my laptop. Unfortunately, I don't have any USB-A ports on my laptop which is the reason I'm using the hub.

`#462 - Freq: 62.25 Hz (Avg: 40.18 Hz) - Bandwidth: 1750.85 Mbps (Avg: 1130.22 Mbps) - Msg size: 3.52`

Myzhar, 2022-10-28:
I guess it's instead a problem of power settings of your laptop.
Laptops running Ubuntu 22 are not using the full available elaboration power, furthermore if powered by batteries.
You can read more here:
https://askubuntu.com/questions/1404521/why-cant-i-turn-on-high-performance-ubuntu-22-04-stock-setting

Myzhar, 2022-11-08:
No, it shouldn't. On zed node side one subscriber or 1000 subscribers doesn't make any different.
It's up to the DDS Middleware to distribute the topic to all the subscribers.

swiz23, 2022-11-08:
makes sense. but no matter how i've tuned dds settings on my end, I can't seem to get it to work correctly. Do you have a recommendation? I'm specifically using Cyclone.

closed 2022-12-15 by github-actions[bot]