stereolabs/zed-ros2-wrapper #125: zed_wrapper launch fails

PallottaEnrico, 2022-12-09:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I'm using the zed 2 on an Nvidia Jetson xavier NX with jetpack 5.0 , i've followed the SDK installation and built zed-ros2-wrapper with Ros2 Foxy.
When launching the zed_wrapper it fails with the error code below.

### Steps to Reproduce

ros2 launch zed_wrapper zed2.launch.py

### Expected Result

zed_wrapper node correctly launched and working

### Actual Result

[INFO] [launch]: All log files can be found below /home/ubm/.ros/log/2022-12-09-16-56-44-629773-f1tenth-31297
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:226> exception=TypeError("Allowed value types are bytes, bool, int, float, str, Sequence[bool], Sequence[int], Sequence[float], Sequence[str]. Got <class 'NoneType'>.If the parameter is meant to be a string, try wrapping it in launch_ros.parameter_descriptions.ParameterValue(value, value_type=str)")>
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 4 more times]
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 444, in execute
    self._perform_substitutions(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 399, in _perform_substitutions
    evaluated_parameters = evaluate_parameters(context, self.__parameters)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 160, in evaluate_parameters
    output_params.append(evaluate_parameter_dict(context, param))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 95, in evaluate_parameter_dict
    raise TypeError(
TypeError: Allowed value types are bytes, bool, int, float, str, Sequence[bool], Sequence[int], Sequence[float], Sequence[str]. Got <class 'NoneType'>.If the parameter is meant to be a string, try wrapping it in launch_ros.parameter_descriptions.ParameterValue(value, value_type=str)
Task exception was never retrieved
future: <Task finished name='Task-8' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:226> exception=RuntimeError('Signal event received before subprocess transport available.')>
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/opaque_function.py", line 75, in execute
    return self.__function(context, *self.__args, **self.__kwargs)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/execute_process.py", line 443, in __on_signal_process_event
    raise RuntimeError('Signal event received before subprocess transport available.')
RuntimeError: Signal event received before subprocess transport available.
[INFO] [robot_state_publisher-1]: process started with pid [31300]
[robot_state_publisher-1] Parsing robot urdf xml string.
[robot_state_publisher-1] Link zed2_base_link had 1 children
[robot_state_publisher-1] Link zed2_camera_center had 4 children
[robot_state_publisher-1] Link zed2_baro_link had 0 children
[robot_state_publisher-1] Link zed2_left_camera_frame had 2 children
[robot_state_publisher-1] Link zed2_left_camera_optical_frame had 0 children
[robot_state_publisher-1] Link zed2_temp_left_link had 0 children
[robot_state_publisher-1] Link zed2_mag_link had 0 children
[robot_state_publisher-1] Link zed2_right_camera_frame had 2 children
[robot_state_publisher-1] Link zed2_right_camera_optical_frame had 0 children
[robot_state_publisher-1] Link zed2_temp_right_link had 0 children
[robot_state_publisher-1] 1670601405.559780102 [zed2.zed_state_publisher] [INFO] got segment base_link
[robot_state_publisher-1] 1670601405.560163598 [zed2.zed_state_publisher] [INFO] got segment zed2_baro_link
[robot_state_publisher-1] 1670601405.560227824 [zed2.zed_state_publisher] [INFO] got segment zed2_base_link
[robot_state_publisher-1] 1670601405.560273137 [zed2.zed_state_publisher] [INFO] got segment zed2_camera_center
[robot_state_publisher-1] 1670601405.560314897 [zed2.zed_state_publisher] [INFO] got segment zed2_left_camera_frame
[robot_state_publisher-1] 1670601405.560356914 [zed2.zed_state_publisher] [INFO] got segment zed2_left_camera_optical_frame
[robot_state_publisher-1] 1670601405.560398547 [zed2.zed_state_publisher] [INFO] got segment zed2_mag_link
[robot_state_publisher-1] 1670601405.560480149 [zed2.zed_state_publisher] [INFO] got segment zed2_right_camera_frame
[robot_state_publisher-1] 1670601405.560554583 [zed2.zed_state_publisher] [INFO] got segment zed2_right_camera_optical_frame
[robot_state_publisher-1] 1670601405.560600056 [zed2.zed_state_publisher] [INFO] got segment zed2_temp_left_link
[robot_state_publisher-1] 1670601405.560671033 [zed2.zed_state_publisher] [INFO] got segment zed2_temp_right_link
[ERROR] [robot_state_publisher-1]: process[robot_state_publisher-1] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [robot_state_publisher-1]: sending signal 'SIGTERM' to process[robot_state_publisher-1]
[ERROR] [robot_state_publisher-1]: process has died [pid 31300, exit code -15, cmd '/opt/ros/foxy/lib/robot_state_publisher/robot_state_publisher --ros-args -r __node:=zed_state_publisher -r __ns:=/zed2 --params-file /tmp/launch_params_g3hxm877'].

### ZED Camera model

ZED2

### Environment

```shell
OS : Ubuntu 20.04
GPU : Nvidia Jetson Xavier NX
ZED SKD version : v.3.8.2
ROS version : Ros2 Foxy
```

Myzhar, 2022-12-09:
Hi @PallottaEnrico 
can I ask you to try to add this parameter to the launch command line?
`svo_path:='Live'`

PallottaEnrico, 2022-12-09:
I've added that parameter, it still gives an error, this is the output:

ros2 launch zed_wrapper zed2.launch.py svo_path:='Live'
[INFO] [launch]: All log files can be found below /home/ubm/.ros/log/2022-12-09-17-30-43-358059-f1tenth-33457
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [33461]
[INFO] [zed_wrapper-2]: process started with pid [33463]
[robot_state_publisher-1] Parsing robot urdf xml string.
[robot_state_publisher-1] Link zed2_base_link had 1 children
[robot_state_publisher-1] Link zed2_camera_center had 4 children
[robot_state_publisher-1] Link zed2_baro_link had 0 children
[robot_state_publisher-1] Link zed2_left_camera_frame had 2 children
[robot_state_publisher-1] Link zed2_left_camera_optical_frame had 0 children
[robot_state_publisher-1] Link zed2_temp_left_link had 0 children
[robot_state_publisher-1] Link zed2_mag_link had 0 children
[robot_state_publisher-1] Link zed2_right_camera_frame had 2 children
[robot_state_publisher-1] Link zed2_right_camera_optical_frame had 0 children
[robot_state_publisher-1] Link zed2_temp_right_link had 0 children
[robot_state_publisher-1] 1670603444.212782089 [zed2.zed_state_publisher] [INFO] got segment base_link
[robot_state_publisher-1] 1670603444.213084273 [zed2.zed_state_publisher] [INFO] got segment zed2_baro_link
[robot_state_publisher-1] 1670603444.213145010 [zed2.zed_state_publisher] [INFO] got segment zed2_base_link
[robot_state_publisher-1] 1670603444.213193107 [zed2.zed_state_publisher] [INFO] got segment zed2_camera_center
[robot_state_publisher-1] 1670603444.213234932 [zed2.zed_state_publisher] [INFO] got segment zed2_left_camera_frame
[robot_state_publisher-1] 1670603444.213311414 [zed2.zed_state_publisher] [INFO] got segment zed2_left_camera_optical_frame
[robot_state_publisher-1] 1670603444.213372728 [zed2.zed_state_publisher] [INFO] got segment zed2_mag_link
[robot_state_publisher-1] 1670603444.213447513 [zed2.zed_state_publisher] [INFO] got segment zed2_right_camera_frame
[robot_state_publisher-1] 1670603444.213521083 [zed2.zed_state_publisher] [INFO] got segment zed2_right_camera_optical_frame
[robot_state_publisher-1] 1670603444.213566140 [zed2.zed_state_publisher] [INFO] got segment zed2_temp_left_link
[robot_state_publisher-1] 1670603444.213606141 [zed2.zed_state_publisher] [INFO] got segment zed2_temp_right_link
[zed_wrapper-2] 1670603444.330255256 [zed2.zed_node] [INFO] ********************************
[zed_wrapper-2] 1670603444.330837543 [zed2.zed_node] [INFO]       ZED Camera Component
[zed_wrapper-2] 1670603444.330938921 [zed2.zed_node] [INFO] ********************************
[zed_wrapper-2] 1670603444.330997962 [zed2.zed_node] [INFO]  * namespace: /zed2
[zed_wrapper-2] 1670603444.331043660 [zed2.zed_node] [INFO]  * node name: zed_node
[zed_wrapper-2] 1670603444.331116397 [zed2.zed_node] [INFO] ********************************
[zed_wrapper-2] 1670603444.331220144 [zed2.zed_node] [INFO] *** DEBUG parameters ***
[zed_wrapper-2] 1670603444.331352499 [zed2.zed_node] [INFO]  * Debug mode: FALSE
[zed_wrapper-2] 1670603444.331463254 [zed2.zed_node] [INFO]  * Debug sensors: FALSE
[zed_wrapper-2] 1670603444.331558296 [zed2.zed_node] [INFO] *** GENERAL parameters ***
[zed_wrapper-2] 1670603444.331854111 [zed2.zed_node] [INFO]  * Camera model: zed2 - ZED 2
[zed_wrapper-2] 1670603444.332030596 [zed2.zed_node] [INFO]  * SDK Verbose: 1
[zed_wrapper-2] 1670603444.332134726 [zed2.zed_node] [INFO]  * SVO: Live
[zed_wrapper-2] 1670603444.332218664 [zed2.zed_node] [INFO]  * SVO Loop: FALSE
[zed_wrapper-2] 1670603444.332297546 [zed2.zed_node] [INFO]  * SVO Realtime: TRUE
[zed_wrapper-2] 1670603444.332407021 [zed2.zed_node] [INFO]  * Camera name: zed2
[zed_wrapper-2] 1670603444.332498095 [zed2.zed_node] [INFO]  * Camera ID: 0
[zed_wrapper-2] 1670603444.332649715 [zed2.zed_node] [INFO]  * Camera SN: 0
[zed_wrapper-2] 1670603444.332775798 [zed2.zed_node] [INFO]  * Camera timeout [sec]: 5
[zed_wrapper-2] 1670603444.332900601 [zed2.zed_node] [INFO]  * Camera reconnection temptatives: 5
[zed_wrapper-2] 1670603444.332996731 [zed2.zed_node] [INFO]  * Camera framerate: 60
[zed_wrapper-2] 1670603444.333081245 [zed2.zed_node] [INFO]  * GPU ID: -1
[zed_wrapper-2] 1670603444.333178688 [zed2.zed_node] [INFO]  * Camera resolution: 2 - HD720
[zed_wrapper-2] 1670603444.333346116 [zed2.zed_node] [INFO]  * Region of interest:
[zed_wrapper-2] 1670603444.333467271 [zed2.zed_node] [INFO]  * Camera self calibration: TRUE
[zed_wrapper-2] 1670603444.333581257 [zed2.zed_node] [INFO]  * Camera flip: FALSE
[zed_wrapper-2] 1670603444.333740365 [zed2.zed_node] [INFO]  * [DYN] Publish framerate [Hz]:  60
[zed_wrapper-2] 1670603444.333802063 [zed2.zed_node] [INFO] *** VIDEO parameters ***
[zed_wrapper-2] 1670603444.333912594 [zed2.zed_node] [INFO]  * Use old extrinsic parameters: 0
[zed_wrapper-2] 1670603444.333995156 [zed2.zed_node] [INFO]  * [DYN] Image downsample factor: 1
[zed_wrapper-2] 1670603444.334092854 [zed2.zed_node] [INFO]  * [DYN] Brightness: 4
[zed_wrapper-2] 1670603444.334175608 [zed2.zed_node] [INFO]  * [DYN] Contrast: 4
[zed_wrapper-2] 1670603444.334298075 [zed2.zed_node] [INFO]  * [DYN] Hue: 0
[zed_wrapper-2] 1670603444.334407902 [zed2.zed_node] [INFO]  * [DYN] Saturation: 4
[zed_wrapper-2] 1670603444.334522080 [zed2.zed_node] [INFO]  * [DYN] Sharpness: 4
[zed_wrapper-2] 1670603444.334601122 [zed2.zed_node] [INFO]  * [DYN] Gamma: 8
[... cut ...]
[zed_wrapper-2] [ZED][ERROR] [Init]  Cannot load SVO file. Check that the path of the SVO is correct, avoid white spaces in filename, add ".svo" extension to filename.
[ERROR] [zed_wrapper-2]: process has died [pid 33463, exit code 1, cmd '/home/ubm/zed2ws/install/zed_wrapper/lib/zed_wrapper/zed_wrapper --ros-args -r __node:=zed_node -r __ns:=/zed2 --params-file /home/ubm/zed2ws/install/zed_wrapper/share/zed_wrapper/config/common.yaml --params-file /home/ubm/zed2ws/install/zed_wrapper/share/zed_wrapper/config/zed2.yaml --params-file /tmp/launch_params_jqayj4cl'].

Myzhar, 2022-12-09:
@PallottaEnrico I'm sorry, `svo_path:='live'`, without the capital 'L' letter

PallottaEnrico, 2022-12-09:
@Myzhar works now, thank you!

Myzhar, 2022-12-09:
I'm sorry for the caused inconvenience, we will fix the launch file as soon as possible