stereolabs/zed-ros2-wrapper #126: Loading composable nodes into a container

MrOCW, 2022-12-15:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Attempting to launch the stereolabs::ZedCamera node into a container using a python launch file returns the following error:
```
Failed to load node 'zed_node' of type 'stereolabs::ZedCamera' in container '/zed_container': Component constructor threw an exception: Invalid topic name: topic name must not be empty string:
  ''
   ^
```

### Steps to Reproduce

Replace zed_camera.launch.py:
```
#!/usr/bin/env python3

import os

from ament_index_python.packages import get_package_share_directory

from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument, SetLaunchConfiguration
from launch.conditions import IfCondition, UnlessCondition
from launch.substitutions import LaunchConfiguration, Command
from launch_ros.actions import Node, ComposableNodeContainer
from launch_ros.descriptions import ComposableNode

def generate_launch_description():
    # Launch configuration variables
    svo_path = LaunchConfiguration('svo_path')

    camera_name = LaunchConfiguration('camera_name')
    camera_model = LaunchConfiguration('camera_model')

    node_name = LaunchConfiguration('node_name')

    config_common_path = LaunchConfiguration('config_common_path')
    config_camera_path = LaunchConfiguration('config_camera_path')

    base_frame = LaunchConfiguration('base_frame')
    cam_pos_x = LaunchConfiguration('cam_pos_x')
    cam_pos_y = LaunchConfiguration('cam_pos_y')
    cam_pos_z = LaunchConfiguration('cam_pos_z')
    cam_roll = LaunchConfiguration('cam_roll')
    cam_pitch = LaunchConfiguration('cam_pitch')
    cam_yaw = LaunchConfiguration('cam_yaw')

    publish_urdf = LaunchConfiguration('publish_urdf')
    xacro_path = LaunchConfiguration('xacro_path')

    # ZED Configurations to be loaded by ZED Node
    default_config_common = os.path.join(
        get_package_share_directory('zed_wrapper'),
        'config',
        'common.yaml'
    )

    # URDF/xacro file to be loaded by the Robot State Publisher node
    default_xacro_path = os.path.join(
        get_package_share_directory('zed_wrapper'),
        'urdf',
        'zed_descr.urdf.xacro'
    )

    # Declare the launch arguments
    declare_camera_name_cmd = DeclareLaunchArgument(
[... 163 lines cut ...]
```

### Expected Result

Loads the composable node into the composable node container and publishes topics

### Actual Result

```
Failed to load node 'zed_node' of type 'stereolabs::ZedCamera' in container '/zed_container': Component constructor threw an exception: Invalid topic name: topic name must not be empty string:
  ''
   ^
```

### ZED Camera model

ZED2i

### Environment

```shell
Jetson Orin AGX
Jetpack 5.0.2
ROS2 Foxy
```

MrOCW, 2022-12-15:
@Myzhar for some reason, it seems that the config parameters arent being loaded properly. overriding depth.quality: 0 seems to work fine but when depth.quality: 1,2,3 or 4, the issue surfaces

Myzhar, 2022-12-15:
The first thing that I see is this

`namespace="/",` is wrong.
The namespace must be the same for each node:
`namespace=camera_name`

but I can't exclude that there's something other wrong in your launch file.

MrOCW, 2022-12-15:
I've changed the namespaces to zed2i, still having the same issue

Myzhar, 2022-12-15:
I suspect it's something related to this:
https://github.com/ros2/launch_ros/issues/185

Myzhar, 2022-12-15:
> I've changed the namespaces to zed2i, still having the same issue. If the issue was the namespace, why does this seem to happen for depth only? shouldnt it apply to regular images as well?

Can you post the full output log?

MrOCW, 2022-12-15:
```
[robot_state_publisher-1] Parsing robot urdf xml string.
[robot_state_publisher-1] Link zed2i_base_link had 1 children
[robot_state_publisher-1] Link zed2i_camera_center had 2 children
[robot_state_publisher-1] Link zed2i_left_camera_frame had 1 children
[robot_state_publisher-1] Link zed2i_left_camera_optical_frame had 0 children
[robot_state_publisher-1] Link zed2i_right_camera_frame had 1 children
[robot_state_publisher-1] Link zed2i_right_camera_optical_frame had 0 children
[robot_state_publisher-1] 1671088233.996592800 [zed2i.zed_state_publisher] [INFO] got segment base_link
[robot_state_publisher-1] 1671088233.996710370 [zed2i.zed_state_publisher] [INFO] got segment zed2i_base_link
[robot_state_publisher-1] 1671088233.996726659 [zed2i.zed_state_publisher] [INFO] got segment zed2i_camera_center
[robot_state_publisher-1] 1671088233.996736163 [zed2i.zed_state_publisher] [INFO] got segment zed2i_left_camera_frame
[robot_state_publisher-1] 1671088233.996743619 [zed2i.zed_state_publisher] [INFO] got segment zed2i_left_camera_optical_frame
[robot_state_publisher-1] 1671088233.996750819 [zed2i.zed_state_publisher] [INFO] got segment zed2i_right_camera_frame
[robot_state_publisher-1] 1671088233.996757859 [zed2i.zed_state_publisher] [INFO] got segment zed2i_right_camera_optical_frame
[component_container-2] 1671088234.231927745 [zed2i.zed_container] [INFO] Load Library: /home/warthog/external/zed/install/lib/libzed_camera_component.so
[component_container-2] 1671088234.269301201 [zed2i.zed_container] [INFO] Found class: rclcpp_components::NodeFactoryTemplate<stereolabs::ZedCamera>
[component_container-2] 1671088234.269406803 [zed2i.zed_container] [INFO] Instantiate class: rclcpp_components::NodeFactoryTemplate<stereolabs::ZedCamera>
[component_container-2] 1671088234.273205792 [zed2i.zed_node] [INFO] ********************************
[component_container-2] 1671088234.273267329 [zed2i.zed_node] [INFO]       ZED Camera Component 
[component_container-2] 1671088234.273315810 [zed2i.zed_node] [INFO] ********************************
[component_container-2] 1671088234.273344867 [zed2i.zed_node] [INFO]  * namespace: /zed2i
[component_container-2] 1671088234.273360099 [zed2i.zed_node] [INFO]  * node name: zed_node
[component_container-2] 1671088234.273392452 [zed2i.zed_node] [INFO] ********************************
[component_container-2] 1671088234.273533638 [zed2i.zed_node] [INFO] *** DEBUG parameters ***
[component_container-2] 1671088234.273689961 [zed2i.zed_node] [INFO]  * Debug mode: FALSE
[component_container-2] 1671088234.273779947 [zed2i.zed_node] [INFO]  * Debug sensors: FALSE
[component_container-2] 1671088234.273848941 [zed2i.zed_node] [INFO] *** GENERAL parameters ***
[component_container-2] 1671088234.274002096 [zed2i.zed_node] [INFO]  * Camera model: zed2i - ZED 2i
[component_container-2] 1671088234.274098354 [zed2i.zed_node] [INFO]  * SDK Verbose: 1
[component_container-2] 1671088234.274225268 [zed2i.zed_node] [INFO]  * SVO: live
[component_container-2] 1671088234.274303862 [zed2i.zed_node] [INFO]  * SVO Loop: FALSE
[component_container-2] 1671088234.274361687 [zed2i.zed_node] [INFO]  * SVO Realtime: FALSE
[component_container-2] 1671088234.274416152 [zed2i.zed_node] [INFO]  * Camera name: zed2i
[component_container-2] 1671088234.274476441 [zed2i.zed_node] [INFO]  * Camera ID: 0
[component_container-2] 1671088234.274530362 [zed2i.zed_node] [INFO]  * Camera SN: 0
[component_container-2] 1671088234.274589532 [zed2i.zed_node] [INFO]  * Camera timeout [sec]: 5
[component_container-2] 1671088234.274644285 [zed2i.zed_node] [INFO]  * Camera reconnection temptatives: 5
[component_container-2] 1671088234.274697118 [zed2i.zed_node] [INFO]  * Camera framerate: 15
[component_container-2] 1671088234.274754495 [zed2i.zed_node] [INFO]  * GPU ID: -1
[component_container-2] 1671088234.274819392 [zed2i.zed_node] [INFO]  * Camera resolution: 3 - VGA
[component_container-2] 1671088234.274900386 [zed2i.zed_node] [INFO]  * Region of interest: []
[component_container-2] 1671088234.274955459 [zed2i.zed_node] [INFO]  * Camera self calibration: TRUE
[component_container-2] 1671088234.275015268 [zed2i.zed_node] [INFO]  * Camera flip: FALSE
[component_container-2] 1671088234.275091622 [zed2i.zed_node] [INFO]  * [DYN] Publish framerate [Hz]:  15
[component_container-2] 1671088234.275112358 [zed2i.zed_node] [INFO] *** VIDEO parameters ***
[component_container-2] 1671088234.275207112 [zed2i.zed_node] [INFO]  * Use old extrinsic parameters: 0
[component_container-2] 1671088234.275514702 [zed2i.zed_node] [INFO]  * [DYN] Image downsample factor: 1 
[component_container-2] 1671088234.275624368 [zed2i.zed_node] [INFO]  * [DYN] Brightness: 4
[component_container-2] 1671088234.275931799 [zed2i.zed_node] [INFO]  * [DYN] Contrast: 4
[component_container-2] 1671088234.276016216 [zed2i.zed_node] [INFO]  * [DYN] Hue: 0
[component_container-2] 1671088234.276083738 [zed2i.zed_node] [INFO]  * [DYN] Saturation: 4
[component_container-2] 1671088234.276140795 [zed2i.zed_node] [INFO]  * [DYN] Sharpness: 4
[component_container-2] 1671088234.276240317 [zed2i.zed_node] [INFO]  * [DYN] Gamma: 8
[component_container-2] 1671088234.276301662 [zed2i.zed_node] [INFO]  * [DYN] Auto Exposure/Gain: TRUE
[component_container-2] 1671088234.276354911 [zed2i.zed_node] [INFO]  * [DYN] Exposure: 80
[component_container-2] 1671088234.276413600 [zed2i.zed_node] [INFO]  * [DYN] Gain: 80
[... cut ...]
[component_container-2] 1671088236.743796852 [zed2i.zed_container] [ERROR] Component constructor threw an exception: Invalid topic name: topic name must not be empty string:
[component_container-2]   ''
[component_container-2]    ^
[component_container-2] 
[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'zed_node' of type 'stereolabs::ZedCamera' in container '/zed2i/zed_container': Component constructor threw an exception: Invalid topic name: topic name must not be empty string:
  ''
   ^

```

Myzhar, 2022-12-15:
Instead, if you set the `depth.quality` to `0` in `common.yaml` it works, right?

MrOCW, 2022-12-15:
nope, it does not read the common.yaml. i have to set it in the launch file like this
```
                ComposableNode(
                    package='zed_components',
                    plugin='stereolabs::ZedCamera',
                    name="zed_node",
                    parameters=[
                        # YAML files
                        config_common_path,  # Common parameters
                        config_camera_path,  # Camera related parameters
                        # Overriding
                        {
                            'general.camera_name': camera_name,
                            'general.camera_model': camera_model,
                            'depth.quality':0,
                            'general.svo_file': svo_path,
                            'pos_tracking.base_frame': base_frame
                        }
                    ],
                ),
            ],
        output="screen",
```

Myzhar, 2022-12-15:
OK, how are you passing the `config_camera_path` value? It seems empty in the launch file

MrOCW, 2022-12-15:
I run the zed2i.launch.py which passes it into this launch file

Myzhar, 2022-12-15:
Does it work if you change `'depth.quality':0,` using other depth modes?

MrOCW, 2022-12-15:
do you mean using 1,2,3,4? no it does not work. Only 0 works

Myzhar, 2022-12-15:
I was trying to test your composable launch file, but it does not adapt to the new launch files that are available in the latest version of the `master` branch.
I pushed a big update on 6 Dec that allows using the value of the `camera_model` launch parameter in the `zed_camera.launch.py` to automatically load the correct default camera configuration file.
The new launch file uses indeed the powerful `OpaqueFunction` launch action.
I suggest you upgrade your repository and try to adapt your "Composition" launch file to the new version of `zed_camera.launch.py` which is more reliable.

MrOCW, 2022-12-15:
updated to latest version. exactly the same issue.

Myzhar, 2022-12-15:
OK, please share the launch file so I can test it.
However, I suspect that the problem is related to the Foxy launch system and the issue in the link I shared above.
The composition using a C++ container works correctly as you can test with the RGBD->RGB converter example:
https://github.com/stereolabs/zed-ros2-examples/tree/master/tutorials/zed_rgb_convert

MrOCW, 2022-12-15:
upgrading it to humble fixed this issue indeed! thank you @Myzhar for your help :)

closed 2022-12-15 by MrOCW

Myzhar, 2022-12-15:
Thanks for confirming that it was a problem with Foxy