stereolabs/zed-ros2-wrapper #128: disto CameraParameter contains only first five coefficients

ozendelait, 2022-12-15:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

The distortion coefficients within the zed2-ros2-wrapper camera_info messages (as well as the disto CameraParameters returned by the ZED API for calibration_parameters_raw
https://www.stereolabs.com/docs/api/structsl_1_1CameraParameters.html#a5e66eff1633b3ba90dbbf12121163179 ) only contains the first five coefficients k1 k2 p1 p2 k3 even if the conf File contains k4 k5 k6 (or s1 s2 s3 s4).
I confirmed that the additional rational/thin prism coefficients are indeed applied to correctly undistort the images, however only the first five of the 8/12 parameters are returned by the API

### Steps to Reproduce

1. Have a camera connected /svo with a .conf file where the [LEFT_DISTO]/[RIGHT_DISTO] contains values for k4,k5,k6 
2. Launch ROS2 node zed2-ros2-wrapper; general/self_calib (==camera_disable_self_calib) set to false
3. check ../left_raw/camera_info distortion coefficients (parameter "D", "distortion_model")

### Expected Result

D has 8 coefficients
distortion_model is "rational_polynomial"

### Actual Result

D has 5 coefficients (the first 5 of the expected 8)
distortion_model is "plumb_bob"

### ZED Camera model

ZED2i

### Environment

```shell
ZED SDK 3.8.2
```

### Anything else?

Same bug when checking via python ZED API (e.g. pyzed camera.get_camera_information().calibration_parameters_raw.left_cam.disto )
The ROS1/ROS2 wrappers have hard-coded plumb_bob/resize(5) behaviour:
https://github.com/stereolabs/zed-ros2-wrapper/blob/master/zed_components/src/zed_camera/src/zed_camera_component.cpp#L1941
Changing k4/k5/k6 in the conf file does change the rectified images -> Coeffs are used

Myzhar, 2022-12-15:
Hi @ozendelait 
can you please compile and run this simple test program and post the output?
[test_camera_params.zip](https://github.com/stereolabs/zed-ros2-wrapper/files/10238678/test_camera_params.zip)

ozendelait, 2022-12-15:
Hi, yes; correct. I get this:
* Left sensor rectified: 
  - fx: 521.396
  - fy: 521.396
  - h_fov: 101.665
  - v_fov: 69.2487

* Right sensor rectified: 
  - fx: 521.396
  - fy: 521.396
  - h_fov: 101.665
  - v_fov: 69.2487

* Left sensor raw: 
  - fx: 534.12
  - fy: 534.125
  - h_fov: 105.628
  - v_fov: 73.1086
  - distortions: -0.92338 2.25279 0.000322705 3.9471e-05 0.0839577 -0.826327 2.11526 0.234659 0 0 0 0 

* Right sensor raw: 
  - fx: 534.72
  - fy: 534.605
  - h_fov: 105.566
  - v_fov: 73.0594
  - distortions: -1.02549 2.26474 -0.000252286 -5.28801e-05 0.184219 -0.924819 2.10794 0.353717 0 0 0 0

Myzhar, 2022-12-15:
OK, I noted the modification to be applied by the ZED Wrapper. 
I will add it in the next few days if I find the right moment.
Regarding Python output, I reported the issue to the Python team.

Myzhar, 2022-12-19:
Fixed https://github.com/stereolabs/zed-ros2-wrapper/commit/f3d9fa719cdda165d6ee0835faad98b6d7690f5a

closed 2022-12-19 by Myzhar