stereolabs/zed-ros2-wrapper #132: Unable to build wrapper on Jetpack 5.0.2 w/ ZED SDK 3.8.2

acampbellb, 2022-12-30:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I am getting a bunch of warnings and eventually some unknown errors building the wrapper. I am installing the wrapper
on Jetpack 5.0.2 w/ ZED SDK 3.8.2. I have followed the exact steps documented in
https://www.stereolabs.com/blog/getting-started-with-ros2-on-jetson-agx-orin/

Any idea what the issue might be?

### Steps to Reproduce

mkdir -p ~/ros2_ws/src
cd ~/ros2_ws/src/
git clone  --recursive https://github.com/stereolabs/zed-ros2-wrapper.git
cd ..
rosdep install --from-paths src --ignore-src -r -y
colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release

### Expected Result

Wrapper components are built

### Actual Result

colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release
Starting >>> zed_interfaces
[1.358s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/jetson/ros2_ws/install/zed_multi_camera' in the environment variable AMENT_PREFIX_PATH doesn't exist
[1.359s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/jetson/ros2_ws/install/zed_display_rviz2' in the environment variable AMENT_PREFIX_PATH doesn't exist
[1.359s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/jetson/ros2_ws/install/zed_tutorial_video' in the environment variable AMENT_PREFIX_PATH doesn't exist
[1.359s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/jetson/ros2_ws/install/zed_tutorial_pos_tracking' in the environment variable AMENT_PREFIX_PATH doesn't exist
[1.360s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/jetson/ros2_ws/install/zed_tutorial_depth' in the environment variable AMENT_PREFIX_PATH doesn't exist
[1.360s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/jetson/ros2_ws/install/zed_topic_benchmark_interfaces' in the environment variable AMENT_PREFIX_PATH doesn't exist
[1.360s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/jetson/ros2_ws/install/zed_rgb_convert' in the environment variable AMENT_PREFIX_PATH doesn't exist
[1.361s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/jetson/ros2_ws/install/rviz_plugin_zed_od' in the environment variable AMENT_PREFIX_PATH doesn't exist
[1.361s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/jetson/ros2_ws/install/zed_ros2' in the environment variable AMENT_PREFIX_PATH doesn't exist
[1.361s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/jetson/ros2_ws/install/zed_wrapper' in the environment variable AMENT_PREFIX_PATH doesn't exist
[1.362s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/jetson/ros2_ws/install/zed_components' in the environment variable AMENT_PREFIX_PATH doesn't exist
[1.362s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/jetson/ros2_ws/install/zed_interfaces' in the environment variable AMENT_PREFIX_PATH doesn't exist
[1.362s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/jetson/ros2_ws/install/zed_multi_camera' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[1.363s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/jetson/ros2_ws/install/zed_display_rviz2' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[1.363s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/jetson/ros2_ws/install/zed_tutorial_video' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[1.363s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/jetson/ros2_ws/install/zed_tutorial_pos_tracking' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[1.363s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/jetson/ros2_ws/install/zed_tutorial_depth' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[1.364s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/jetson/ros2_ws/install/zed_topic_benchmark_interfaces' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[1.364s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/jetson/ros2_ws/install/zed_rgb_convert' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[1.364s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/jetson/ros2_ws/install/rviz_plugin_zed_od' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[1.364s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/jetson/ros2_ws/install/zed_ros2' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[1.365s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/jetson/ros2_ws/install/zed_wrapper' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[1.365s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/jetson/ros2_ws/install/zed_components' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[1.365s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/jetson/ros2_ws/install/zed_interfaces' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[Processing: zed_interfaces]                             
[Processing: zed_interfaces]                                     
[Processing: zed_interfaces]                                        
--- stderr: zed_interfaces                                       
* Found ROS2 foxy
* ROS2 foxy is officially supported by this package.
---
Finished <<< zed_interfaces [1min 36s]
Starting >>> zed_components
[Processing: zed_components]                                  
[Processing: zed_components]                                  
[Processing: zed_components]                                       
[Processing: zed_components]                                       
[Processing: zed_components]                                       
[Processing: zed_components]                                       
[Processing: zed_components]                                       
--- stderr: zed_components                                      
* Found ROS2 foxy
* ROS2 foxy is officially supported by this package.
---
Finished <<< zed_components [3min 32s]
Starting >>> zed_wrapper
[Processing: zed_wrapper]                                  
--- stderr: zed_wrapper                                     
* Found ROS2 foxy
* ROS2 foxy is officially supported by this package.
---
Finished <<< zed_wrapper [38.3s]
Starting >>> zed_ros2
--- stderr: zed_ros2                                 
* Found ROS2 foxy
* ROS2 foxy is officially supported by this package.
---
Finished <<< zed_ros2 [3.14s]

Summary: 4 packages finished [5min 50s]
  4 packages had stderr output: zed_components zed_interfaces zed_ros2 zed_wrapper

### ZED Camera model

ZED2i

### Environment

```shell
OS: Ubuntu 20.04
Jetpack: 5.0.2
ZED SDK: 382
Hardware: Jetson Xavier NX
```

Myzhar, 2022-12-30:
@acampbellb there are no error messages. All the packages are correctly built.
There are a few warnings about missing pre-installed packages, try to clone and install also the `zed-ros2-examples` repository

acampbellb, 2022-12-30:
This does not indicate an error?
4 packages had stderr output: zed_components zed_interfaces zed_ros2 zed_wrapper

acampbellb, 2022-12-30:
I was confused by the stderr output, it appears this is fine.

closed 2022-12-30 by acampbellb