stereolabs/zed-ros2-wrapper #134: Missing xarco file when trying to run zed2.launch

Kars007, 2023-01-08:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hello currently running a docker container with foxy on it. I installed zed sdk, as well as zed wrapper. But when running ros2 launch zed_wrapper zed2.launch.py,
the following error happened. it is missing a FileNotFoundError: [Errno 2] No such file or directory: 'xacro'. What is going wrong?

root@a269d5d0e6f6:/ros_ws# ros2 launch zed_wrapper zed2.launch.py
[INFO] [launch]: All log files can be found below /root/.ros/log/2023-01-08-19-10-44-198961-a269d5d0e6f6-74
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /ros_ws/build/launch/launch/launch_service.py:226> exception=SubstitutionFailure("file not found: [Errno 2] No such file or directory: 'xacro'")>
Traceback (most recent call last):
  File "/ros_ws/build/launch/launch/substitutions/command.py", line 108, in perform
    result = subprocess.run(
  File "/usr/lib/python3.8/subprocess.py", line 493, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.8/subprocess.py", line 858, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.8/subprocess.py", line 1704, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'xacro'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/ros_ws/build/launch/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/ros_ws/build/launch/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/ros_ws/build/launch/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/ros_ws/build/launch/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/ros_ws/build/launch/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 4 more times]
  File "/ros_ws/build/launch/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/ros_ws/build/launch/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/install/lib/python3.8/site-packages/launch_ros/actions/node.py", line 444, in execute
    self._perform_substitutions(context)
  File "/opt/ros/foxy/install/lib/python3.8/site-packages/launch_ros/actions/node.py", line 399, in _perform_substitutions
    evaluated_parameters = evaluate_parameters(context, self.__parameters)
  File "/opt/ros/foxy/install/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 160, in evaluate_parameters
    output_params.append(evaluate_parameter_dict(context, param))
  File "/opt/ros/foxy/install/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 71, in evaluate_parameter_dict
    evaluated_value = perform_substitutions(context, list(value))
  File "/ros_ws/build/launch/launch/utilities/perform_substitutions_impl.py", line 26, in perform_substitutions
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/ros_ws/build/launch/launch/utilities/perform_substitutions_impl.py", line 26, in <listcomp>
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/ros_ws/build/launch/launch/launch_context.py", line 232, in perform_substitution
    return substitution.perform(self)
  File "/ros_ws/build/launch/launch/substitutions/command.py", line 114, in perform
    raise SubstitutionFailure(f'file not found: {ex}')
launch.substitutions.substitution_failure.SubstitutionFailure: file not found: [Errno 2] No such file or directory: 'xacro'

### Steps to Reproduce

1. Run docker container dustynv/ros:foxy-pytorch-l4t-r35.1.0
2. Install ZED sdk 3.8
3. source ${ROS_ENVIRONMENT} && \
    cd ${ROS_ROOT} && \
    mkdir src/slam && \
    rosinstall_generator --deps --exclude-path ros_ws/src --rosdistro foxy \
    diagnostic_updater \
    xacro \
    > ros2.foxy.zed.rosinstall && \
    cat ros2.foxy.zed.rosinstall && \
    vcs import src/slam < ros2.foxy.zed.rosinstall && \
    apt-get update && \
    rosdep install --from-paths src/slam --ignore-src --rosdistro foxy -y --skip-keys "Pangolin libopencv-dev libopencv-contrib-dev libopencv-imgproc-dev python-opencv python3-opencv" && \
    rm -rf /var/lib/apt/lists/* && \
    apt-get clean && \
    colcon build --symlink-install --base-paths src/slam
4. source ${ROS_ENVIRONMENT} && \
    cd ${ROS_ROOT} && \
    git clone https://github.com/stereolabs/zed-ros2-wrapper src/slam/zed-ros2-wrapper && \
    apt-get update && \
    rosdep install --from-paths src/slam --ignore-src --rosdistro foxy -y --skip-keys "rtabmap find_object_2d Pangolin libopencv-dev libopencv-contrib-dev libopencv-imgproc-dev python-opencv python3-opencv" && \
    rm -rf /var/lib/apt/lists/* && \
    apt-get clean && \
    colcon build --symlink-install --base-paths src/slam/zed-ros2-wrapper --cmake-args=-DCMAKE_BUILD_TYPE=Release
5.  ros2 launch zed_wrapper zed2.launch.py,
...

### Expected Result

run like it suppose to do with camera topic and so on.

### Actual Result

Erorr as mentioned above

### ZED Camera model

ZED2

### Environment

```shell
OS: Linux Ubuntu 20.04 Jepack 5.0.2
CPU: ARm
GPU: NVIDIA jetson Xavier
ZED SDK version 3.8
ROS2 version: foxy
```

Myzhar, 2023-01-08:
Hi @Kars007 
Dusty's docker images do not contain the `xacro` package, so you must install it from the source.

Myzhar, 2023-01-08:
Hi @Kars007 
This is not a bug, but a support request.
Please write a post on [Stereolabs community](https://community.stereolabs.com) to obtain direct support

closed 2023-01-08 by Myzhar