stereolabs/zed-ros2-wrapper #136: Error opening camera: NO GPU DETECTED

lucyannofrota, 2023-02-20:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I'm trying to use the ZED2 camera in a docker container. The docker container has ROS2 Foxy installed and the ZED SDK. The ZED wrapper was built without a problem.

I got the error: "Error opening camera: NO GPU DETECTED" when I try to run `ros2 launch zed_wrapper zed2.launch.py`.

### Steps to Reproduce

1. Run the container
> `xhost +si:localuser:root`
> `docker run --gpus all -it --privileged -e DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix  stereolabs/zed:3.8-gl-devel-cuda11.4-ubuntu20.04`

2. ros2 launch zed_wrapper zed2.launch.py

### Expected Result

Being able to launch the node

### Actual Result

> ros2 launch zed_wrapper zed2.launch.py 

[INFO] [launch]: All log files can be found below /home/ros/.ros/log/2023-02-20-20-01-38-583565-ISR-Desktop-453653
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [453663]
[INFO] [zed_wrapper-2]: process started with pid [453665]
[robot_state_publisher-1] Parsing robot urdf xml string.
[robot_state_publisher-1] Link zed2_base_link had 1 children
[robot_state_publisher-1] Link zed2_camera_center had 4 children
[robot_state_publisher-1] Link zed2_baro_link had 0 children
[robot_state_publisher-1] Link zed2_left_camera_frame had 2 children
[robot_state_publisher-1] Link zed2_left_camera_optical_frame had 0 children
[robot_state_publisher-1] Link zed2_temp_left_link had 0 children
[robot_state_publisher-1] Link zed2_mag_link had 0 children
[robot_state_publisher-1] Link zed2_right_camera_frame had 2 children
[robot_state_publisher-1] Link zed2_right_camera_optical_frame had 0 children
[robot_state_publisher-1] Link zed2_temp_right_link had 0 children
[robot_state_publisher-1] [INFO] [1676923298.810269616] [zed2.zed_state_publisher]: got segment base_link
[robot_state_publisher-1] [INFO] [1676923298.810340142] [zed2.zed_state_publisher]: got segment zed2_baro_link
[robot_state_publisher-1] [INFO] [1676923298.810350758] [zed2.zed_state_publisher]: got segment zed2_base_link
[robot_state_publisher-1] [INFO] [1676923298.810358424] [zed2.zed_state_publisher]: got segment zed2_camera_center
[robot_state_publisher-1] [INFO] [1676923298.810365561] [zed2.zed_state_publisher]: got segment zed2_left_camera_frame
[robot_state_publisher-1] [INFO] [1676923298.810372624] [zed2.zed_state_publisher]: got segment zed2_left_camera_optical_frame
[robot_state_publisher-1] [INFO] [1676923298.810379578] [zed2.zed_state_publisher]: got segment zed2_mag_link
[robot_state_publisher-1] [INFO] [1676923298.810386440] [zed2.zed_state_publisher]: got segment zed2_right_camera_frame
[robot_state_publisher-1] [INFO] [1676923298.810393285] [zed2.zed_state_publisher]: got segment zed2_right_camera_optical_frame
[robot_state_publisher-1] [INFO] [1676923298.810400098] [zed2.zed_state_publisher]: got segment zed2_temp_left_link
[robot_state_publisher-1] [INFO] [1676923298.810406934] [zed2.zed_state_publisher]: got segment zed2_temp_right_link
[zed_wrapper-2] [INFO] [1676923298.823930703] [zed2.zed_node]: ********************************
[zed_wrapper-2] [INFO] [1676923298.824029919] [zed2.zed_node]:       ZED Camera Component 
[zed_wrapper-2] [INFO] [1676923298.824042079] [zed2.zed_node]: ********************************
[zed_wrapper-2] [INFO] [1676923298.824050966] [zed2.zed_node]:  * namespace: /zed2
[zed_wrapper-2] [INFO] [1676923298.824059983] [zed2.zed_node]:  * node name: zed_node
[zed_wrapper-2] [INFO] [1676923298.824067461] [zed2.zed_node]: ********************************
[zed_wrapper-2] [INFO] [1676923298.824085818] [zed2.zed_node]: *** DEBUG parameters ***
[zed_wrapper-2] [INFO] [1676923298.824114016] [zed2.zed_node]:  * Debug mode: FALSE
[zed_wrapper-2] [INFO] [1676923298.824138156] [zed2.zed_node]:  * Debug sensors: FALSE
[zed_wrapper-2] [INFO] [1676923298.824150443] [zed2.zed_node]: *** GENERAL parameters ***
[zed_wrapper-2] [INFO] [1676923298.824178806] [zed2.zed_node]:  * Camera model: zed2 - ZED 2
[zed_wrapper-2] [INFO] [1676923298.824206457] [zed2.zed_node]:  * SDK Verbose: 1
[zed_wrapper-2] [INFO] [1676923298.824230790] [zed2.zed_node]:  * SVO: live
[zed_wrapper-2] [INFO] [1676923298.824249800] [zed2.zed_node]:  * SVO Loop: FALSE
[zed_wrapper-2] [INFO] [1676923298.824266142] [zed2.zed_node]:  * SVO Realtime: TRUE
[zed_wrapper-2] [INFO] [1676923298.824283309] [zed2.zed_node]:  * Camera name: zed2
[zed_wrapper-2] [INFO] [1676923298.824305322] [zed2.zed_node]:  * Camera ID: 0
[... 145 lines cut ...]

ZED2

### Environment

```shell
OS: Ubuntu 20.04
CPU: Intel(R) Core(TM) i7-3770
GPU: 2x GTX 780

Docker: 23.0.1
ROS: Foxy

NVIDIA-SMI 470.161.03   Driver Version: 470.161.03   CUDA Version: 11.7

nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2022 NVIDIA Corporation
Built on Wed_Jun__8_16:49:14_PDT_2022
Cuda compilation tools, release 11.7, V11.7.99
Build cuda_11.7.r11.7/compiler.31442593_0
```

### Anything else?

Inside the container, I was able to execute applications using the GPU like OpengGL apps.

lucyannofrota, 2023-02-22:
Thank you for your reply @Myzhar.
There was confusion in the information that I presented earlier. I'm not using the container `stereolabs/zed:3.8-gl-devel-cuda11.4-ubuntu20.04` this container don't even has ROS installed! Sory about that.

I'm actually running a custom container based on `nvidia/cuda:11.7.0-devel-ubuntu20.04`.

I've installed the ZED dependencies based on: https://github.com/stereolabs/zed-docker/blob/master/3.X/ubuntu/devel/Dockerfile.

### The ZED SDK part of the dockerfile looks like this:

ENV NVIDIA_VISIBLE_DEVICES all

\# ENV NVIDIA_DRIVER_CAPABILITIES \
\#    ${NVIDIA_DRIVER_CAPABILITIES:+$NVIDIA_DRIVER_CAPABILITIES,}compute,video,utility,graphics

ENV NVIDIA_DRIVER_CAPABILITIES all

RUN echo "Europe/Paris" > /etc/localtime ; echo "CUDA Version 11.7" > /usr/local/cuda/version.txt

USER ros
ENV USER=ros
\# Install zstd
RUN sudo apt-get update -y || true
RUN sudo apt-get install --no-install-recommends zstd

\# Setup the ZED SDK
RUN sudo apt-get update -y || true
RUN sudo apt-get install --no-install-recommends lsb-release wget less udev sudo build-essential cmake -y
RUN sudo wget -q -O ZED_SDK_38_Linux_Ubuntu20.run https://download.stereolabs.com/zedsdk/3.8/cu117/ubuntu20
RUN sudo chmod +x ZED_SDK_38_Linux_Ubuntu20.run

RUN ./ZED_SDK_38_Linux_Ubuntu20.run -- silent

RUN sudo ln -sf /lib/x86_64-linux-gnu/libusb-1.0.so.0 /usr/lib/x86_64-linux-gnu/libusb-1.0.so \ 
    && sudo rm ZED_SDK_38_Linux_Ubuntu20.run \
    && sudo rm -rf /var/lib/apt/lists/*

FROM zed-sdk as zed-api

\# # ZED Python API
RUN sudo apt-get update -y || true
RUN sudo apt-get install --no-install-recommends python3 python3-pip libpng-dev libgomp1 -y && \ 
    wget download.stereolabs.com/zedsdk/pyzed -O /usr/local/zed/get_python_api.py && \
    python3 /usr/local/zed/get_python_api.py && \
    python3 -m pip install numpy opencv-python pyopengl *.whl && \
    sudo rm *.whl ; sudo rm -rf /var/lib/apt/lists/*

\# # Make some tools happy
RUN sudo mkdir -p /root/Documents/ZED/

\# ROS diagnostic_updater
RUN sudo apt-get update -y || true
RUN sudo apt-get install --no-install-recommends \
    ros-foxy-diagnostic-updater

### Extra notes:
I already had the variables `NVIDIA_VISIBLE_DEVICES` and `NVIDIA_DRIVER_CAPABILITIES` set as you suggested.

I don think the problem is related to those variables because I was able to run Gazebo with OpenGL and GPU support.

Myzhar, 2023-02-22:
Hi @lucyannofrota 
I suggest you compare your dockerfile with this:
https://github.com/stereolabs/zed-ros2-wrapper/blob/master/docker/Dockerfile.u22-cu117-humble-release
available in the `docker` folder of this repository.
It's highly optimized and it contains the latest updates.

lucyannofrota, 2023-02-22:
I tried to use the suggested container as it is as well as the altered version (Ubuntu 20.04 + ROS Foxy) and got the same error in both cases!
"https://github.com/stereolabs/zed-ros2-wrapper/blob/master/docker/Dockerfile.u22-cu117-humble-release"

[zed_wrapper-2] [WARN] [1677091650.892548286] [zed2.zed_node]: Error opening camera: NO GPU DETECTED
[zed_wrapper-2] [INFO] [1677091650.892599583] [zed2.zed_node]: Please verify the USB3 connection

nvidia-smi works and lists my two GPUs!

Myzhar, 2023-02-22:
What is the model of the NVIDIA GPU card installed on the host?

Myzhar, 2023-02-22:
The NVIDIA driver version seems old, have you tried upgrading it?

Myzhar, 2023-02-23:
Let me know if by updating the driver you fix the issue.
I have also a concern regarding the double GPU configuration. Normally the ZED SDK takes the first available, but maybe you need to manually force it by setting the `general.gpu_id` parameter to 0 or 1: 
https://github.com/stereolabs/zed-ros2-wrapper/blob/master/zed_wrapper/config/common.yaml#L22

lucyannofrota, 2023-02-23:
I was not able to install the new drivers! 470 is the last NVIDIA driver that supports GTX780, consequently, the last cuda available is 11.4.

I've tested the container with `NVIDIA_VISIBLE_DEVICES=0` as well as setting `general.gpu_id = 0` or `general.gpu_id = 1` and I got the same error.

I think I just have found the source of the issue. The graphics cards + drivers + Nvidia-container-runtime are working as intended! But apparently exists an incompatibility of ZED SDK 3.8 with Maxwell and Kepler GPUs. This information is listed as a known issue ("Maxwell and Kepler GPUs (compute capabilities 35, 50 and 52) are not supported using CUDA 11.X, only with CUDA 10.2") in (https://www.stereolabs.com/developers/release/).

Considering this. I think I have two options:
1. Try ZED SDK 3.6
2. Downgrade to CUDA 10.2

What do you recommend?
I'm inclined to try the first option "ZED SDK 3.6". There's a major loss using ZED SDK 3.6 instead of ZED SDK 3.8? I'm working with ROS Foxy and I have two setups. One desktop for development (Ubuntu 20.04 + 2xGTX780) and a Jetson AGX Xavier with JetPack 5.1 (L4T R35.2.1) for deployment.

Myzhar, 2023-02-23:
You can get the wrapper for SDK v3.6 in the releases

lucyannofrota, 2023-02-23:
The SDK v3.6 didn't work as well!

Myzhar, 2023-02-24:
I expected that, in my opinion your problem is related to an incompatibility issue between the NVIDIA docker and the GPU driver.

What is the result of this command?
`sudo docker run --rm --runtime=nvidia --gpus all nvidia/cuda:11.7.0-devel-ubuntu20.04 nvidia-smi`

Are you able to run one of the CUDA examples inside a Docker container?

lucyannofrota, 2023-02-24:
Yes. I can run this command! All the previous `nvidia-smi` provided before were from inside containers.

### `sudo docker run --rm --runtime=nvidia --gpus all nvidia/cuda:11.7.0-devel-ubuntu20.04 nvidia-smi` output:

<pre><font color="#4E9A06"><b>lucyanno@ISR-Desktop</b></font>:<font color="#3465A4"><b>~/Semantic-Mapping-ROS2-Containers</b></font>$ sudo docker run --rm --runtime=nvidia --gpus all nvidia/cuda:11.7.0-devel-ubuntu20.04 nvidia-smi

==========
== CUDA ==
==========

CUDA Version 11.7.0

Container image Copyright (c) 2016-2022, NVIDIA CORPORATION &amp; AFFILIATES. All rights reserved.

This container image and its contents are governed by the NVIDIA Deep Learning Container License.
By pulling and using the container, you accept the terms and conditions of this license:
https://developer.nvidia.com/ngc/nvidia-deep-learning-container-license

A copy of this license is made available in this container at /NGC-DL-CONTAINER-LICENSE for your convenience.

*************************
** DEPRECATION NOTICE! **
*************************
THIS IMAGE IS DEPRECATED and is scheduled for DELETION.
    https://gitlab.com/nvidia/container-images/cuda/blob/master/doc/support-policy.md

Fri Feb 24 10:28:41 2023       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 470.161.03   Driver Version: 470.161.03   CUDA Version: 11.7     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:01:00.0 N/A |                  N/A |
| 37%   29C    P8    N/A /  N/A |    503MiB /  3021MiB |     N/A      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
|   1  NVIDIA GeForce ...  Off  | 00000000:02:00.0 N/A |                  N/A |
| 37%   24C    P8    N/A /  N/A |     12MiB /  3022MiB |     N/A      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|  No running processes found                                                 |
+-----------------------------------------------------------------------------+
</pre>

lucyannofrota, 2023-02-24:
I did tryied to run the CUDA examples at "https://github.com/NVIDIA/cuda-samples/tree/v11.6".

I was able to compile the samples. But I was not able to execute than.

lucyannofrota, 2023-02-24:
I did tryied to run the CUDA examples at "https://github.com/NVIDIA/cuda-samples/tree/v11.6".

I managed to compile the samples. But I couldn't execute than ("No device detected"). So the problem is related to the NVIDIA runtime!

Thank you verry mutch for your help @Myzhar but after that I don't think this problem is related with the ZED SDK at all.
It should be a CUDA + NVIDIA driver compatibility issue with this specific achitecture. So the solution should be downgrade to CUDA 10.2 but it isn't supported by ubuntu 20.04.

lucyannofrota, 2023-03-01:
I ended up switching my graphics card and it worked perfectly with the GTX1060. I think we can close this topic knowing that the problem is related to an incompatibility of CUDA 11.4 with Maxwell and Keppler cards. Thanks for your help @Myzhar.

closed 2023-03-01 by lucyannofrota