stereolabs/zed-ros2-wrapper #137: Record .svo file and run object detection at the same time

JWaynee, 2023-02-27:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

It would be great if the wrapper could record a .svo file and run object detection in parallel. 

### Use-Case

Doing some test outdoors with active object detection.
Afterwards rerun the ros2 wrapper in the lab with different parameters to see if different parameter set would have led to a better result.

Myzhar, 2023-02-27:
Hi @JWaynee 
this is already available. You must simply call the services to enable them.
Object Detection can start immediately if you set it to true in `common.yaml` without using the service call:
https://www.stereolabs.com/docs/ros2/zed-node/#services

JWaynee, 2023-02-27:
Hey @Myzhar,
thanks for the quick response. Sorry for missing this info in the ReadMe. Can you give me some additional information to the parameters of the service call:

- bitrate, which bitrate is this referring to.  I assume there is no data transmission but it's directly written on the hard drive
- compression_mode, do the modes correspond to the modes listed here: https://www.stereolabs.com/docs/video/recording/ and does 0 correspond to the first mode the list?
- input_transcode, what does this parameter do

Thanks
Jeremy

JWaynee, 2023-02-28:
Thanks again for the quick response!

closed 2023-02-28 by JWaynee