stereolabs/zed-ros2-wrapper #138: Zed ros Wrapper for Ros2 galactic

jheanelestrada, 2023-03-01:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

Is there any way to install zed ros wrapper in Ros2 Galactic?

### Use-Case

I am running an Ubuntu 20.04 and ros2 galactic

Myzhar, 2023-03-01:
Hi @jheanelestrada 
We support only LTS versions, hence Foxy (close to EOL) and Humble.
You can try to install the Wrapper also with Galactic, I know that someone did it successfully, but we do not provide support for it.

jheanelestrada, 2023-03-01:
Did you know how he do it? Can you send me the link to test?

Myzhar, 2023-03-01:
You can search in the closed issues. There was a discussion about that.

JiaqiangZhang, 2024-04-08:
Hi, @jheanelestrada ,
I tried install ros2 wrapper on galactic, but I didn't succeed. I ended up installing it on foxy. Did you find a way to install the wrapper on galactic?

closed 2024-04-08 by Myzhar

jheanel, 2024-04-11:
Hi @JiaqiangZhang I had some revisions in the src file where the error shows not supported version. You just need to uncomment those include files and retain the last one.

JiaqiangZhang, 2024-04-11:
>  I had some revisions in the src file where the error shows not supported version. You just need to comment out those include files and retain the last one.

Hi, @jheanel . Thanks for your reply, I successfully built ros2 wrapper on Galactic by adding several lines on the src files. For others who also want to build the wrapper on Galactic, you can follow commit [604a06f Add ROS2 Iron support](https://github.com/stereolabs/zed-ros2-wrapper/commit/604a06f53de2ad530b280e51c41ee6ea9ce36800), and add Galactic (Copy the FOXY part, and change it to Galactic).

GuangyiWilliamLiu, 2025-02-10:
> Hi, [@jheanel](https://github.com/jheanel) . Thanks for your reply, I successfully built ros2 wrapper on Galactic by adding several lines on the src files. For others who also want to build the [...]

Also, search FOUND_FOXY and then replace it with FOUND_GALACTIC