stereolabs/zed-ros2-wrapper #143: colcon build not successful Ubuntu 20.04 and ZED 2i

guidout, 2023-03-31:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I downloaded the latest release and this is the error I get when building with colcon

`Starting >>> zed_components
--- stderr: zed_components                         
CMake Warning at CMakeLists.txt:110 (find_package):
  By not providing "Findzed_interfaces.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "zed_interfaces", but CMake did not find one.

  Could not find a package configuration file provided by "zed_interfaces"
  with any of the following names:

    zed_interfacesConfig.cmake
    zed_interfaces-config.cmake

  Add the installation prefix of "zed_interfaces" to CMAKE_PREFIX_PATH or set
  "zed_interfaces_DIR" to a directory containing one of the above files.  If
  "zed_interfaces" provides a separate development package or SDK, be sure it
  has been installed.

CMake Error at /opt/ros/foxy/share/ament_cmake_target_dependencies/cmake/ament_target_dependencies.cmake:77 (message):
  ament_target_dependencies() the passed package name 'zed_interfaces' was
  not found before
Call Stack (most recent call first):
  CMakeLists.txt:197 (ament_target_dependencies)

---
Failed   <<< zed_components [1.90s, exited with code 1]
                                
Summary: 0 packages finished [2.02s]
  1 package failed: zed_components
  1 package had stderr output: zed_components
  2 packages not processed
`

### Steps to Reproduce

1. mkdir -p zed_test/src
2. cd zed_test/src
3. curl -sSL  https://github.com/stereolabs/zed-ros2-wrapper/archive/refs/tags/foxy-humble-v3.8.2.zip --output zed-ros2-wrapper.zip
4. unzip zed-ros2-wrapper.zip -d . && rm zed-ros2-wrapper.zip
5. mv zed-ros2-wrapper-foxy-humble-v3.8.2/ zed-ros2-wrapper/
6. cd ..
7. rosdep install --from-paths src --ignore-src -r -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
zed_components: Cannot locate rosdep definition for [zed_interfaces]
zed_ros2: Cannot locate rosdep definition for [zed_interfaces]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully

8. colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release --parallel-workers $(nproc)
Starting >>> zed_components
--- stderr: zed_components                         
CMake Warning at CMakeLists.txt:110 (find_package):
  By not providing "Findzed_interfaces.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "zed_interfaces", but CMake did not find one.

  Could not find a package configuration file provided by "zed_interfaces"
  with any of the following names:

    zed_interfacesConfig.cmake
    zed_interfaces-config.cmake

  Add the installation prefix of "zed_interfaces" to CMAKE_PREFIX_PATH or set
  "zed_interfaces_DIR" to a directory containing one of the above files.  If
  "zed_interfaces" provides a separate development package or SDK, be sure it
  has been installed.

CMake Error at /opt/ros/foxy/share/ament_cmake_target_dependencies/cmake/ament_target_dependencies.cmake:77 (message):
  ament_target_dependencies() the passed package name 'zed_interfaces' was
  not found before
Call Stack (most recent call first):
  CMakeLists.txt:197 (ament_target_dependencies)

---
Failed   <<< zed_components [1.94s, exited with code 1]

Summary: 0 packages finished [2.06s]
  1 package failed: zed_components
  1 package had stderr output: zed_components
  2 packages not processed

### Expected Result

colcon successfully compiles

### Actual Result

Starting >>> zed_components
--- stderr: zed_components                         
CMake Warning at CMakeLists.txt:110 (find_package):
  By not providing "Findzed_interfaces.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "zed_interfaces", but CMake did not find one.

  Could not find a package configuration file provided by "zed_interfaces"
  with any of the following names:

    zed_interfacesConfig.cmake
    zed_interfaces-config.cmake

  Add the installation prefix of "zed_interfaces" to CMAKE_PREFIX_PATH or set
  "zed_interfaces_DIR" to a directory containing one of the above files.  If
  "zed_interfaces" provides a separate development package or SDK, be sure it
  has been installed.

CMake Error at /opt/ros/foxy/share/ament_cmake_target_dependencies/cmake/ament_target_dependencies.cmake:77 (message):
  ament_target_dependencies() the passed package name 'zed_interfaces' was
  not found before
Call Stack (most recent call first):
  CMakeLists.txt:197 (ament_target_dependencies)

---
Failed   <<< zed_components [1.94s, exited with code 1]

Summary: 0 packages finished [2.06s]
  1 package failed: zed_components
  1 package had stderr output: zed_components
  2 packages not processed

### ZED Camera model

ZED2i

### Environment

```shell
Output of ZED_Diagnostic

ZED Diagnostic Tool

- Running ZED SDK Diagnostic : OK  
    ZED SDK version: 3.8.2
    CUDA version: V11.7.99
- Running Processor Diagnostic : OK  
    Processor:   12th Gen Intel(R) Core(TM) i9-12900H
    Motherboard:  03M8N5, Dell Inc.
- Running Graphics Card Diagnostic : OK  
    Graphics card:  NVIDIA RTX A2000 8GB Laptop GPU
- Running Camera Test Diagnostic : OK  
    Camera: ZED 2i
    Firmware: 1523
- Running USB Diagnostic : OK  
    USB Bandwidth:  OK
```

Myzhar, 2023-03-31:
Hi @guidout 
if you install the ZED ROS2 Wrapper from the zip file, you must also get the `zed-ros2-interfaces` package because GitHub does not add submodules in the release files:
https://github.com/stereolabs/zed-ros2-interfaces/releases/tag/foxy-humble-v3.8.2

guidout, 2023-03-31:
> Hi @guidout if you install the ZED ROS2 Wrapper from the zip file, you must also get the `zed-ros2-interfaces` package because GitHub does not add submodules in the release files: https://github.com/stereolabs/zed-ros2-interfaces/releases/tag/foxy-humble-v3.8.2

Oh ok. I'll try right now. Although I got different errors when just cloning the repo.
I'll report back soon, thank you

guidout, 2023-03-31:
> Hi @guidout if you install the ZED ROS2 Wrapper from the zip file, you must also get the `zed-ros2-interfaces` package because GitHub does not add submodules in the release files: https://github.com/stereolabs/zed-ros2-interfaces/releases/tag/foxy-humble-v3.8.2

I can confirm that adding the interfaces worked! Thank you!