stereolabs/zed-ros2-wrapper #145: Zed Wrapper randomly dies when set pose service is called

coalman321, 2023-04-19:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

On our robot, we launch the zed wrapper with all of our other sensors. We then launch our navigation with a pose converter that calls the Set Pose service in order to get a position and orientation relative to our own odometry frame. Occasionally, when we call the Set Pose service (1 in 10-12 times), the zed wrapper dies with error code -11. 

### Steps to Reproduce

1. Launch ZED ROS 2 wrapper
2. Call set pose 10-20 times until it dies

### Expected Result

Set pose works

### Actual Result

`[ERROR] [zed_wrapper-11]: process has died [pid 54022, exit code -11, cmd '/home/ros/colcon_deploy/install/zed_wrapper/lib/zed_wrapper/zed_wrapper --ros-args -r __node:=zed_node -r __ns:=/puddles/zed2i --params-file /home/ros/colcon_deploy/install/zed_wrapper/share/zed_wrapper/config/common.yaml --params-file /home/ros/colcon_deploy/install/zed_wrapper/share/zed_wrapper/config/zed2i.yaml --params-file /tmp/launch_params_77eslm20'].`

### ZED Camera model

ZED2i

### Environment

```shell
OS: Ubuntu 20.04
CPU: ARM
GPU: NVIDIA Jetson AGX Orin
ZED SDK: 3.8
Jetpack SDK: 5.1
ROS 2 Humble
```

coalman321, 2023-05-23:
This still has not been resolved.

IzzyCoder, 2023-05-29:
Same issue on Jetson NX with ZED SDK 3.8.2 and ZED Mini (also with SDK 4.0.1).
In our case the issue seem to happen only, when running the wrapper as a composable node (with component_container_mt). Which makes me think that this might a race condition issue.

markcfong561, 2023-06-29:
This has not been fixed yet

markcfong561, 2023-07-31:
@Myzhar could this be looked at?

andystoeggle, 2023-08-11:
Same issue here when calling the stop_svo_rec service on my Laptop (Thinkpad T490) as well as on the Jetson Xavier AGX:

```
[zed_wrapper-2] [INFO] [1691742368.031871536] [zed2i.zed_node]: SVO Recording started: 
[zed_wrapper-2] [INFO] [1691742368.031941371] [zed2i.zed_node]:  * Bitrate: 0
[zed_wrapper-2] [INFO] [1691742368.031966139] [zed2i.zed_node]:  * Compression: LOSSLESS
[zed_wrapper-2] [INFO] [1691742368.031984120] [zed2i.zed_node]:  * Framerate: 0
[zed_wrapper-2] [INFO] [1691742368.031998830] [zed2i.zed_node]:  * Input Transcode: FALSE
[zed_wrapper-2] [INFO] [1691742368.032015174] [zed2i.zed_node]:  * Filename: /media/andy/e44cb8f2-cfe8-4644-9e05-9b290aa899961/test_01.svo
[zed_wrapper-2] [ERROR] [1691742368.032048026] [zed2i.zed_node]: Error saving frame to SVO
[zed_wrapper-2] [ERROR] [1691742368.219456540] [zed2i.zed_node]: Error saving frame to SVO
[zed_wrapper-2] [INFO] [1691742376.345337048] [zed2i.zed_node]: ** Stop SVO Recording service called **
[zed_wrapper-2] [WARN] [1691742376.415200856] [zed2i.zed_node]: SVO Recording stopped
[ERROR] [zed_wrapper-2]: process has died [pid 7031, exit code -11, cmd '/home/andy/zhaw/ba/AutonomousSystem_ROS/install/zed_wrapper/lib/zed_wrapper/zed_wrapper --ros-args -r __node:=zed_node -r __ns:=/zed2i --params-file /home/andy/zhaw/ba/AutonomousSystem_ROS/install/zed_wrapper/share/zed_wrapper/config/common.yaml --params-file /home/andy/zhaw/ba/AutonomousSystem_ROS/install/zed_wrapper/share/zed_wrapper/config/zed2i.yaml --params-file /tmp/launch_params_cqwfebvn'].
```

Myzhar, 2023-08-11:
Please update to the latest ZED SDK.
The bug should have been fixed.

andystoeggle, 2023-08-14:
@Myzhar tried it and you're right, works fine with SDK 4.0.6 and the latest version of the ROS2 wrapper. Thank you so much for your fast reply and please excuse the circumstances.

closed 2023-09-19 by github-actions[bot]