stereolabs/zed-ros2-wrapper #146: colcon build error: sl/Fusion.hpp: No such file or directory

blackjacket996, 2023-04-21:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

After I successfully install the SDK in Jetson Xavier , I still met a erros “sl/Fusion.hpp: No such file or directory” when I build the zed_ros2_wrapper.

### Steps to Reproduce

1.install ZED_SDK
2.clone repo to ros2 workspace
3.run colcon build --symlink-install (just follow the steps)
...

### Expected Result

build successfully

### Actual Result

met a error when colcon build

### ZED Camera model

ZED2

### Environment

```shell
Jetson AGX Xavier
ros galactic
```

Myzhar, 2023-04-21:
Hi @blackjacket996 
what version of the ZED SDK did you install?
According to the error you installed v3.x, while the master branch of the ZED ROS2 Wrapper works with ZED SDK v4.x.
If you want to use the ZED SDK v3.8.2 you must use the latest release tag https://github.com/stereolabs/zed-ros2-wrapper/releases/tag/foxy-humble-v3.8.2

Please also note that ROS 2 Galactic is not officially supported and could not work.

blackjacket996, 2023-04-21:
> Hi @blackjacket996 what version of the ZED SDK did you install? According to the error you installed v3.x, while the master branch of the ZED ROS2 Wrapper works with ZED SDK v4.x. If you want to use [...]

Hi.it seems that I installed the lasest version V4.0. 
 
[image]

Myzhar, 2023-04-21:
Is it an upgrade or did you install from zero?

blackjacket996, 2023-04-21:
> Is it an upgrade or did you install from zero?
> 
> Please run `ZED_Explorer` and check for the ZED SDK version in the status bar

I have checked it and it is v4.0.1. I installed sdk 3.x before, but I have deleted the files.
[image]

Myzhar, 2023-04-21:
OK, so the problem is that you have not cleaned the ROS 2 workspace after the upgrade.
You must delete the `build` and `install` folders in the root of your workspace and build it again

blackjacket996, 2023-04-21:
> OK, so the problem is that you have not cleaned the ROS 2 workspace after the upgrade. You must delete the `build` and `install` folders in the root of your workspace and build it again

Thank you so much! It works well now~

closed 2023-04-21 by blackjacket996