stereolabs/zed-ros2-wrapper #148: Problem colcon build zed_wrapper

TeamRoboTo, 2023-04-29:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hello everyone, while building this package in the docker container dustynv/ros:foxy-pytorch-l4t-r32.7.1, i have this error:
[image].
can someone figure it out and help us? 
Thanks in advance for your patience.

### Steps to Reproduce

1. pull the dustynv/ros:foxy-pytorch-l4t-r32.7.1
2. install the ZED_SDK 3.8.2
3. pull and build zed_wrapper for ros2

### Expected Result

clear build of zed_wrapper package

### Actual Result

package installation fail

### ZED Camera model

ZED Mini

### Environment

```shell
NVIDIA NVIDIA Jetson Nano Developer Kit
 L4T 32.7.1 [ JetPack 4.6.1 ]
   Ubuntu 18.04.6 LTS
   Kernel Version: 4.9.299-tegra
 CUDA 10.2.300
   CUDA Architecture: 5.3
 OpenCV version: 4.1.1
   OpenCV Cuda: NO
 CUDNN: 8.2.1.32
 TensorRT: 8.2.1.8
 Vision Works: 1.6.0.501
 VPI: 1.2.3
 Vulcan: 1.2.70
ZED_SDK: 3.8.2
```

### Anything else?

nope

Myzhar, 2023-04-29:
Hi @TeamRoboTo 
You must satisfy all the dependencies to build the wrapper correctly.
Can you share your Dockerfile?

TeamRoboTo, 2023-05-04:
@Myzhar here you can find the file .sh we used to build the docker image 
https://github.com/dusty-nv/jetson-containers/blob/master/scripts/docker_build_ros.sh

TeamRoboTo, 2023-05-04:
I am confused, i'm using ZED SDK v3.8.2

TeamRoboTo, 2023-05-04:
ok thank you now we managed to build the package but when we try to tun th ecommand "$ ros2 launch zed_wrapper zedm.launch.py" I get the following error:
ros2 launch zed_wrapper zedm.launch.py
[INFO] [launch]: All log files can be found below /root/.ros/log/2023-05-04-18-43-33-682404-ubuntu-8124
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/install/lib/python3.8/site-packages/launch/launch_service.py:226> exception=SubstitutionFailure('executed command failed. Command: xacro /ros2_ws/install/zed_wrapper/share/zed_wrapper/urdf/zed_descr.urdf.xacro camera_name:=zedm camera_model:=zedm base_frame:=base_link gnss_frame:= cam_pos_x:=0.0 cam_pos_y:=0.0 cam_pos_z:=0.0 cam_roll:=0.0 cam_pitch:=0.0 cam_yaw:=0.0')>
Traceback (most recent call last):
  File "/opt/ros/foxy/install/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/install/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/install/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/install/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/install/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 4 more times]
  File "/opt/ros/foxy/install/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/install/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/install/lib/python3.8/site-packages/launch_ros/actions/node.py", line 444, in execute
    self._perform_substitutions(context)
  File "/opt/ros/foxy/install/lib/python3.8/site-packages/launch_ros/actions/node.py", line 399, in _perform_substitutions
    evaluated_parameters = evaluate_parameters(context, self.__parameters)
  File "/opt/ros/foxy/install/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 164, in evaluate_parameters
    output_params.append(evaluate_parameter_dict(context, param))
  File "/opt/ros/foxy/install/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 71, in evaluate_parameter_dict
    evaluated_value = perform_substitutions(context, list(value))
  File "/opt/ros/foxy/install/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in perform_substitutions
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/foxy/install/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in <listcomp>
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/foxy/install/lib/python3.8/site-packages/launch/launch_context.py", line 232, in perform_substitution
    return substitution.perform(self)
  File "/opt/ros/foxy/install/lib/python3.8/site-packages/launch/substitutions/command.py", line 116, in perform
    raise SubstitutionFailure(f'executed command failed. Command: {command_str}')
launch.substitutions.substitution_failure.SubstitutionFailure: executed command failed. Command: xacro /ros2_ws/install/zed_wrapper/share/zed_wrapper/urdf/zed_descr.urdf.xacro camera_name:=zedm camera_model:=zedm base_frame:=base_link gnss_frame:= cam_pos_x:=0.0 cam_pos_y:=0.0 cam_pos_z:=0.0 cam_roll:=0.0 cam_pitch:=0.0 cam_yaw:=0.0

Myzhar, 2023-05-04:
You miss the `xacro` dependency and it's surely not the only missing dependency, you must surely install also `diagnostic`

Myzhar, 2023-05-09:
You probably did not install `xacro` correctly

Icon45, 2023-05-11:
Hi, im getting the same error, but i  used the latest zed sdk version and cloned this repo.

closed 2023-06-17 by github-actions[bot]