stereolabs/zed-ros2-wrapper #151: How optimize AI model once

rafal-gorecki, 2023-05-10:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

Hi,
I create docker with zed-ros2-wrapper and i wonder how to disable, build docker, or add volumen to eliminate every time optimization. Do you have any solutions?

Myzhar, 2023-05-10:
Hi @rafal-gorecki 
the reply is in the README in the `docker` folder:
https://github.com/stereolabs/zed-ros2-wrapper/tree/master/docker#ai-module

Myzhar, 2023-05-11:
@rafal-gorecki please close the issue if solved

closed 2023-05-12 by rafal-gorecki