stereolabs/zed-ros2-wrapper #157: zed_wrapper package fail to build

nctran108, 2023-05-23:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hi I try to build the ros2 zed wrapper in mine ubuntu 22.04 but the wrapper package is fail to build.
[image]

### Steps to Reproduce

$ cd ~/ros2_ws/src/ #use your current ros 2 workspace folder
$ git clone --recursive https://github.com/stereolabs/zed-ros2-wrapper.git
$ cd ..
$ rosdep install --from-paths src --ignore-src -r -y
$ colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release
$ echo source $(pwd)/install/local_setup.bash >> ~/.bashrc
$ source ~/.bashrc

### Expected Result

the wrapper successful build

### Actual Result

zed_wrapper package fail to build.

### ZED Camera model

ZED2

### Environment

```shell
Ubuntu 22.04
GPU: NVIDIA GeForce RTX 4080
driver: NVIDIA 530.30.02
CUDA: 12.1
```

Myzhar, 2023-05-23:
@nctran108 installing the missing library should solve your problem:
`$ sudo apt install libturbojpeg libturbojpeg0-dev`

nctran108, 2023-05-23:
@Myzhar Thanks, it works.

closed 2023-06-28 by github-actions[bot]