stereolabs/zed-ros2-wrapper #162: Error of No GPU Compatible even after the reinstallation of Nvidia Driver

segapss, 2023-06-19:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hello, 
I have installed this ZED-ROS2-Wrapper for Stereo ZED Mini, but it still causes No GPU Compatible error as follows.
[image]
My laptop environment: Ubuntu 22.04, ROS2 Humble, Nvidia GTX 950m graphic card, Graphic Driver: Nvidia driver 530.41, CUDA version: 12.1.
Don't ask me for reinstallation of Graphic Driver. Because I've already done it. 
Thanks in advance.

### Steps to Reproduce

1. Install Nvidia Driver and CUDA installation.
2. Run ros2 launch zed_wrapper zedm.launch.py

### Expected Result

Successful ZED node operation

### Actual Result

[image]
This unpleasant output: No GPU Compatible.

### ZED Camera model

ZED Mini

### Environment

```shell
Ubuntu 22.04, ROS2 Humble, Nvidia GTX 950m graphic card, Graphic Driver: Nvidia driver 530.41, CUDA version: 12.1. ZED SDK: 4.0.2
```

Myzhar, 2023-06-19:
Hi @segapss 
This is not a bug, but a wrong configuration. 
The NVIDIA GTX 950m is based on the old Maxwell architecture, so you cannot use it with CUDA 12.x and CUDA 11.x, but only with CUDA 10.2, as explained in the ZED SDK download page.

closed 2023-06-19 by Myzhar

segapss, 2023-06-19:
@Myzhar  Thanks. But I remember that it has worked under Ubuntu 20.04, ROS2 Foxy with lower version of ZED SDK. Then what should I do to run ZED ROS2 Wrapper? Because CUDA 10.2 doesn't support Ubuntu 22.04, so I wonder which S/W should I change. Downgrade of CUDA, Ubuntu, or ROS2?

Myzhar, 2023-06-19:
It worked for previous versions of the ZED SDK, not v4.x.
You can downgrade OS if you want to use the latest SDK and latest wrapper, otherwise downgrade the SDK and the wrapper

segapss, 2023-06-19:
@Myzhar Thanks. Then version adjustment should be the only option.