stereolabs/zed-ros2-wrapper #165: /zedm/zed_node/depth/depth_registered turning black when using ROS2 Driver

Siming-He, 2023-07-09:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I'm using ROS2 driver for positional tracking. I observe that the /zedm/zed_node/depth/depth_registered image turns fully black sometimes and cause loss of track. The problem happens for all depth mode when using ros2. But it doesn't happen for ZED_DEPTH_VIEWER.

### Steps to Reproduce

1. Followed this https://github.com/stereolabs/zed-ros2-wrapper to setup the workspace
2. `ros2 launch zed_wrapper zedm.launch.py `
3. looking at the depth images from ` rviz2 zedm.rviz ` which is given in ros2 zed examples

### Expected Result

I expect to see clear depth values (different colors in the depth image) all the time.

### Actual Result

The depth images turn black sometimes and cause loss of tracking.

### ZED Camera model

ZED Mini

### Environment

```shell
OS: Ubuntu 22.04
CPU: 12th Gen Intel(R) Core(TM) i9-12900H
GPU: NVIDIA GeForce RTX 3050 Ti
SDK: ZED_SDK_Ubuntu22_cuda11.8_v4.0.5.zstd
ROS2 Humble Hawksbill
```

Myzhar, 2023-07-09:
Hi @Siming-He 
Please read this
https://support.stereolabs.com/hc/en-us/articles/5365701074967-Why-is-the-depth-map-so-dark-

Positional tracking behavior are not related to dark depth maps

Siming-He, 2023-07-09:
> Hi @Siming-He Please read this https://support.stereolabs.com/hc/en-us/articles/5365701074967-Why-is-the-depth-map-so-dark-
> 
> Positional tracking behavior are not related to dark depth maps

Thank you for the reply! So the link is talking about that the depth map is dark because object is too close which is indeed a visualization issue. But if you look at the video I attached, the position of my camera doesn't change to much, but the depth suddenly turned to black for some period. It shouldn't be the issue of depth image being too dark right?

Myzhar, 2023-07-09:
That's a normal behavior of rviz when it's set to normalize the depth map visualization according to the depth max value.
If you change the depth normalization values in order to match the expected max depth range you will notice that black depth maps disappear.

Myzhar, 2023-07-09:
Positional tracking is losing track because of your environment: dark room with high light sources (screen and lamp)

Siming-He, 2023-07-09:
> That's a normal behavior of rviz when it's set to normalize the depth map visualization according to the depth max value. If you change the depth normalization values in order to match the expected max depth range you will notice that black depth maps disappear.

Thank you for answering my questions. It makes sense to me now that the sudden change of "color" is due to normalization of depth. I have one last question: when I look into the topic `/zedm/zed_node/depth/depth_info,` the max depth is negative?
`header:
  stamp:
    sec: 1688761281
    nanosec: 161146658
  frame_id: zedm_left_camera_optical_frame
min_depth: 2.5231311321258545
max_depth: -1.0545660257339478
---
header:
  stamp:
    sec: 1688761281
    nanosec: 227656658
  frame_id: zedm_left_camera_optical_frame
min_depth: 2.5231311321258545
max_depth: -1.0545660257339478
---
header:
  stamp:
    sec: 1688761281
    nanosec: 294539658
  frame_id: zedm_left_camera_optical_frame
min_depth: 2.5231311321258545
max_depth: -1.0545660257339478
---
header:
  stamp:
    sec: 1688761281
    nanosec: 361149658
  frame_id: zedm_left_camera_optical_frame
min_depth: 2.5231311321258545
max_depth: -1.0545660257339478
`

Siming-He, 2023-07-17:
The issue is indeed normalization during visualization.

closed 2023-07-17 by Siming-He