stereolabs/zed-ros2-wrapper #168: Colcon build Error, Unable to detect `zed-config.cmake`

lonebots, 2023-07-26:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Recently I was trying to setup `zed-ros2-wrapper` locally. 
I am using Ubuntu 22.04, with ROS 2 Humble and trying to set ZED 4.0 with CUDA 11.8 for one of my projects

### Steps to Reproduce

1. cloned the `zed-ros2-wrapper` repo
2. install ZED SDK and with CUDA 
3. verified that `/usr/local/zed` contains the `zed-config.cmake` file
4. Tried doing a `colcon build` and the build system was unable detect `zed-config.cmake` 

### Expected Result

The build successfully completes

### Actual Result

The actual result is given below.

```bash
Starting >>> zed_components
--- stderr: zed_components                                                         
CMake Error at CMakeLists.txt:78 (find_package):
  By not providing "FindZED.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "ZED", but
  CMake did not find one.

  Could not find a package configuration file provided by "ZED" (requested
  version 3) with any of the following names:

    ZEDConfig.cmake
    zed-config.cmake

  Add the installation prefix of "ZED" to CMAKE_PREFIX_PATH or set "ZED_DIR"
  to a directory containing one of the above files.  If "ZED" provides a
  separate development package or SDK, be sure it has been installed.

---
Failed   <<< zed_components [0.06s, exited with code 1]

```

### ZED Camera model

ZED2

### Environment

```shell
OS : Ubuntu 22.04
ZED version : 4.0
CUDA : 11.8
Graphics Card : NVIDIA GeForce GTX 1050 Ti
```

### Anything else?

What I tried ?
1. I  tried to set the following environment variables in the `.bashrc` file 

```bash
#CUDA PATH
export CUDA_BIN_PATH=/usr/local/cuda-11.8
#ZED PATH
export ZED_DIR=/usr/local/zed
export ZED_INCLUDE_DIRS=/usr/local/zed/include
export ZED_LIBRARY_DIR=/usr/local/zed/lib

```

Patrick-AA, 2023-08-08:
Hello can you try to manually change the two occurrences of `find_package(ZED 3 REQUIRED)` to `find_package(ZED 4 REQUIRED)` ?
We will soon publish a fix for that.

lonebots, 2023-08-08:
Thank you for the comment ❤️ 
Actually in this case the issue was totally different from the version number, It was a simple permission issue and by default when individual user was installing zed on the common device, the installation got over written so only that particular user can exclusive access writes over it, 
The fix I could find was to switch the `owner:user-group` to something common, which everyone can access and also give owner level authority to all `user-group` users.