stereolabs/zed-ros2-wrapper #169: Cannot build the Zed Components package

sandipsharan, 2023-07-28:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hello, I am facing the same problem as #58 and I have installed ZED SDK in the computer and confirmed it with doing **_ls /usr/local/zed_** and I can see the zed-config.cmake file in there but even after setting the ZED_DIR to that path I am experiencing the same issue here. Can you help me with this? I have also attached a screenshot the error I'm getting. 
[image]

### Steps to Reproduce

1. Installed zed-ros2-wrapper, ZED SDK and the zed-ros2-examples package 
2. Sourced the workspace and tried to build the package using the command given in the readme of the repo
3. Ended up with this error
...

### Expected Result

Clean build output

### Actual Result

Couldn't find the zed-config.cmake file

### ZED Camera model

ZED2i

### Environment

```shell
OS: Ubuntu 22.04
ZED SDK version: v4.0.1
ROS Humble
```

pkraison, 2023-07-31:
Make sure you have installed the correct version of Zed SDK. it should be compatible with the ros driver branch.

Myzhar, 2023-08-04:
Can you try to manually change the two occurrences of `find_package(ZED 3 REQUIRED` to `find_package(ZED 4 REQUIRED`?

We will release a patch for this

closed 2023-08-11 by sandipsharan