stereolabs/zed-ros2-wrapper #170: ZEDX running using docker stereolabs/zedbot:zed-ros2-wrapper_l4t35_1_humble_ cant connec to cameras running on jetson agx orin

DailyNir, 2023-07-31:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

i am running 2 zedx cameras connected with the propritery connection of zed to the jetson agx orin, running ubuntu 20.04 jetpack 5.1.1 cuda 11.4 . 
trying to run the docker image stereolabs/zedbot:zed-ros2-wrapper_l4t35_1_humble_ ,  pre built, and i get the following errors 
[zed_wrapper - 2]
[INFO]
[1690793838.710382443]
[zedx.zed_node]: ZED SDK Version: 4.0.1 - Build 70396_69497376
[zed_wrapper- 2]
[INFO] [1690793838.713748749] [zedx.zed_node]:
* CAMERA OPENING *
[zed_wrapper-21
[ZED][INFO] Logging level INFO
[zed_wrapper- 2]
[ZED SD][Input] Using GMSL input... Switched to HD1200
[zed_wrapper- 2] [ZED][INFO] [Init]
Depth mode: NEURAL
[zed_wrapper -2] [ZED][ERROR] [ZED] Cannot initialize the camera.
Try another resolution
[ERROR] [zed_wrapper-2]: process has died [pid 7283, exit code -11, cmd /root/ros2_ws/install/zed_wrapper/Lib/zed_wrapper/zed_wrapper-ros-args ~г _node:=zed_node - r _ns:=/zedx - -params-file /root/ro 52_ws/install/zed_wrapper/share/zed_wrapper/config/common.yaml - -params-file /root/ros2_ws/tnstall/zed_wrapper/share/zed_wrapper/config/zed.yaml - params-file /tmp/launch_params_00c6w1tx'].
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[robot_state_publisher-1] [INFO] [1690793844.352043189] [rclcpp]: signal_handler(signum=2)
[INFO] [robot_state_publisher-1]: process has finished cleanly [pid 7281] root@a80bd4ccicab:~/ros2_ws/install/zed_wrapper/share/zed_wrapper/config

### Steps to Reproduce

i followed the steps in this link https://github.com/stereolabs/zed-ros2-wrapper/tree/master/docker after doing:docker pull stereolabs/zedbot:zed-ros2-wrapper_l4t35_1_humble_ i just run docker run --runtime nvidia -it --privileged --ipc=host --pid=host -e DISPLAY \
  -v /dev/shm:/dev/shm \
  -v /tmp/:/tmp/ -v /var/nvidia/nvcam/settings/:/var/nvidia/nvcam/settings/ \
  -v /tmp/zed_ai/:/usr/local/zed/resources/ \
stereolabs/zedbot:zed-ros2-wrapper_l4t35_1_humble_

### Expected Result

opening cameras and seeing an live video feed 

### Actual Result

errors, trying to tell me to change resolution, tried that also, didnt work 

### ZED Camera model

ZED

### Environment

```shell
OS:Ubuntu 20.04
GPU - jetson agx orin
JETPACK 5.1.1 +SDK for Jetpack 5.1.1
Cuda 11.4
```

Myzhar, 2023-07-31:
Hi @DailyNir 
is it working outside Docker?

DailyNir, 2023-07-31:
yes it works well with the api, i haven't checked with ros2 wrapper, the main goal is to run zedx cameras with the ros2 wrapper via docker image you provided

Myzhar, 2023-07-31:
Working on the fix (upgrade ZED SDK from v4.0.1 to v4.0.5)

nirlevymenteebot, 2023-07-31:
So I should build my own image with the 4.0.5 SDK?

Myzhar, 2023-07-31:
No, it will be automatically pushed to our docker hub as soon as the CI process is completed. 
A few hours I guess

Myzhar, 2023-07-31:
@nirlevymenteebot you can try now with the new Docker images
https://hub.docker.com/r/stereolabs/zedbot/tags

DailyNir, 2023-08-01:
hi, run the same command, and did ros2 launch as documented and got the following errors: 
zed_wrapper-2] [INFO] [1690870722.696182584] [zedx.zed_node]: *** CAMERA OPENING ***
[zed_wrapper-2] [ZED][INFO] Logging level INFO
[zed_wrapper-2] [ZED][INFO] [Init]  Depth mode: NEURAL
[zed_wrapper-2] [ZED-X][Warning] Failed to open zed_x service
[zed_wrapper-2] [ZED][ERROR] [ZED] Cannot initialize the camera. Try another resolution
[ERROR] [zed_wrapper-2]: process has died [pid 7426, exit code -11, cmd '/root/ros2_ws/install/zed_wrapper/lib/zed_wrapper/zed_wrapper --ros-args -r __node:=zed_node -r __ns:=/zedx --params-file /root/ros2_ws/install/zed_wrapper/share/zed_wrapper/config/common.yaml --params-file /root/ros2_ws/install/zed_wrapper/share/zed_wrapper/config/zedx.yaml --params-file /tmp/launch_params_aocearvp'].

Myzhar, 2023-08-02:
Still debugging, sorry. We are facing the same issue and we cannot understand the cause.
We'll update the issue as soon as we solve the problem.

DailyNir, 2023-08-17:
i saw you updated to 4.0.6 but still not working, update us when you can

DailyNir, 2023-09-17:
Still waiting on you, let us know what is up with the docker image

Myzhar, 2023-09-17:
We tested the docker image and it works as expected.
Did you install the ZED X driver in the host (outside docker)?

DailyNir, 2023-09-17:
Yes. Should I update to 4.0.6 or it is for 4.0.5?

Myzhar, 2023-09-17:
You should use the latest Driver v0.5.4 and update the Docker image to the latest version that is updated to the SDK v4.0.6

DailyNir, 2023-09-20:
just checked it again using same versions and updated docker and getting same error .

Myzhar, 2023-09-20:
Please install the ZED SDK in the host (no Docker), run the command `sudo ZED_Diagnostic --dmesg` and send the report file that it generates to support@stereolabs.com

DailyNir, 2023-09-20:
sent it

DailyNir, 2023-09-20:
here is the file's content: 
####################################
############ DMESG LOG #############
####################################

[   18.074794] zedx 30-0018: Driver Version : v0.5.4
[   18.074801] zedx 30-0018: Probing v4l2 sensor.
[   18.078996] zedx 30-0018: tegracam sensor driver:video0_v2.0.6
[   25.149221] zedx 30-0018: zedx_links_check: 2 link(s) detected
[   30.956284] zedx 30-0018: Multiple ZED-X camera detect success
[   31.184917] zedx 30-0018: zedx_set_frame_rate: Unsupported value
[   31.512660] tegra-camrtc-capture-vi tegra-capture-vi: subdev zedx 30-0018 bound
[   31.516085] zedx 30-0018: Detected ZED-X sensor
[   31.516593] zedx 30-0010: Driver Version : v0.5.4
[   31.516596] zedx 30-0010: Probing v4l2 sensor.
[   31.516759] zedx 30-0010: tegracam sensor driver:video1_v2.0.6
[   31.523470] zedx 30-0018: zedx_open: Accessing the camera
[   31.625438] zedx 30-0010: zedx_links_check: 2 link(s) detected
[   31.850917] zedx 30-0010: zedx_set_frame_rate: Unsupported value
[   32.180665] tegra-camrtc-capture-vi tegra-capture-vi: subdev zedx 30-0010 bound
[   32.181841] zedx 30-0010: Detected ZED-X sensor
[   32.182013] zedx 31-0018: Driver Version : v0.5.4
[   32.182015] zedx 31-0018: Probing v4l2 sensor.
[   32.182161] zedx 31-0018: tegracam sensor driver:video2_v2.0.6
[   32.185050] zedx 30-0010: zedx_open: Accessing the camera
[   32.289401] zedx 31-0018: zedx_links_check: 2 link(s) detected
[   32.515340] zedx 31-0018: zedx_set_frame_rate: Unsupported value
[   32.845027] tegra-camrtc-capture-vi tegra-capture-vi: subdev zedx 31-0018 bound
[   32.846351] zedx 31-0018: Detected ZED-X sensor
[   32.846575] zedx 31-0010: Driver Version : v0.5.4
[   32.846579] zedx 31-0010: Probing v4l2 sensor.
[   32.846833] zedx 31-0010: tegracam sensor driver:video3_v2.0.6
[   32.850371] zedx 31-0018: zedx_open: Accessing the camera
[   32.953489] zedx 31-0010: zedx_links_check: 2 link(s) detected
[   33.178826] zedx 31-0010: zedx_set_frame_rate: Unsupported value
[   33.510093] tegra-camrtc-capture-vi tegra-capture-vi: subdev zedx 31-0010 bound
[   33.510945] zedx 31-0010: Detected ZED-X sensor
[   33.514692] zedx 31-0010: zedx_open: Accessing the camera
[  124.838806] zedx 30-0010: zedx_open: Accessing the camera
[  124.839130] zedx 30-0018: zedx_open: Accessing the camera
[  124.839242] zedx 31-0010: zedx_open: Accessing the camera
[  124.839464] zedx 31-0018: zedx_open: Accessing the camera
[ 1051.423671] zedx 30-0010: zedx_open: Accessing the camera
[ 1051.423747] zedx 30-0018: zedx_open: Accessing the camera
[ 1051.423817] zedx 31-0010: zedx_open: Accessing the camera
[ 1051.423884] zedx 31-0018: zedx_open: Accessing the camera
[ 1228.116565] zedx 30-0010: zedx_open: Accessing the camera
[ 1228.116641] zedx 30-0018: zedx_open: Accessing the camera
[ 1228.116707] zedx 31-0010: zedx_open: Accessing the camera
[ 1228.116771] zedx 31-0018: zedx_open: Accessing the camera

####################################
############ Driver list ###########
####################################

/usr/lib/modules/5.10.104-tegra/kernel/drivers/stereolabs/max96712/max96712.ko==> exists
/usr/lib/modules/5.10.104-tegra/kernel/drivers/stereolabs/zedx/sl_zedx.ko ==> exists

################################################
############ ZEDX Daemon Status LS #############
################################################

● zed_x_daemon.service - ZED-X Daemon service
     Loaded: loaded (/etc/systemd/system/zed_x_daemon.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2023-09-20 10:51:33 IDT; 24min ago
   Main PID: 1447 (ZEDX_Daemon)
      Tasks: 3 (limit: 74725)
     Memory: 9.5M
     CGroup: /system.slice/zed_x_daemon.service
             └─1447 /usr/sbin/ZEDX_Daemon
[... 43 lines cut ...]

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: UU -- -- -- -- -- -- --                        

################################################
############ I2C Tracz /dev/i2C-33 #############
################################################

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: UU -- -- -- -- -- -- --                        
#########################

Myzhar, 2023-09-20:
The cameras are correctly detected. Can you open them with ZED Explorer and ZED Depth Viewer?

DailyNir, 2023-09-20:
Yes as I said everything works fine, the ros2 wrapper the zed explorer etc, the dicker is the only one saying it can't start zedx service.

DailyNir, 2023-09-22:
this was the issue from the beginning , it always says : cant start the zedx service try different resolution. let us know when the docker image is fixed and we can run it smoothly .

Myzhar, 2023-09-23:
Hi @DailyNir 
Did you install the ZED X driver on the Jetson, "outside Docker"?

DailyNir, 2023-09-23:
Hi, let me describe the setup .
Jetson AGX Orin. All the updated drivers jet pack 5.1.1 and zed x 0.5.4 drivers for jet pack 5.1.1.
Running docker doing docker pull from the docker hub of the original image there.

Running docker with the command line from GitHub especially for zed x .

Runs ok.
Then doing
ros2 launch zedx ……..
Per you line for running zed x in the GitHub page.

Getting can’t start zedx service … etc etc ..

DailyNir, 2023-10-09:
Hi, are there any updates on the subject? We have been waiting for couple of months for this issue . 

Let us know.

closed 2023-11-15 by github-actions[bot]

bmegli, 2024-01-29:
@DailyNir, potential cause:
- https://github.com/stereolabs/zed-docker/issues/52

ZED-SDK downloads calibration files if missing.

If your system is offline (no internet connection), and you are running from docker you don't have those calibration files pre-downloaded. If that is your problem the fix is to map calibration file paths to persistent storage on host and run the pipeline at least once online.