stereolabs/zed-ros2-wrapper #171: ZED Wrapper Build Error

alberthli, 2023-08-01:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

When building the ROS2 wrapper in a Docker container, I get a build error. 

### Steps to Reproduce

Build using the following Dockerfile:
```
FROM stereolabs/zed:4.0-devel-cuda12.1-ubuntu20.04 as zed
SHELL ["/bin/bash", "-c"]

# installing essential dependencies
RUN apt-get update -y --allow-insecure-repositories && apt-get install curl -y
RUN curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key -o /usr/share/keyrings/ros-archive-keyring.gpg
RUN echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(. /etc/os-release && echo $UBUNTU_CODENAME) main" | sudo tee /etc/apt/sources.list.d/ros2.list > /dev/null
RUN apt-get update -y
RUN apt-get install -y \
    build-essential \
    cmake \
    git \
    python-is-python3 \
    python3-pip \
    nano \
    mesa-utils \
    ros-foxy-ros-base \
    python3-argcomplete \
    ros-dev-tools

# franka_ros2 dependencies + others
RUN apt-get install --allow-unauthenticated -y \
    ros-foxy-xacro \
    ros-foxy-angles \
    ros-foxy-rviz-visual-tools \
    ros-foxy-joint-state-publisher \
    ros-foxy-joint-state-publisher-gui \
    ros-foxy-ament-cmake-clang-format \
    ros-foxy-rosidl-generator-cpp \
    ros-foxy-rosidl-default-generators \
    ros-foxy-image-transport \
    ros-foxy-ament-lint-auto \
    ros-foxy-ament-cmake-copyright \
    ros-foxy-ament-cmake-cppcheck \
    ros-foxy-ament-cmake-lint-cmake \
    ros-foxy-ament-cmake-pep257 \
    ros-foxy-ament-cmake-uncrustify \
    ros-foxy-ament-cmake-xmllint \
    ros-foxy-nmea-msgs \
    ros-foxy-graph-msgs \
    ros-foxy-diagnostic-updater \
    ros-foxy-geographic-msgs \
    ros-foxy-robot-localization \
    libopencv-dev \
    python3-opencv

# clone zed_ros2_wrapper and build
WORKDIR /root/dev_ws/src
RUN git clone --recursive https://github.com/stereolabs/zed-ros2-wrapper.git
WORKDIR /root/dev_ws
RUN rosdep init
RUN rosdep update
RUN rosdep install --from-paths src --ignore-src -r -y
RUN source /opt/ros/foxy/setup.bash && colcon build --symlink-install --event-handlers console_direct+ --base-paths src \
  --cmake-args ' -DCMAKE_BUILD_TYPE=Release' \
  ' -DCMAKE_LIBRARY_PATH=/usr/local/cuda/lib64/stubs' \
  ' -DCMAKE_CXX_FLAGS="-Wl,--allow-shlib-undefined"' --parallel-workers $(nproc)
```

### Expected Result

The build should complete without error.

### Actual Result

The build fails. This is not the same error as #169. Log attached here:
[build.log](https://github.com/stereolabs/zed-ros2-wrapper/files/12223886/build.log)

### ZED Camera model

ZED2i

### Environment

```shell
OS: Ubuntu 22.04 (Ubuntu 20.04 in container)
CPU: AMD Ryzen Threadripper PRO 5995WX 64-Cores
GPU: 2x RTX 4090s
Zed SDK version: 4.0.5 (base image specified in the Dockerfile)
ROS Distro: ROS2 Foxy
```

### Anything else?

Build was working with this Dockerfile somewhat recently (due to cached Docker builds, I'm not sure whether this passed on the 4.0.4 SDK version vs. 4.0.5).

Myzhar, 2023-08-01:
Hi @alberthli 
local compile passed and CI passed.
What version of GCC do you use in the container?

alberthli, 2023-08-01:
The output of `gcc --version` is `gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0`.

Myzhar, 2023-08-01:
I identified the possible cause of the error.
Publishing the fix in a few minutes

PR #172 (2023-08-01): Fix mAiInstanceID atomix init
Fixes #171
closed 2023-08-01 by Myzhar