stereolabs/zed-ros2-wrapper #173: running locally foxy and zed ros2 wrapper on jetson agx orin sdk 4.0.5 Error

DailyNir, 2023-08-02:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

running zed x camera first time in zed ros2 wrapper and jetpack 5.1.1 jetson agx orin sdk 4.0.5(latest i think) getting this error after fresh install of ros2 foxy : 
ros2 launch zed_wrapper zedx.launch.py
[INFO] [launch]: All log files can be found below /home/jetson_dev/.ros/log/2023-08-02-15-36-06-269495-linux-149755
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:226> exception=SubstitutionFailure('executed command failed. Command: xacro /home/jetson_dev/ros2_ws/install/zed_wrapper/share/zed_wrapper/urdf/zed_descr.urdf.xacro camera_name:=zedx camera_model:=zedx base_frame:=base_link gnss_frame:= cam_pos_x:=0.0 cam_pos_y:=0.0 cam_pos_z:=0.0 cam_roll:=0.0 cam_pitch:=0.0 cam_yaw:=0.0')>
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 4 more times]
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 444, in execute
    self._perform_substitutions(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 399, in _perform_substitutions
    evaluated_parameters = evaluate_parameters(context, self.__parameters)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 164, in evaluate_parameters
    output_params.append(evaluate_parameter_dict(context, param))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 71, in evaluate_parameter_dict
    evaluated_value = perform_substitutions(context, list(value))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in perform_substitutions
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in <listcomp>
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_context.py", line 232, in perform_substitution
    return substitution.perform(self)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/substitutions/command.py", line 116, in perform
    raise SubstitutionFailure(f'executed command failed. Command: {command_str}')
launch.substitutions.substitution_failure.SubstitutionFailure: executed command failed. Command: xacro /home/jetson_dev/ros2_ws/install/zed_wrapper/share/zed_wrapper/urdf/zed_descr.urdf.xacro camera_name:=zedx camera_model:=zedx base_frame:=base_link gnss_frame:= cam_pos_x:=0.0 cam_pos_y:=0.0 cam_pos_z:=0.0 cam_roll:=0.0 cam_pitch:=0.0 cam_yaw:=0.0

### Steps to Reproduce

i followed all the steps in the git repo for ros2 foxy 
then run : 
ros2 launch zed_wrapper zedx.launch.py

### Expected Result

seeing the camera

### Actual Result

- error : Task exception was never retrieved

### ZED Camera model

ZED Mini

### Environment

```shell
OS: Ubuntu 20.04
Jetson AGX orin
SDK 4.0.5(installed last week from the website) 
ZED X
```

### Anything else?

having a lot of issues since upgrading to sdk 4.0.5

Myzhar, 2023-08-02:
@DailyNir did you clean the workspace (delete `log`, `install`, and `build` folder) after the ZED SDK upgrade and before building?

DailyNir, 2023-08-02:
yes, the upgrade was last week and just today i downloaded the zed2 ros wrapper locally, if i can get sdk 4.0.1 to try it would be great it worked before with older sdk locally

Myzhar, 2023-08-02:
There is no reason why it should not work with ZED SDK v4.0.5 and it was working with SDK v4.0.1.
This is something more related with Foxy that cannot correctly handle empty strings as launch parameters.

Can you try this?
`ros2 launch zed_wrapper zedx.launch.py gnss_frame:=''`
or
`ros2 launch zed_wrapper zedx.launch.py gnss_frame:=""`
or 
`ros2 launch zed_wrapper zedx.launch.py gnss_frame:='dummy'`

DailyNir, 2023-08-02:
(base) jetson_dev@linux:~/ros2_ws$ ros2 launch zed_wrapper zedx.launch.py gnss_frame:=''
malformed launch argument 'gnss_frame:=', expected format '<name>:=<value>'
(base) jetson_dev@linux:~/ros2_ws$ ros2 launch zed_wrapper zedx.launch.py gnss_frame:=""
malformed launch argument 'gnss_frame:=', expected format '<name>:=<value>'
(base) jetson_dev@linux:~/ros2_ws$ ros2 launch zed_wrapper zedx.launch.py gnss_frame:=
malformed launch argument 'gnss_frame:=', expected format '<name>:=<value>'
(base) jetson_dev@linux:~/ros2_ws$ ros2 launch zed_wrapper zedx.launch.py gnss_frame:='dummy'
[INFO] [launch]: All log files can be found below /home/jetson_dev/.ros/log/2023-08-02-16-44-53-289502-linux-17678
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:226> exception=SubstitutionFailure('executed command failed. Command: xacro /home/jetson_dev/ros2_ws/install/zed_wrapper/share/zed_wrapper/urdf/zed_descr.urdf.xacro camera_name:=zedx camera_model:=zedx base_frame:=base_link gnss_frame:=dummy cam_pos_x:=0.0 cam_pos_y:=0.0 cam_pos_z:=0.0 cam_roll:=0.0 cam_pitch:=0.0 cam_yaw:=0.0')>
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 4 more times]
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 444, in execute
    self._perform_substitutions(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 399, in _perform_substitutions
    evaluated_parameters = evaluate_parameters(context, self.__parameters)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 164, in evaluate_parameters
    output_params.append(evaluate_parameter_dict(context, param))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 71, in evaluate_parameter_dict
    evaluated_value = perform_substitutions(context, list(value))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in perform_substitutions
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in <listcomp>
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_context.py", line 232, in perform_substitution
    return substitution.perform(self)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/substitutions/command.py", line 116, in perform
    raise SubstitutionFailure(f'executed command failed. Command: {command_str}')
launch.substitutions.substitution_failure.SubstitutionFailure: executed command failed. Command: xacro /home/jetson_dev/ros2_ws/install/zed_wrapper/share/zed_wrapper/urdf/zed_descr.urdf.xacro camera_name:=zedx camera_model:=zedx base_frame:=base_link gnss_frame:=dummy cam_pos_x:=0.0 cam_pos_y:=0.0 cam_pos_z:=0.0 cam_roll:=0.0 cam_pitch:=0.0 cam_yaw:=0.0

Myzhar, 2023-08-02:
I try to replicate the problem on my Jetson. Please wait for a reply, unless you find a solution.

Myzhar, 2023-08-02:
I tested your configuration on Jetson with JP5.1.1, ZED SDK v4.0.5, ZED X Driver v0.5.4, ROS 2 Foxy, and the ZED ROS 2 Wrapper from the `master` branch and I can run the nodes without problems.

Can you please verify that everything is updated to the latest available version?

DailyNir, 2023-08-03:
worked after i reinstalled everything, but mainly i had to deactivate the conda environment (base) and after that it run fine, why is that?(i should point out the conda files and environment are on a different nvme drive on the jetson and not on the system memory)

Myzhar, 2023-08-03:
Python and ROS must always be correctly configured. I suggest you ask ROS channels for more information

closed 2023-08-03 by Myzhar