stereolabs/zed-ros2-wrapper #174: Zed Node terminates after throwing an instance of 'std::out_of_range'

Aniket-Gujarathi, 2023-08-08:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I am using the ZEDXM camera with this project to launch the zed_nodes, operating on a Jetson Orin AGX platform. I am using the [Dockerfile.l4t35_1-humble-devel](https://github.com/stereolabs/zed-ros2-wrapper/blob/master/docker/Dockerfile.l4t35_1-humble-devel) dockerfile provided. I am able to build the docker container and launch the camera node, but after some time of working fine, the zed_node terminates and gives an error:
```
[zed_wrapper-2] terminate called after throwing an instance of 'std::out_of_range'
[zed_wrapper-2]   what():  deque::_M_range_check: __n (which is 11)>= this->size() (which is 1)
[ERROR] [zed_wrapper-2]: process has died [pid 590, exit code -6, cmd '/root/ros2_ws/install/zed_wrapper/lib/zed_wrapper/zed_wrapper --ros-args -r __node:=zed_node -r __ns:=/zedxm --params-file /root/ros2_ws/install/zed_wrapper/share/zed_wrapper/config/common.yaml --params-file /root/ros2_ws/install/zed_wrapper/share/zed_wrapper/config/zedxm.yaml --params-file /tmp/launch_params_b448xxaz'].```
 

### Steps to Reproduce

1. I modify the Dockerfile to change the L4T_MINOR version to 3 instead of 1 (as per my system's specifications).
2. I build the docker container provided [here](https://github.com/stereolabs/zed-ros2-wrapper/blob/master/docker/Dockerfile.l4t35_1-humble-devel).
3. After successfully building the container, I run the zed_node using `ros2 launch zed_wrapper zed_camera.launch.py camera_model:=zedxm publish_tf:=false`. I am publishing the tf using an external node and so set the publish_tf argument to false. 
4. The cameras start working fine initally
```[zed_wrapper-2] [INFO] [1691502633.894414206] [zedxm.zed_node]: *** TF FRAMES ***                                                                                                                          
[zed_wrapper-2] [INFO] [1691502633.894435613] [zedxm.zed_node]:  * Map                  -> map                                                                                                             
[zed_wrapper-2] [INFO] [1691502633.894451260] [zedxm.zed_node]:  * Odometry             -> odom                                                                                                            
[zed_wrapper-2] [INFO] [1691502633.894465499] [zedxm.zed_node]:  * Base                 -> base_link                                                                                                       
[zed_wrapper-2] [INFO] [1691502633.894478554] [zedxm.zed_node]:  * Camera               -> zedxm_camera_center
[zed_wrapper-2] [INFO] [1691502633.894491289] [zedxm.zed_node]:  * Left                 -> zedxm_left_camera_frame
[zed_wrapper-2] [INFO] [1691502633.894504057] [zedxm.zed_node]:  * Left Optical         -> zedxm_left_camera_optical_frame
[zed_wrapper-2] [INFO] [1691502633.894517592] [zedxm.zed_node]:  * RGB                  -> zedxm_left_camera_frame
[zed_wrapper-2] [INFO] [1691502633.894530487] [zedxm.zed_node]:  * RGB Optical          -> zedxm_left_camera_frame
[zed_wrapper-2] [INFO] [1691502633.894542806] [zedxm.zed_node]:  * Right                -> zedxm_right_camera_frame
[zed_wrapper-2] [INFO] [1691502633.894559733] [zedxm.zed_node]:  * Right Optical        -> zedxm_right_camera_optical_frame
[zed_wrapper-2] [INFO] [1691502633.894574421] [zedxm.zed_node]:  * IMU                  -> zedxm_imu_link
[zed_wrapper-2] [INFO] [1691502633.894643921] [zedxm.zed_node]: *** PUBLISHED TOPICS ***
[zed_wrapper-2] [INFO] [1691502633.964395792] [zedxm.zed_node]: Advertised on topic: /zedxm/zed_node/rgb/image_rect_color
[zed_wrapper-2] [INFO] [1691502633.975521096] [zedxm.zed_node]: Advertised on topic: /zedxm/zed_node/rgb/image_rect_gray
[zed_wrapper-2] [INFO] [1691502633.975615874] [zedxm.zed_node]: Advertised on topic: /zedxm/zed_node/rgb/camera_info
[zed_wrapper-2] [INFO] [1691502633.978543128] [zedxm.zed_node]: Advertised on topic: /zedxm/zed_node/rgb_raw/image_raw_color
[zed_wrapper-2] [INFO] [1691502633.988578063] [zedxm.zed_node]: Advertised on topic: /zedxm/zed_node/rgb_raw/image_raw_gray
[zed_wrapper-2] [INFO] [1691502633.988667114] [zedxm.zed_node]: Advertised on topic: /zedxm/zed_node/rgb_raw/camera_info
[zed_wrapper-2] [INFO] [1691502633.991640221] [zedxm.zed_node]: Advertised on topic: /zedxm/zed_node/left/image_rect_color
[zed_wrapper-2] [INFO] [1691502633.993473010] [zedxm.zed_node]: Advertised on topic: /zedxm/zed_node/left/image_rect_gray
[zed_wrapper-2] [INFO] [1691502633.993529166] [zedxm.zed_node]: Advertised on topic: /zedxm/zed_node/left/camera_info
[zed_wrapper-2] [INFO] [1691502634.006919619] [zedxm.zed_node]: Advertised on topic: /zedxm/zed_node/left_raw/image_raw_color
[zed_wrapper-2] [INFO] [1691502634.009066438] [zedxm.zed_node]: Advertised on topic: /zedxm/zed_node/left_raw/image_raw_gray
[zed_wrapper-2] [INFO] [1691502634.009113315] [zedxm.zed_node]: Advertised on topic: /zedxm/zed_node/left_raw/camera_info
[zed_wrapper-2] [INFO] [1691502634.011101263] [zedxm.zed_node]: Advertised on topic: /zedxm/zed_node/right/image_rect_color
[zed_wrapper-2] [INFO] [1691502634.012926885] [zedxm.zed_node]: Advertised on topic: /zedxm/zed_node/right/image_rect_gray
[zed_wrapper-2] [INFO] [1691502634.013008896] [zedxm.zed_node]: Advertised on topic: /zedxm/zed_node/right/camera_info
[zed_wrapper-2] [INFO] [1691502634.015129829] [zedxm.zed_node]: Advertised on topic: /zedxm/zed_node/right_raw/image_raw_color
[zed_wrapper-2] [INFO] [1691502634.017006008] [zedxm.zed_node]: Advertised on topic: /zedxm/zed_node/right_raw/image_raw_gray
[zed_wrapper-2] [INFO] [1691502634.017051893] [zedxm.zed_node]: Advertised on topic: /zedxm/zed_node/right_raw/camera_info
[zed_wrapper-2] [INFO] [1691502634.040982661] [zedxm.zed_node]: Advertised on topic: /zedxm/zed_node/stereo/image_rect_color
[zed_wrapper-2] [INFO] [1691502634.046191158] [zedxm.zed_node]: Advertised on topic: /zedxm/zed_node/stereo_raw/image_raw_color
[zed_wrapper-2] [INFO] [1691502634.048886010] [zedxm.zed_node]: Advertised on topic: /zedxm/zed_node/imu/data
[zed_wrapper-2] [INFO] [1691502634.049692203] [zedxm.zed_node]: Advertised on topic: /zedxm/zed_node/imu/data_raw
[zed_wrapper-2] [INFO] [1691502634.051175253] [zedxm.zed_node]: Advertised on topic: /zedxm/zed_node/temperature/imu
[zed_wrapper-2] [INFO] [1691502634.051932105] [zedxm.zed_node]: Advertised on topic: /zedxm/zed_node/left_cam_imu_transform
[zed_wrapper-2] [INFO] [1691502634.051995301] [zedxm.zed_node]: Camera-IMU Translation: 
[zed_wrapper-2]  -0.002 -0.023 0.00022
[zed_wrapper-2] [INFO] [1691502634.052056673] [zedxm.zed_node]: Camera-IMU Rotation:
[zed_wrapper-2] FFFFCFE60C70
[zed_wrapper-2] 0.999990 -0.001235 0.004384
[zed_wrapper-2] 0.001255 0.999989 -0.004636
[zed_wrapper-2] -0.004378 0.004641 0.999980
[zed_wrapper-2] 
[zed_wrapper-2] [INFO] [1691502634.052083424] [zedxm.zed_node]: *** Subscribers ***
```
5. After running perfectly for some time (around an hour of operation), the node terminates giving the error mentioned above.
...

### Expected Result

The node should not terminate abruptly over extended operation time.

### Actual Result

```
[zed_wrapper-2] terminate called after throwing an instance of 'std::out_of_range'
[zed_wrapper-2]   what():  deque::_M_range_check: __n (which is 11)>= this->size() (which is 1)
[ERROR] [zed_wrapper-2]: process has died [pid 590, exit code -6, cmd '/root/ros2_ws/install/zed_wrapper/lib/zed_wrapper/zed_wrapper --ros-args -r __node:=zed_node -r __ns:=/zedxm --params-file /root/ros2_ws/install/zed_wrapper/share/zed_wrapper/config/common.yaml --params-file /root/ros2_ws/install/zed_wrapper/share/zed_wrapper/config/zedxm.yaml --params-file /tmp/launch_params_b448xxaz'].```

### ZED Camera model

ZED Mini

### Environment

```shell
OS: Ubuntu 20.04
OS Type: 64-bit
Processor: ARM
GPU: Nvidia Tegra Orin
Nvidia Jetpack version: 5.1.1
ZED SDK Version: L4T 35.3 for Jetpack 5.1.1
```

Aniket-Gujarathi, 2023-08-08:
Here is the configuration file that I use:

    ros__parameters:
    
        general:
            svo_file: "" # usually overwritten by launch file
            svo_loop: false # Enable loop mode when using an SVO as input source
            svo_realtime: true # if true the SVO will be played trying to respect the original framerate eventually skipping frames, otherwise every frame will be processed respecting the `pub_frame_rate` setting
            camera_timeout_sec: 5
            camera_max_reconnect: 5
            camera_flip: false
            zed_id: 0 # IMPORTANT: to be used in simulation to distinguish individual cameras im multi-camera configurations - usually overwritten by launch file
            serial_number: 0 # usually overwritten by launch file
            pub_resolution: 'MEDIUM' # The resolution used for output. 'HD2K', 'HD1080', 'HD1200', 'HD720', 'MEDIUM', 'SVGA', 'VGA', 'LOW'
            pub_frame_rate: 30.0 # [DYNAMIC] - frequency of publishing of visual images and depth images
            gpu_id: -1
            region_of_interest: '[]' # A polygon defining the ROI where the ZED SDK perform the processing ignoring the rest. Coordinates must be normalized to '1.0' to be resolution independent.
            #region_of_interest: '[[0.25,0.33],[0.75,0.33],[0.75,0.5],[0.5,0.75],[0.25,0.5]]' # A polygon defining the ROI where the ZED SDK perform the processing ignoring the rest. Coordinates must be normalized to '1.0' to be resolution independent.
            #region_of_interest: '[[0.25,0.25],[0.75,0.25],[0.75,0.75],[0.25,0.75]]' # A polygon defining the ROI where the ZED SDK perform the processing ignoring the rest. Coordinates must be normalized to '1.0' to be resolution independent.
            #region_of_interest: '[[0.5,0.25],[0.75,0.5],[0.5,0.75],[0.25,0.5]]' # A polygon defining the ROI where the ZED SDK perform the processing ignoring the rest. Coordinates must be normalized to '1.0' to be resolution independent.
            sdk_verbose: 1

        video:
            brightness: 4 # [DYNAMIC] Not available for ZED X/ZED X Mini
            contrast: 4 # [DYNAMIC] Not available for ZED X/ZED X Mini
            hue: 0 # [DYNAMIC] Not available for ZED X/ZED X Mini
            saturation: 4 # [DYNAMIC]
            sharpness: 4 # [DYNAMIC]
            gamma: 8 # [DYNAMIC]
            auto_exposure_gain: true # [DYNAMIC]
            exposure: 80 # [DYNAMIC]
            gain: 80 # [DYNAMIC]
            auto_whitebalance: true # [DYNAMIC]
            whitebalance_temperature: 42 # [DYNAMIC] - [28,65] works only if `auto_whitebalance` is false
            qos_history: 1 # '1': KEEP_LAST - '2': KEEP_ALL
            qos_depth: 1 # Queue size if using KEEP_LAST
            qos_reliability: 1 # '1': RELIABLE - '2': BEST_EFFORT -
            qos_durability: 2 # '1': TRANSIENT_LOCAL - '2': VOLATILE

        depth:
            depth_mode: 'NONE' # Matches the ZED SDK setting: 'NONE', 'PERFORMANCE', 'QUALITY', 'ULTRA', 'NEURAL' - Note: if 'NONE' all the modules that requires depth extraction are disabled by default (Pos. Tracking, Obj. Detection, Mapping, ...)
            depth_stabilization: 1 # Forces positional tracking to start if major than 0 - Range: [0,100]
            openni_depth_mode: false # 'false': 32bit float [meters], 'true': 16bit unsigned int [millimeters]
            point_cloud_freq: 10.0 # [DYNAMIC] - frequency of the pointcloud publishing (equal or less to `grab_frame_rate` value)
            depth_confidence: 50 # [DYNAMIC]
            depth_texture_conf: 100 # [DYNAMIC]
            remove_saturated_areas: true # [DYNAMIC]
            qos_history: 1 # '1': KEEP_LAST - '2': KEEP_ALL
            qos_depth: 1 # Queue size if using KEEP_LAST
            qos_reliability: 1 # '1': RELIABLE - '2': BEST_EFFORT -
            qos_durability: 2 # '1': TRANSIENT_LOCAL - '2': VOLATILE 

        pos_tracking:
            pos_tracking_enabled: false # True to enable positional tracking from start            
            imu_fusion: true # enable/disable IMU fusion. When set to false, only the optical odometry will be used.
            publish_tf: true # [usually overwritten by launch file] publish `odom -> base_link` TF
            publish_map_tf: true # [usually overwritten by launch file] publish `map -> odom` TF
            publish_imu_tf: true # [usually overwritten by launch file] enable/disable the IMU TF broadcasting
            base_frame: "base_link" # usually overwritten by launch file
            map_frame: "map"
            odometry_frame: "odom"
            area_memory_db_path: ""
            area_memory: true # Enable to detect loop closure
            depth_min_range: 0.0 # Set this value for removing fixed zones of the robot in the FoV of the camerafrom the visual odometry evaluation
            set_as_static: false # If 'true' the camera will be static and not move in the environment
            set_gravity_as_origin: true # If 'true' align the positional tracking world to imu gravity measurement. Keep the yaw from the user initial pose.
            floor_alignment: false # Enable to automatically calculate camera/floor offset
            initial_base_pose: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0] # Initial position of the `base_frame` in the map -> [X, Y, Z, R, P, Y]
            init_odom_with_first_valid_pose: true # Enable to initialize the odometry with the first valid pose
            path_pub_rate: 2.0 # [DYNAMIC] - Camera trajectory publishing frequency
[... cut ...]
            debug_video_depth: false
            debug_camera_controls: false
            debug_point_cloud: false
            debug_positional_tracking: false
            debug_gnss: false
            debug_sensors: false
            debug_mapping : false
            debug_terrain_mapping : false
            debug_object_detection : false

Currently, I am just using the docker container to run the zed camera. Also, after crashing, the node is not able to detect the camera sometimes and I have to restart the Zed daemon.

Patrick-AA, 2023-08-10:
Hello,
This is not a bug of the ZED ROS2 Wrapper, but something related to the behavior of the ZED SDK.
Please report your problem to [support@stereolabs.com](mailto:support@stereolabs.com)

Patrick-AA, 2023-08-24:
I will close this issue as the problem has been fixed in SDK version 4.0.6.

closed 2023-08-24 by Patrick-AA