stereolabs/zed-ros2-wrapper #175: FPS at Jetson Xavier NX

eladpar, 2023-08-09:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Ive been trying to work with the ros2 wrapper on a jetson.
I set the FPS parameter to 60/30 and I always get around 15-20 at the rgb image topic.
I would say that this is just the limitation of the jetson, but when using the ZED Explorer, it seems to work smoothly with 60 fps and 30 fps.

### Steps to Reproduce

1. buy a jetson xavier nx
2. install ros2
3. launch the node
4. ros2 hz /rgb_topic
...

### Expected Result

30 fps

### Actual Result

15-20 hz

### ZED Camera model

ZED2i

### Environment

```shell
OS: Ubuntu 20
CPU: ARM
Plaform: jetson Xavier nx
git tag foxy-humble-v3.8
```

Patrick-AA, 2023-08-10:
Hello,
You can monitor these values using the [node diagnostic](https://www.stereolabs.com/docs/ros2/#node-diagnostic).
You can also check [composition](https://www.stereolabs.com/docs/ros2/ros2-composition/) to optimize the performance of your system.

eladpar, 2023-08-10:
> Hello, You can monitor these values using the [node diagnostic](https://www.stereolabs.com/docs/ros2/#node-diagnostic). You can also check [composition](https://www.stereolabs.com/docs/ros2/ros2-composition/) to optimize the performance of your system.

Thanks for the quick reply!
I have used composition with no big improvement.
and i've used debug mode =True (seems like node diagnostics).
as mentioned in previous issues in order to get the mean rate of publishing from the node itself.
is seems like the node publishes at a maximum of 20 fps , and my question does it make sense?
keeping in mind that at the viewer it says 60 fps without any problem.

closed 2023-09-16 by github-actions[bot]