stereolabs/zed-ros2-wrapper #176: Cannot build zed_components

KSorte, 2023-08-09:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I am trying to build the zed_wrapper for ros2 humble. zed_components not building. 
CMake Error at CMakeLists.txt:78 (find_package):
  By not providing "FindZED.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "ZED", but
  CMake did not find one.

  Could not find a package configuration file provided by "ZED" (requested
  version 4) with any of the following names:

    ZEDConfig.cmake
    zed-config.cmake

  Add the installation prefix of "ZED" to CMAKE_PREFIX_PATH or set "ZED_DIR"
  to a directory containing one of the above files.  If "ZED" provides a
  separate development package or SDK, be sure it has been installed.

### Steps to Reproduce

1.git clone --recursive https://github.com/stereolabs/zed-ros2-wrapper.git   [inside src of workspace]
2. colcon build --symlink-install --packages-select zed_wrapper zed_ros2 zed_components zed_interfaces [inside ws]

### Expected Result

The package should have built right?

### Actual Result

Got the following error while building zed_components
CMake Error at CMakeLists.txt:78 (find_package):
  By not providing "FindZED.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "ZED", but
  CMake did not find one.

  Could not find a package configuration file provided by "ZED" (requested
  version 4) with any of the following names:

    ZEDConfig.cmake
    zed-config.cmake

  Add the installation prefix of "ZED" to CMAKE_PREFIX_PATH or set "ZED_DIR"
  to a directory containing one of the above files.  If "ZED" provides a
  separate development package or SDK, be sure it has been installed.

### ZED Camera model

ZED

### Environment

```shell
Running ROS2 Humble
```

closed 2023-08-09 by KSorte

Michael-Equi, 2024-04-15:
Hello I have this exact issue. What was the solution?

hotung1027, 2025-01-16:
it should be related to permissin issues, for my case.
the whole /usr/local/zed directory is under ownership root:zed,
so I chown -R $USER:$USER /usr/local/zed, to allow cmake to find the share library.