stereolabs/zed-ros2-wrapper #179: colcon build error.

Hensbag, 2023-08-24:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I want to install use my zed2 camera with             

cuda_version": "11.6
cudnn_version": "8.4.1.50-1+cuda11.6
tensorrt_version": "8.4.2-1+cuda11.6
I used ZED sdk 3.7.7 
when I try to build my zed_ws it ' s failing
I use humble on ubuntu 22.04

### Steps to Reproduce

1.I installed Cuda tensorrt cudnn 
2.tried to build my package

### Expected Result

colcon build succes

### Actual Result

Starting >>> zed_interfaces
Finished <<< zed_interfaces [0.70s]                     
Starting >>> zed_components
--- stderr: zed_components                             
/home/hasan/zed_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp: In member function ‘void stereolabs::ZedCamera::initParameters()’:
/home/hasan/zed_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:267:5: error: ‘set_on_parameters_set_callback’ was not declared in this scope
  267 |     set_on_parameters_set_callback(std::bind(&ZedCamera::callback_paramChange, this, _1));
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /opt/ros/humble/include/rclcpp/rclcpp/logging.hpp:24,
                 from /opt/ros/humble/include/rclcpp/rclcpp/client.hpp:40,
                 from /opt/ros/humble/include/rclcpp/rclcpp/callback_group.hpp:24,
                 from /opt/ros/humble/include/rclcpp/rclcpp/any_executable.hpp:20,
                 from /opt/ros/humble/include/rclcpp/rclcpp/memory_strategy.hpp:25,
                 from /opt/ros/humble/include/rclcpp/rclcpp/memory_strategies.hpp:18,
                 from /opt/ros/humble/include/rclcpp/rclcpp/executor_options.hpp:20,
                 from /opt/ros/humble/include/rclcpp/rclcpp/executor.hpp:37,
                 from /opt/ros/humble/include/rclcpp/rclcpp/executors/multi_threaded_executor.hpp:25,
                 from /opt/ros/humble/include/rclcpp/rclcpp/executors.hpp:21,
                 from /opt/ros/humble/include/rclcpp/rclcpp/rclcpp.hpp:155,
                 from /home/hasan/zed_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/include/zed_camera_component.hpp:31,
                 from /home/hasan/zed_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:25:
/home/hasan/zed_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp: In member function ‘void stereolabs::ZedCamera::getGeneralParams()’:
/home/hasan/zed_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:327:35: warning: too many arguments for format [-Wformat-extra-args]
  327 |         RCLCPP_WARN(get_logger(), "'pub_frame_rate' cannot be bigger than 'grab_frame_rate'", paramName.c_str());
      |                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/hasan/zed_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp: In member function ‘bool stereolabs::ZedCamera::getCamera2BaseTransform()’:
/home/hasan/zed_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:2443:108: error: no matching function for call to ‘rclcpp::Duration::Duration(double)’
 2443 |                 mTfBuffer->lookupTransform(mCameraFrameId, mBaseFrameId, TIMEZERO_SYS, rclcpp::Duration(0.1));
      |                                                                                                            ^
In file included from /opt/ros/humble/include/rclcpp/rclcpp/time.hpp:24,
                 from /opt/ros/humble/include/rclcpp/rclcpp/clock.hpp:24,
                 from /home/hasan/zed_ws/src/zed-ros2-wrapper/zed_components/src/tools/include/sl_tools.h:49,
                 from /home/hasan/zed_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/include/zed_camera_component.hpp:29,
                 from /home/hasan/zed_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:25:
/opt/ros/humble/include/rclcpp/rclcpp/duration.hpp:48:3: note: candidate: ‘template<class Rep, class Period> rclcpp::Duration::Duration(const std::chrono::duration<_Rep1, _Period1>&)’
   48 |   Duration(const std::chrono::duration<Rep, Period> & duration)  // NOLINT(runtime/explicit)
      |   ^~~~~~~~
/opt/ros/humble/include/rclcpp/rclcpp/duration.hpp:48:3: note:   template argument deduction/substitution failed:
/home/hasan/zed_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:2443:108: note:   mismatched types ‘const std::chrono::duration<_Rep1, _Period1>’ and ‘double’
 2443 |                 mTfBuffer->lookupTransform(mCameraFrameId, mBaseFrameId, TIMEZERO_SYS, rclcpp::Duration(0.1));
      |                                                                                                            ^
In file included from /opt/ros/humble/include/rclcpp/rclcpp/time.hpp:24,
                 from /opt/ros/humble/include/rclcpp/rclcpp/clock.hpp:24,
[... 143 lines cut ...]
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< zed_components [6.17s, exited with code 2]

Summary: 1 package finished [7.04s]
  1 package failed: zed_components
  1 package had stderr output: zed_components
  2 packages not processed

### ZED Camera model

ZED2

### Environment

```shell
ubuntu 22.04
nvidia rtx2060 6gb
ros humble
zed sdk 3.7.7
```

### Anything else?

also I'm trying to imply this camera to autonomous driving stack called autoware which is not supported by this autonomous stack.any help would be appreciated .

Patrick-AA, 2023-08-25:
Hello,

Are you using the tag for SDK version 3.7 [foxy-v3.7](https://github.com/stereolabs/zed-ros2-wrapper/tree/foxy-v3.7) ?

Hensbag, 2023-08-25:
there is a sdk version 3.7.7 'https://download.stereolabs.com/zedsdk/3.7/cu117/ubuntu22' Since I use Cuda 11.6 and cudnn_version": "8.4.1.50-1+cuda11.6 tensorrt_version": "8.4.2-1+cuda11.6 it seemed ok with me since the autonomous driving stack autoware using it.

PenghuySrean, 2023-08-29:
I am running wsl2 ubuntu 20 (ros2 foxy), my cuda version is 12, hence I have installed the compatible version of zed-sdk accordingly. while trying to build the package I found this error threw on the terminal.

```
CMake Error at /mnt/c/Program Files (x86)/ZED SDK/zed-config.cmake:55 (message):
  You've selected the 32bit version of Unix Makefiles.

   Please delete the cache (file->Delete Cache) and use the 64bit version. (Unix Makefiles Win64)
Call Stack (most recent call first):
  CMakeLists.txt:78 (find_package)
```

I am not sure how to the change the configuration from 32bit to 64bit. I am not sure where the 32 bit was configured.

closed 2023-09-11 by Hensbag