stereolabs/zed-ros2-wrapper #181: colcon build error in docker

eladpar, 2023-09-03:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

hey im trying to create a ros2 zed wrapper docker in amd64 ans it is not compiling
here is my Dockerfile:

```
ARG UBUNTU_RELEASE_YEAR=20
ARG CUDA_MAJOR=12
ARG CUDA_MINOR=1.0

FROM nvidia/cuda:${CUDA_MAJOR}.${CUDA_MINOR}-devel-ubuntu${UBUNTU_RELEASE_YEAR}.04
ARG UBUNTU_RELEASE_YEAR=20
ARG CUDA_MAJOR=12
ARG CUDA_MINOR=1.0
ARG ZED_SDK_MAJOR=4
ARG ZED_SDK_MINOR=0

ENV NVIDIA_DRIVER_CAPABILITIES \
    ${NVIDIA_DRIVER_CAPABILITIES:+$NVIDIA_DRIVER_CAPABILITIES,}compute,video,utility

RUN echo "Europe/Paris" > /etc/localtime ; echo "CUDA Version ${CUDA_MAJOR}.${CUDA_MINOR}.0" > /usr/local/cuda/version.txt

# Setup the ZED SDK
RUN apt-get update -y || true ; apt-get install --no-install-recommends lsb-release wget less udev sudo zstd build-essential cmake python3 python3-pip libpng-dev libgomp1 -y ; \
    #python3 -m pip install --upgrade pip ; \
    python3 -m pip install numpy opencv-python ; \
    wget -q -O ZED_SDK_Linux_Ubuntu${UBUNTU_RELEASE_YEAR}.run https://download.stereolabs.com/zedsdk/${ZED_SDK_MAJOR}.${ZED_SDK_MINOR}/cu${CUDA_MAJOR}${CUDA_MINOR%.*}/ubuntu${UBUNTU_RELEASE_YEAR} && \
    chmod +x ZED_SDK_Linux_Ubuntu${UBUNTU_RELEASE_YEAR}.run ; ./ZED_SDK_Linux_Ubuntu${UBUNTU_RELEASE_YEAR}.run -- silent skip_tools skip_cuda && \
    ln -sf /lib/x86_64-linux-gnu/libusb-1.0.so.0 /usr/lib/x86_64-linux-gnu/libusb-1.0.so && \
    rm ZED_SDK_Linux_Ubuntu${UBUNTU_RELEASE_YEAR}.run && \
    rm -rf /var/lib/apt/lists/*

WORKDIR /usr/local/zed/
WORKDIR /external_libs_build

# ========================
# ROS2-FOXY INSTALLATION
# ========================

ENV DEBIAN_FRONTEND=noninteractive

RUN apt-get update && \
	apt-get install -y --no-install-recommends \
	locales && \
	locale-gen en_US.UTF-8 && \
	update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8 && \
	apt-get clean && \
	rm -rf /var/lib/apt/lists/*

ENV LANG=en_US.UTF-8

RUN apt-get update && \
	apt-get install -y --no-install-recommends \
 	software-properties-common && \
	add-apt-repository universe && \
	apt-get clean && \
	rm -rf /var/lib/apt/lists/*

RUN apt update && \
	curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key \
	-o /usr/share/keyrings/ros-archive-keyring.gpg

RUN sh -c 'echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(. /etc/os-release && echo $UBUNTU_CODENAME) main" > /etc/apt/sources.list.d/ros2-latest.list'

RUN apt-get update && \
[... 175 lines cut ...]
66.23 make[1]: *** [CMakeFiles/Makefile2:79: CMakeFiles/zed_wrapper.dir/all] Error 2
66.23 make: *** [Makefile:141: all] Error 2
66.23 ---
66.23 Failed   <<< zed_wrapper [4.52s, exited with code 2]
66.26 
66.26 Summary: 2 packages finished [51.4s]
66.26   1 package failed: zed_wrapper
66.26   2 packages had stderr output: zed_components zed_wrapper
66.26   1 package not processed
------
Dockerfile:82
--------------------
  81 |     
  82 | >>> RUN bash -c '. /opt/ros/foxy/setup.bash && \ 
  83 | >>> sudo rosdep init && \
  84 | >>> rosdep update && \
  85 | >>> git clone  --recursive https://github.com/stereolabs/zed-ros2-wrapper.git && \
  86 | >>> cd .. && \
  87 | >>> rosdep install --from-paths src --ignore-src -r -y && \
  88 | >>> colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release --parallel-workers $(nproc) && \
  89 | >>> . install/setup.bash'
  90 |     
--------------------
ERROR: failed to solve: process "/bin/sh -c bash -c '. /opt/ros/foxy/setup.bash && sudo rosdep init && rosdep update && git clone  --recursive https://github.com/stereolabs/zed-ros2-wrapper.git && cd .. && rosdep install --from-paths src --ignore-src -r -y && colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release --parallel-workers $(nproc) && . install/setup.bash'" did not complete successfully: exit code: 2
```

Patrick-AA, 2023-09-04:
Hello,

Can you try the colcon build command with something similar to [Dockerfile.l4t35_1-humble-release](https://github.com/stereolabs/zed-ros2-wrapper/blob/78172a486abf9d97e8e4c0bc293b267bd7e1623c/docker/Dockerfile.l4t35_1-humble-release#L78-L85C28) ?

Mat198, 2023-09-28:
I think this can solve your issue:
https://stackoverflow.com/questions/59691207/docker-build-with-nvidia-runtime

eladpar, 2023-09-28:
> Hello,
> 
> Can you try the colcon build command with something similar to [Dockerfile.l4t35_1-humble-release](https://github.com/stereolabs/zed-ros2-wrapper/blob/78172a486abf9d97e8e4c0bc293b267bd7e1623c/docker/Dockerfile.l4t35_1-humble-release#L78-L85C28) ?

this worked but although its not foxy..

> I think this can solve your issue: https://stackoverflow.com/questions/59691207/docker-build-with-nvidia-runtime

haven't tried this, it might work

closed 2023-11-04 by github-actions[bot]