stereolabs/zed-ros2-wrapper #184: Unable to increase FPS for ZED Mini

sandeepdutta, 2023-10-04:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I am unable to increase the FPS for ZED Mini in the ros wrapper. I tried increasing the grab_rate in "zedm.yaml"
but it gave a waring and reverted back to 15 FPS. I want to capture @ WVGA resolution , according to the sec I should about
100 FPS . How do I set this up in ROS.

Sandeep

### Steps to Reproduce

1.  Set publish_rate to > 15 in common.yaml
2. Set grab_rate >= publish_rate in zedm.yaml
3. launch zedm.launch.py
Observe warnings

### Expected Result

Publish rate increase

### Actual Result

Publish defaults down to 15 FPS

### ZED Camera model

ZED Mini

### Environment

```shell
OS : Jet Pack 5.2
CPU : ARM
GPU; Nvidia Jetson ORIN AGX
Latest Zed SDK 4.0
ROS Humble
```

Myzhar, 2023-10-05:
Hi @sandeepdutta 
can you please share your `common.yaml` and `zedm.yaml` files?
Please also post the output of the command `$ ros2 launch zed_wrapper zedm.launch.py`

sandeepdutta, 2023-10-05:
Hi Myzhar,

I have uploaded the three files you requested, note if I change the publish rate only it will say publish_rate cannot
be greater than grab_rate and reverts it to grab_rate. If I increase grab_rate it says it is an invalid value and reverts it to 15.

[zed_upload.zip](https://github.com/stereolabs/zed-ros2-wrapper/files/12819606/zed_upload.zip)

Myzhar, 2023-10-05:
`grab_frame_rate: 20` -> this is not valid. In VGA mode you can set it to 15,30,60, or 100

closed 2023-10-05 by sandeepdutta