stereolabs/zed-ros2-wrapper #187: Docker build error | colcon build | foxy-humble-v3.8.2

ironoa, 2023-10-12:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

On branch foxy-humble-v3.8.2 the docker image `Dockerfile.l4t35_1-humble-release` is not building properly.

### Steps to Reproduce

1. `docker build -f Dockerfile.l4t35_1-humble-release .`

### Expected Result

Image being built successfully 

### Actual Result

```
25.32 CMake Error at CMakeLists.txt:78 (find_package):
25.32 -- Configuring incomplete, errors occurred!
25.32   Could not find a configuration file for package "ZED" that is compatible
25.32   with requested version "4".
25.32 
25.32   The following configuration files were considered but not accepted:
25.32 
25.32     /usr/local/zed/zed-config.cmake, version: 3.8.2
25.32 
25.32 
25.32 
25.33 -- Check for working CXX compiler: /usr/bin/c++ - skipped
25.33 -- Detecting CXX compile features
25.33 -- Detecting CXX compile features - done
25.33 -- Found ament_cmake: 1.3.4 (/opt/ros/humble/install/share/ament_cmake/cmake)
25.36 --- stderr: zed_components
25.36 CMake Error at CMakeLists.txt:78 (find_package):
25.36   Could not find a configuration file for package "ZED" that is compatible
25.36   with requested version "4".
25.36 
25.36   The following configuration files were considered but not accepted:
25.36 
25.36     /usr/local/zed/zed-config.cmake, version: 3.8.2
25.36 
25.36 
25.36 
25.36 ---
25.36 Failed   <<< zed_components [0.30s, exited with code 1]
```

### ZED Camera model

ZED

### Environment

```shell
- ARM
- Docker version 24.0.6, build ed223bc
```

ironoa, 2023-10-12:
I fixed it [like that](https://github.com/ironoa/zed-ros2-wrapper/commit/787271d)

A generic solution should be crafted for branch master

Myzhar, 2023-10-12:
Thank you for providing a working solution

closed 2023-10-12 by Myzhar