stereolabs/zed-ros2-wrapper #188: Rolling Support

peterdavidfagan, 2023-10-25:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

Support ROS rolling in order to enable applications that leverage the latest versions of ROS packages.

### Use-Case

Robot learning research that depends on updates to `moveit_py`.

peterdavidfagan, 2023-10-25:
Thanks @Myzhar much appreciated, for now I will remove these [checks](https://github.com/stereolabs/zed-ros2-wrapper/blob/66e6d123615294c19c710de4813f7e6268107dc8/zed_components/src/zed_camera/src/zed_camera_component.cpp#L33) and test the launch files.

Myzhar, 2023-10-25:
Maybe you can modify this part in all the CMakeLists.txt files too:
https://github.com/stereolabs/zed-ros2-wrapper/blob/66e6d123615294c19c710de4813f7e6268107dc8/zed_wrapper/CMakeLists.txt#L7-L44
so you can add officially the `ROLLING` support

Myzhar, 2023-10-25:
a nice and good at line 38 before the `else()`
```
   elseif(${FOUND_ROS2_DISTRO} STREQUAL "rolling") 
     #message("* ROS2 ${FOUND_ROS2_DISTRO} is officially supported by this package.") 
     add_definitions(-DFOUND_ROLLING) 
```
and you are good

peterdavidfagan, 2023-10-25:
Ok yes this is good advice, also a lot better than the quick solution I was going for to test things. Shall I open a PR with these changes?

Myzhar, 2023-10-25:
Yes, you're welcome to open a PR