stereolabs/zed-ros2-wrapper #190: UDP communication interferece between ZED2i and Robot arm via ROS2

Bardhitoo, 2023-10-27:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

My team and I are currently facing the issue of robot disconnecting when connecting ZED2i camera with laptop (via ROS2) making it impossible to use both at the same time.  

Everything runs smoothly when computer is connected to either robot or ZED 2i. Once both are connected to the computer, the issue comes up. The problem seems to be consistent regardless of the order of connection with laptop (1. ZED2i -> 2. Robot and 1. Robot -> 2. ZED2i). 

### Steps to Reproduce

1. Connect camera and robot to PC 
2. Launch communication between robot and PC
3. Launch communication between ZED and PC ( `ros2 launch zed_wrapper zed2i.launch.py`)
4. 
...

### Expected Result

No errors in communication between   ZED2i <=> PC <=> Robot

### Actual Result

`[control_node-1] 1698357420.879862 [0]        tev: ddsi_udp_conn_write to udp/239.255.0.1:7400 failed with retcode -1
1698357420.902612 [0]        tev: ddsi_udp_conn_write to udp/239.255.0.1:7400 failed with retcode -1
[robot_manager_node-2] 1698357420.979563 [0]        tev: ddsi_udp_conn_write to udp/239.255.0.1:7400 failed with retcode -1
[robot_manager_node-2] 1698357420.979697 [0]        tev: ddsi_udp_conn_write to udp/239.255.0.1:7400 failed with retcode -1
[control_node-1] 1698357420.979805 [0]        tev: ddsi_udp_conn_write to udp/239.255.0.1:7400 failed with retcode -1
[control_node-1] 1698357420.979821 [0]        tev: ddsi_udp_conn_write to udp/239.255.0.1:7400 failed with retcode -1
1698357421.002722 [0]        tev: ddsi_udp_conn_write to udp/239.255.0.1:7400 failed with retcode -1
[robot_manager_node-2] 1698357421.079865 [0]        tev: ddsi_udp_conn_write to udp/239.255.0.1:7400 failed with retcode -1
[control_node-1] 1698357421.079892 [0]        tev: ddsi_udp_conn_write to udp/239.255.0.1:7400 failed with retcode -1
1698357421.102879 [0]        tev: ddsi_udp_conn_write to udp/239.255.0.1:7400 failed with retcode -1
[control_node-1] 1698357421.179963 [0]        tev: ddsi_udp_conn_write to udp/239.255.0.1:7400 failed with retcode -1
[robot_manager_node-2] 1698357421.180049 [0]        tev: ddsi_udp_conn_write to udp/239.255.0.1:7400 failed with retcode -1
1698357421.202999 [0]        tev: ddsi_udp_conn_write to udp/239.255.0.1:7400 failed with retcode -1
[control_node-1] 1698357421.280036 [0]        tev: ddsi_udp_conn_write to udp/239.255.0.1:7400 failed with retcode -1
[robot_manager_node-2] 1698357421.280232 [0]        tev: ddsi_udp_conn_write to udp/239.255.0.1:7400 failed with retcode -1`

### ZED Camera model

ZED2i

### Environment

```shell
OS: Pop!_OS
CPU: 12th gen Intel Core, i9-12900H (20), 
GPU: Nvidia rtx A2000
CUDA version: 12.1
Nvidia drivers: 535.113.01, 
ZED Software SDK: 4.0 
ROS Version: ROS2 Iron
```

Myzhar, 2023-10-27:
Hi @Bardhitoo 
are you sure this is not a ROS 2 issue instead of a ZED ROS2 Wrapper problem?
The ZED Wrapper does not use UDP communication if not through the ROS 2 DDS Middleware. 
Please check your ROS 2 configuration, the DDS Middleware that you are using, and the UDP ports that it requires, maybe they are in conflict.

Start from here:
* ROS 2 DOC: https://docs.ros.org/en/humble/Concepts/Intermediate/About-Domain-ID.html
* ROS 2 answers: https://answers.ros.org/question/295400/what-default-ports-are-used-in-ros2-dds-implementations/

Myzhar, 2023-10-27:
Let's continue the discussion on the [Stereolabs forum](https://community.stereolabs.com/t/udp-communication-interference-between-zed2i-and-robot-arm-via-ros2/3809/2). Closing the issue.

closed 2023-10-27 by Myzhar