stereolabs/zed-ros2-wrapper #197: No EGL display

stevejohn16, 2023-11-27:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I get the following error when I run zed-wrapper2 in a docker
https://drive.google.com/file/d/1wymepKWRCCl2FItUoUl_ZDjV5UfXuktC/view?usp=sharing

### Steps to Reproduce

1. Run the docker: docker run --runtime nvidia -it --privileged --ipc=host --pid=host -e DISPLAY   -v /dev/shm:/dev/shm -v /tmp/.X11-unix/:/tmp/.X11-unix voidwalking-zed:latest
2. Source the file
3. Run the launch file: ros2 launch zed_wrapper [zed2.launch.py](http://zed2.launch.py/) gnss_frame:=navsatfix_frame

### Expected Result

https://drive.google.com/file/d/1VQZ-VdbFvpuYgZT8X2qcf7sE1crlN7VK/view?usp=sharing

### Actual Result

https://drive.google.com/file/d/1wymepKWRCCl2FItUoUl_ZDjV5UfXuktC/view?usp=sharing

### ZED Camera model

ZED2

### Environment

```shell
Computer envrionment:
Linux nano 4.9.253-tegra #1 SMP PREEMPT Mon Jul 26 12:13:06 PDT 2021 aarch64 aarch64 aarch64 GNU/Linux

Docker environment:

FROM dustynv/ros:humble-ros-base-l4t-r35.1.0

ARG ZED_SDK_MAJOR=4
ARG ZED_SDK_MINOR=0
ARG ZED_SDK_PATCH=5
ARG JETPACK_MAJOR=4
ARG JETPACK_MINOR=6
ARG L4T_MAJOR=32
ARG L4T_MINOR=7

ARG ROS2_DIST=humble       # ROS2 distribution

# ZED ROS2 Wrapper dependencies version
ARG XACRO_VERSION=2.0.8
ARG DIAGNOSTICS_VERSION=3.0.0
ARG AMENT_LINT_VERSION=0.12.4
ARG GEOGRAPHIC_INFO_VERSION=1.0.4
ARG ROBOT_LOCALIZATION_VERSION=3.4.2
```

### Anything else?

Unsetting the display solved the issue, not sure how
$ unset DISPLAY

RajUpadhyay, 2023-12-27:
did you do xhost+?
what does xrandr show?

what does echo $DISPLAY show?

its good that you solved the error though.

closed 2024-02-01 by github-actions[bot]

JLBicho, 2024-02-23:
Hi all,

I'm having the same issue

### Device
Jetson Nano
L4T 32.7.4 Jetpack 4.7
Working with screen connected to HDMI

### Docker environment
```
FROM dustynv/ros:humble-ros-base-l4t-r35.1.0

ARG ZED_SDK_MAJOR=4
ARG ZED_SDK_MINOR=0
ARG ZED_SDK_PATCH=7
ARG JETPACK_MAJOR=4
ARG JETPACK_MINOR=6
ARG L4T_MAJOR=35
ARG L4T_MINOR=1

ARG ROS2_DIST=humble       # ROS2 distribution

# ZED ROS2 Wrapper dependencies version
ARG XACRO_VERSION=2.0.8
ARG DIAGNOSTICS_VERSION=3.0.0
ARG AMENT_LINT_VERSION=0.12.4
ARG GEOGRAPHIC_INFO_VERSION=1.0.4
ARG ROBOT_LOCALIZATION_VERSION=3.4.2
```

### Steps to replicate
```
jetson-zed@jetson-zed-desktop:~$ echo $DISPLAY 
:0
jetson-zed@jetson-zed-desktop:~$ xhost +si:localuser:root
localuser:root being added to access control list
jetson-zed@jetson-zed-desktop:~$ docker run --runtime nvidia -it --privileged --ipc=host --pid=host -e NVIDIA_DRIVER_CAPABILITIES=all -e DISPLAY   -v /dev:/dev -v /tmp/.X11-unix/:/tmp/.X11-unix   -v ${HOME}/zed_docker_ai/:/usr/local/zed/resources/   humble_zed:jetson 
ZED ROS2 Docker Image
---------------------
ROS distro:  humble
DDS middleware:  rmw_cyclonedds_cpp
---
Available ZED packages:
zed_components
zed_interfaces
zed_ros2
zed_wrapper
---------------------
sourcing   /opt/ros/humble/install/setup.bash
notfound   /ros_deep_learning/install/setup.bash
ROS_DISTRO humble
ROS_ROOT   /opt/ros/humble
root@9c96f28f86fa:~/ros2_ws# ros2 launch zed_wrapper zed_camera.launch.py camera_model:=zed2i
[INFO] [launch]: All log files can be found below /root/.ros/log/2024-02-23-09-38-42-046490-9c96f28f86fa-6770
[INFO] [launch]: Default logging verbosity is set to INFO
[WARNING] [launch_ros.actions.node]: Parameter file path is not a file: .
[INFO] [robot_state_publisher-1]: process started with pid [6773]
[INFO] [zed_wrapper-2]: process started with pid [6775]
[robot_state_publisher-1] [INFO] [1708677524.002929082] [zed.zed_state_publisher]: got segment zed_camera_center
[robot_state_publisher-1] [INFO] [1708677524.003248881] [zed.zed_state_publisher]: got segment zed_camera_link
[robot_state_publisher-1] [INFO] [1708677524.003325028] [zed.zed_state_publisher]: got segment zed_left_camera_frame
[robot_state_publisher-1] [INFO] [1708677524.003382112] [zed.zed_state_publisher]: got segment zed_left_camera_optical_frame
[robot_state_publisher-1] [INFO] [1708677524.003433520] [zed.zed_state_publisher]: got segment zed_right_camera_frame
[robot_state_publisher-1] [INFO] [1708677524.003481333] [zed.zed_state_publisher]: got segment zed_right_camera_optical_frame
[zed_wrapper-2] nvbuf_utils: Could not get EGL display connection
[zed_wrapper-2] nvbuf_utils: ERROR getting proc addr of eglCreateImageKHR
[zed_wrapper-2] nvbuf_utils: ERROR getting proc addr of eglDestroyImageKHR
[zed_wrapper-2] No EGL Display 
[zed_wrapper-2] nvbufsurftransform: Could not get EGL display connection
[ERROR] [zed_wrapper-2]: process has died [pid 6775, exit code 255, cmd '/root/ros2_ws/install/zed_wrapper/lib/zed_wrapper/zed_wrapper --ros-args -r __node:=zed_node -r __ns:=/zed --params-file /root/ros2_ws/install/zed_wrapper/share/zed_wrapper/config/common.yaml --params-file /root/ros2_ws/install/zed_wrapper/share/zed_wrapper/config/zed2i.yaml --params-file /tmp/launch_params_mfu5j_0h'].
```

Does anyone have any clue on how to solve this?

Thanks in advance!