stereolabs/zed-ros2-wrapper #209: Launchin 2 ZEDX Cams doesn't work

adriandavidauer, 2024-03-07:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I tried to start to camera nodes with `ros2 launch zed_wrapper zed_camera.launch.py camera_model:=zedx zed_id:=0 camera_name:=zed_0` and `ros2 launch zed_wrapper zed_camera.launch.py camera_model:=zedx zed_id:=1 camera_name:=zed_1` and the first node always comes up without any problems but when I start the second node it causes  the first node (almost always) to stop with this error : `[zed_wrapper-2] [ERROR] [1709824226.481411052] [zed.zed_node]: Critical camera error: CAMERA NOT INITIALIZED. Node stopped.`
On very rare occasions I can start both cameras but it is not reproducible :disappointed: 

I tried this on a Jetson AGX Orin with ROS2 Foxy as well as on the same Jetson with a Docker Container with Humble.

I'm also aware of [this tutorial](https://github.com/stereolabs/zed-ros2-examples/tree/master/tutorials/zed_multi_camera) but on Humble it didn't build and on Foxy it failed with `RuntimeError: Included launch description missing required argument 'cam_poses' (description: 'An array containing the array of the pose of the cameras with respect to the base frame link, e.g. [[0.5,0.0,0.0,0.0,0.0,0.0],[0.0,0.2,0.0,0.0,1.571,0.0]],[0.0,-0.2,0.0,0.0,-1.571,0.0],[-0.5,0.0,0.0,0.0,0.0,3.142]]]'), given: [cam_names, cam_models, cam_serials]` and including this argument was not possible because it always resulted in `[INFO] [launch.user]: The size of the `poses` param array must be equal to the size of `names` even after I adjusted the example to the righ amount of brackets.

### Steps to Reproduce

1. Get a Jetson AGX Orin 
2. Install Foxy or Humble
3. Install zed-ros2-wrapper
4. run `ros2 launch zed_wrapper zed_camera.launch.py camera_model:=zedx zed_id:=0 camera_name:=zed_0` and `ros2 launch zed_wrapper zed_camera.launch.py camera_model:=zedx zed_id:=1 camera_name:=zed_1`

### Expected Result

Both cameras should start and don't stop until I stop the node.

### Actual Result

One camera starts and the second camera starts but it crashes the first one.

### ZED Camera model

ZED

### Environment

```shell
OS: UBUNTU 20.04.6 LTS
CPU: ARM
GPU: 2048-core NVIDIA Ampere architecture GPU with 64 Tensor Cores
ZED SDK Version: 4.0.8
ROS: ROS2 Foxy / Humble
```

### Anything else?

I cannot specify ZEDx camera in the drop down.

adriandavidauer, 2024-03-07:
@Myzhar thanks for the quick response!
I tried that already with the same result. 😔

adriandavidauer, 2024-03-08:
@Myzhar Do you have any other suggestions? Additional sidenote: A colleague of mine used two cameras with the Python API on the same system without problems. Seems to be a ROS-specific issue. Would really appreciate the help!
Thanks

adriandavidauer, 2024-03-08:
@Myzhar First node shows
```
.
.
.
[zed_wrapper-2] [INFO] [1709896210.817555971] [zed_0.zed_node]: *** CAMERA OPENING ***
[zed_wrapper-2] [ZED-X][Warning] Failed to verify the match between the L4T version of the host and the container. If you are indeed in a container, be sure to run the same L4T version on both.
[zed_wrapper-2] [INFO] [1709896218.949574412] [zed_0.zed_node]: ZED SDK running on GPU #0
[zed_wrapper-2] [INFO] [1709896218.949687409] [zed_0.zed_node]:  * Camera Model  -> ZED X
[zed_wrapper-2] [INFO] [1709896218.949704913] [zed_0.zed_node]:  * Serial Number -> 42546479
.
.
.
```

Second node shows
```
.
.
.
[zed_wrapper-2] [INFO] [1709896219.208237041] [zed_1.zed_node]: *** CAMERA OPENING ***
[zed_wrapper-2] [ZED-X][Warning] Failed to connect to zed_x_daemon.
[zed_wrapper-2] [ZED-X][Warning] Failed to verify the match between the L4T version of the host and the container. If you are indeed in a container, be sure to run the same L4T version on both.
[zed_wrapper-2] [INFO] [1709896227.016400009] [zed_1.zed_node]: ZED SDK running on GPU #0
[zed_wrapper-2] [INFO] [1709896227.016518990] [zed_1.zed_node]:  * Camera Model  -> ZED X
[zed_wrapper-2] [INFO] [1709896227.016538191] [zed_1.zed_node]:  * Serial Number -> 48345682
.
.
.
```

So this is not the problem. I have the feeling it could have something to do with the zed_x_deamon...maybe some locking mechanism?

Myzhar, 2024-03-08:
What does it happen if you add a latency between the two openings?

PS What is your problem with building the multi-camera tutorial in Humble? I've never heard of something similar

adriandavidauer, 2024-03-08:
Building the example has some problems with angles:
```
--- stderr: angles                    
failed to create symbolic link '/root/ros2_ws/build/angles/ament_cmake_python/angles/angles' because existing path cannot be removed: Is a directory
make[2]: *** [CMakeFiles/ament_cmake_python_symlink_angles.dir/build.make:70: CMakeFiles/ament_cmake_python_symlink_angles] Error 1
make[1]: *** [CMakeFiles/Makefile2:138: CMakeFiles/ament_cmake_python_symlink_angles.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
---
Failed   <<< angles [3.13s, exited with code 2]
Aborted  <<< ament_cppcheck [3.59s]  
Aborted  <<< ament_pycodestyle [3.07s]
Aborted  <<< ament_lint [3.74s]
Aborted  <<< xacro [2.16s]           
Aborted  <<< zed_tutorial_video [6.51s]
Aborted  <<< nmea_msgs [12.3s]        
Aborted  <<< geographic_msgs [12.4s]  

Summary: 1 package finished [16.2s]
  1 package failed: angles
  7 packages aborted: ament_cppcheck ament_lint ament_pycodestyle geographic_msgs nmea_msgs xacro zed_tutorial_video
  3 packages had stderr output: angles geographic_msgs nmea_msgs
  47 packages not processed
```
And starting the second node a minute later does not have any positive effect.

adriandavidauer, 2024-03-08:
If I build without `--symlink-install` I end up with this:
```
--- stderr: rviz_plugin_zed_od               
CMake Error at CMakeLists.txt:100 (find_package):
  By not providing "Findrviz_common.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "rviz_common", but CMake did not find one.

  Could not find a package configuration file provided by "rviz_common" with
  any of the following names:

    rviz_commonConfig.cmake
    rviz_common-config.cmake

  Add the installation prefix of "rviz_common" to CMAKE_PREFIX_PATH or set
  "rviz_common_DIR" to a directory containing one of the above files.  If
  "rviz_common" provides a separate development package or SDK, be sure it
  has been installed.

---
Failed   <<< rviz_plugin_zed_od [4.30s, exited with code 1]
Aborted  <<< self_test [2.46s]              
Aborted  <<< zed_topic_benchmark_component [4.93s]
Aborted  <<< robot_localization [9.05s]     
Aborted  <<< zed_tutorial_pos_tracking [37.4s]

Summary: 41 packages finished [1min 21s]
  1 package failed: rviz_plugin_zed_od
  4 packages aborted: robot_localization self_test zed_topic_benchmark_component zed_tutorial_pos_tracking
  23 packages had stderr output: ament_clang_format ament_clang_tidy ament_copyright ament_cppcheck ament_cpplint ament_flake8 ament_lint ament_lint_cmake ament_mypy ament_pclint ament_pep257 ament_pycodestyle ament_pyflakes ament_uncrustify ament_xmllint diagnostic_aggregator geographic_msgs nmea_msgs robot_localization rviz_plugin_zed_od xacro zed_interfaces zed_topic_benchmark_interfaces
  10 packages not processed
```

Myzhar, 2024-03-08:
OK, they all are dependencies to be satisfied

adriandavidauer, 2024-03-12:
I realized the container in which I was building was based on `ros-base` and not `ros-desktop` so rviz dependencies were missing. Now I can build a step further until it crashes with the following error 
```
CMake Error at cmake_install.cmake:69 (file):
  file INSTALL cannot find
  "/root/ros2_ws/src/zed-ros2-examples/examples/zed_aruco_localization/DIRECTORY":
  No such file or directory.
```
I still don't think the multi cam example in the container will get me anywhere because on the host system it was not working either... It is really frustrating because the error when the camera is shut down is not really telling me something. 
Any other advises on how to start the nodes that they won't crash?

Myzhar, 2024-03-12:
I would recommend to re-create the example in your single package instead of cloning the full zed-ros2-examples repository to not be forced to satisfy all the dependencies that it requires.

anuj-neuralzome, 2024-03-15:
Hi @Myzhar , I have exact same issue what @adriandavidauer mentioned. In my case, I even tried multi-camera tutorial,issue persists there as well , I'm getting data from only one ZedX

anuj-neuralzome, 2024-03-15:
Here is my log output from multi-camera tutorial-
root@GTW-ONX1-E16SUV6C:~/zed_ws# ros2 launch zed_multi_camera zed_multi_camera.launch.py cam_names:='[zed_front,zed_back]' cam_models:='[zedx,zedx]' cam_serials:='[40070159,42786307]'
[INFO] [launch]: All log files can be found below /root/.ros/log/2024-03-15-08-43-23-883313-GTW-ONX1-E16SUV6C-23477
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [launch.user]: * Starting a ZED ROS2 node for camera zed_front(zedx/40070159)
[WARNING] [launch_ros.actions.node]: Parameter file path is not a file: .
[INFO] [launch.user]: * Starting a ZED ROS2 node for camera zed_back(zedx/42786307)
[WARNING] [launch_ros.actions.node]: Parameter file path is not a file: .
[INFO] [robot_state_publisher-1]: process started with pid [23482]
[INFO] [robot_state_publisher-2]: process started with pid [23484]
[INFO] [zed_wrapper-3]: process started with pid [23486]
[INFO] [robot_state_publisher-4]: process started with pid [23488]
[INFO] [zed_wrapper-5]: process started with pid [23490]
[robot_state_publisher-2] [INFO] [1710488605.402347180] [zed_front.zed_state_publisher]: got segment zed_front_camera_center
[robot_state_publisher-2] [INFO] [1710488605.402523603] [zed_front.zed_state_publisher]: got segment zed_front_camera_link
[robot_state_publisher-2] [INFO] [1710488605.402544435] [zed_front.zed_state_publisher]: got segment zed_front_left_camera_frame
[robot_state_publisher-2] [INFO] [1710488605.402555316] [zed_front.zed_state_publisher]: got segment zed_front_left_camera_optical_frame
[robot_state_publisher-2] [INFO] [1710488605.402564180] [zed_front.zed_state_publisher]: got segment zed_front_right_camera_frame
[robot_state_publisher-2] [INFO] [1710488605.402572437] [zed_front.zed_state_publisher]: got segment zed_front_right_camera_optical_frame
[robot_state_publisher-4] [INFO] [1710488605.402347052] [zed_back.zed_state_publisher]: got segment zed_back_camera_center
[robot_state_publisher-4] [INFO] [1710488605.402521395] [zed_back.zed_state_publisher]: got segment zed_back_camera_link
[robot_state_publisher-4] [INFO] [1710488605.402540371] [zed_back.zed_state_publisher]: got segment zed_back_left_camera_frame
[robot_state_publisher-4] [INFO] [1710488605.402552820] [zed_back.zed_state_publisher]: got segment zed_back_left_camera_optical_frame
[robot_state_publisher-4] [INFO] [1710488605.402561620] [zed_back.zed_state_publisher]: got segment zed_back_right_camera_frame
[robot_state_publisher-4] [INFO] [1710488605.402570036] [zed_back.zed_state_publisher]: got segment zed_back_right_camera_optical_frame
[robot_state_publisher-1] [INFO] [1710488605.409451742] [zed_multi.zed_multi_state_publisher]: got segment zed_front_camera_link
[robot_state_publisher-1] [INFO] [1710488605.409643910] [zed_multi.zed_multi_state_publisher]: got segment zed_multi_link
[robot_state_publisher-1] [INFO] [1710488605.409666662] [zed_multi.zed_multi_state_publisher]: got segment zed_rear_camera_link
[zed_wrapper-5] [INFO] [1710488605.468586826] [zed_back.zed_node_1]: ********************************
[zed_wrapper-5] [INFO] [1710488605.468793362] [zed_back.zed_node_1]:       ZED Camera Component 
[zed_wrapper-5] [INFO] [1710488605.468814035] [zed_back.zed_node_1]: ********************************
[zed_wrapper-5] [INFO] [1710488605.468828563] [zed_back.zed_node_1]:  * namespace: /zed_back
[zed_wrapper-5] [INFO] [1710488605.468840308] [zed_back.zed_node_1]:  * node name: zed_node_1
[zed_wrapper-5] [INFO] [1710488605.468849140] [zed_back.zed_node_1]: ********************************
[zed_wrapper-5] [INFO] [1710488605.468876341] [zed_back.zed_node_1]: *** DEBUG parameters ***
[zed_wrapper-5] [INFO] [1710488605.468930199] [zed_back.zed_node_1]:  * Debug Common: FALSE
[zed_wrapper-5] [INFO] [1710488605.468955576] [zed_back.zed_node_1]:  * Debug Simulation: FALSE
[zed_wrapper-5] [INFO] [1710488605.468975801] [zed_back.zed_node_1]:  * Debug Video/Depth: FALSE
[zed_wrapper-5] [INFO] [1710488605.468994298] [zed_back.zed_node_1]:  * Debug Control settings: FALSE
[zed_wrapper-5] [INFO] [1710488605.469011674] [zed_back.zed_node_1]:  * Debug Point Cloud: FALSE
[zed_wrapper-5] [INFO] [1710488605.469028827] [zed_back.zed_node_1]:  * Debug GNSS: FALSE
[zed_wrapper-5] [INFO] [1710488605.469047196] [zed_back.zed_node_1]:  * Debug Positional Tracking: FALSE
[zed_wrapper-5] [INFO] [1710488605.469065532] [zed_back.zed_node_1]:  * Debug sensors: FALSE
[zed_wrapper-5] [INFO] [1710488605.469082845] [zed_back.zed_node_1]:  * Debug Mapping: FALSE
[zed_wrapper-5] [INFO] [1710488605.469099902] [zed_back.zed_node_1]:  * Debug Object Detection: FALSE
[zed_wrapper-5] [INFO] [1710488605.469116830] [zed_back.zed_node_1]:  * Debug Body Tracking: FALSE
[zed_wrapper-5] [INFO] [1710488605.469134687] [zed_back.zed_node_1]:  * Debug Advanced: FALSE
[zed_wrapper-5] [INFO] [1710488605.469181025] [zed_back.zed_node_1]: *** GENERAL parameters ***
[zed_wrapper-5] [INFO] [1710488605.469230979] [zed_back.zed_node_1]:  * SVO: ''
[zed_wrapper-5] [INFO] [1710488605.469360648] [zed_back.zed_node_1]:  * Camera model: zedx - ZED X
[zed_wrapper-5] [INFO] [1710488605.469459340] [zed_back.zed_node_1]:  * SDK Verbose: 1
[zed_wrapper-5] [INFO] [1710488605.469540847] [zed_back.zed_node_1]:  * Camera name: zed_back
[zed_wrapper-5] [INFO] [1710488605.469610449] [zed_back.zed_node_1]:  * Camera SN: 42786307
[zed_wrapper-5] [INFO] [1710488605.469677556] [zed_back.zed_node_1]:  * Camera timeout [sec]: 5
[... cut ...]
[zed_wrapper-3] [INFO] [1710488619.628541953] [zed_front.zed_node_0]:  * T: [-0.01,0.06,0.016]
[zed_wrapper-3] [INFO] [1710488619.628580034] [zed_front.zed_node_0]:  * Q: [0,0,0,1]
[zed_wrapper-3] [WARN] [1710488619.667450688] [zed_front.zed_node_0]: sl::getSensorsData error: CAMERA NOT DETECTED
[zed_wrapper-3] [INFO] [1710488619.820730271] [zed_front.zed_node_0]: Odometry aligned to last tracking pose
[zed_wrapper-3] [INFO] [1710488619.820836995] [zed_front.zed_node_0]:  * Initial odometry [odom -> zed_front_camera_link] - {0.003,0.000,0.002} {0.145,-10.183,-0.077}

adriandavidauer, 2024-03-15:
For me it worked after building the multi_cam tutorial without the other examples. Still not sure why because to my understanding it just starts two different cam nodes with the correct parameters. If I do it manually it still crashes. 
@anuj-neuralzome the `sl::getSensorsData error: CAMERA NOT DETECTED` can be ignored. 
This dockerfile works for me:
```docker
ARG IMAGE_NAME=dustynv/ros:humble-ros-base-l4t-r35.2.1

FROM ${IMAGE_NAME}

ARG ZED_SDK_MAJOR=4
ARG ZED_SDK_MINOR=0
ARG ZED_SDK_PATCH=8
ARG JETPACK_MAJOR=5
ARG JETPACK_MINOR=0
ARG L4T_MAJOR=35
ARG L4T_MINOR=2

ARG ROS2_DIST=humble       # ROS2 distribution

# ZED ROS2 Wrapper dependencies version
ARG XACRO_VERSION=2.0.8
ARG DIAGNOSTICS_VERSION=3.0.0
ARG AMENT_LINT_VERSION=0.12.4
ARG GEOGRAPHIC_INFO_VERSION=1.0.4
ARG ROBOT_LOCALIZATION_VERSION=3.4.2
ENV DEBIAN_FRONTEND noninteractive

# Disable apt-get warnings
RUN apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 42D5A192B819C5DA || true && \
  apt-get update || true && apt-get install -y --no-install-recommends apt-utils dialog && \
  rm -rf /var/lib/apt/lists/*
  
ENV TZ=Europe/Paris

RUN ln -snf /usr/share/zoneinfo/$TZ /etc/localtime && echo $TZ > /etc/timezone && \ 
  apt-get update && \
  apt-get install --yes lsb-release wget less udev sudo build-essential cmake python3 python3-dev python3-pip python3-wheel git jq libpq-dev zstd usbutils && \    
  rm -rf /var/lib/apt/lists/*

# Install the ZED SDK
RUN echo "# R${L4T_MAJOR} (release), REVISION: ${L4T_MINOR}" > /etc/nv_tegra_release && \
  apt-get update -y || true && \
  apt-get install -y --no-install-recommends zstd wget less cmake curl gnupg2 \
  build-essential python3 python3-pip python3-dev python3-setuptools libusb-1.0-0-dev -y && \
  pip install protobuf && \
  wget -q --no-check-certificate -O ZED_SDK_Linux_JP.run \
  https://download.stereolabs.com/zedsdk/${ZED_SDK_MAJOR}.${ZED_SDK_MINOR}/l4t${L4T_MAJOR}.${L4T_MINOR}/jetsons && \
  chmod +x ZED_SDK_Linux_JP.run ; ./ZED_SDK_Linux_JP.run silent skip_tools && \
  rm -rf /usr/local/zed/resources/* && \
  rm -rf ZED_SDK_Linux_JP.run && \
  rm -rf /var/lib/apt/lists/*

# Install the ZED ROS2 Wrapper
ENV ROS_DISTRO ${ROS2_DIST}

# Install the ZED ROS2 Wrapper
WORKDIR /root/ros2_ws/src
RUN git clone --recursive https://github.com/stereolabs/zed-ros2-wrapper.git

# Install missing dependencies
WORKDIR /root/ros2_ws/src
RUN wget https://github.com/ros/xacro/archive/refs/tags/${XACRO_VERSION}.tar.gz -O - | tar -xvz && mv xacro-${XACRO_VERSION} xacro && \
  wget https://github.com/ros/diagnostics/archive/refs/tags/${DIAGNOSTICS_VERSION}.tar.gz -O - | tar -xvz && mv diagnostics-${DIAGNOSTICS_VERSION} diagnostics && \
  wget https://github.com/ament/ament_lint/archive/refs/tags/${AMENT_LINT_VERSION}.tar.gz -O - | tar -xvz && mv ament_lint-${AMENT_LINT_VERSION} ament-lint && \
  wget https://github.com/cra-ros-pkg/robot_localization/archive/refs/tags/${ROBOT_LOCALIZATION_VERSION}.tar.gz -O - | tar -xvz && mv robot_localization-${ROBOT_LOCALIZATION_VERSION} robot-localization && \
  wget https://github.com/ros-geographic-info/geographic_info/archive/refs/tags/${GEOGRAPHIC_INFO_VERSION}.tar.gz -O - | tar -xvz && mv geographic_info-${GEOGRAPHIC_INFO_VERSION} geographic-info && \
  cp -r geographic-info/geographic_msgs/ . && \
  rm -rf geographic-info && \
  git clone https://github.com/ros-drivers/nmea_msgs.git --branch ros2 && \  
  git clone https://github.com/ros/angles.git --branch humble-devel 

# Check that all the dependencies are satisfied
WORKDIR /root/ros2_ws
RUN apt-get update -y || true && rosdep update && \
  rosdep install --from-paths src --ignore-src -r -y && \
  rm -rf /var/lib/apt/lists/*

# Install cython
RUN python3 -m pip install --upgrade cython

# Build the dependencies and the ZED ROS2 Wrapper
RUN /bin/bash -c "source /opt/ros/$ROS_DISTRO/install/setup.bash && \
  colcon build --parallel-workers $(nproc) \
  --event-handlers console_direct+ --base-paths src \
  --cmake-args ' -DCMAKE_BUILD_TYPE=Release' \
  ' -DCMAKE_LIBRARY_PATH=/usr/local/cuda/lib64/stubs' \
  ' -DCMAKE_CXX_FLAGS="-Wl,--allow-shlib-undefined"' \
  ' --no-warn-unused-cli' "

# Setup environment variables
COPY ros_entrypoint_jetson.sh /sbin/ros_entrypoint.sh
RUN sudo chmod 755 /sbin/ros_entrypoint.sh

# Install examples
WORKDIR /root
RUN git clone https://github.com/stereolabs/zed-ros2-examples.git
RUN cp -r zed-ros2-examples/tutorials/zed_multi_camera /root/ros2_ws/src/
WORKDIR /root/ros2_ws
RUN apt update
RUN rosdep update
RUN rosdep install --from-paths src --ignore-src -r -y
RUN . /sbin/ros_entrypoint.sh && colcon build --cmake-args=-DCMAKE_BUILD_TYPE=Release 

ENTRYPOINT ["/sbin/ros_entrypoint.sh"]
```

And I start the container with

```bash
`docker run --rm -it -d --gpus all --runtime nvidia --privileged --network host -e DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix -v /dev:/dev -v /tmp/:/tmp/ -v /var/nvidia/nvcam/settings/:/var/nvidia/nvcam/settings/ -v /etc/systemd/system/zed_x_daemon.service:/etc/systemd/system/zed_x_daemon.service -e NVIDIA_DRIVER_CAPABILITIES=all d-reg.hb.dfki.de/smartrecyclingup/zedbot:zed-ros2-wrapper_l4t35_2_humble_base ros2 launch zed_multi_camera zed_multi_camera.launch.py cam_names:='[zed_front, zed_back]' cam_models:='[zedx, zedx]' cam_serials:='[42546479,48345682]'`
```
Hope that helps

anuj-neuralzome, 2024-03-15:
hey @adriandavidauer ,thanks for that dockerfile,I'll test and see. Are you able to install & run rviz in the very same container?

adriandavidauer, 2024-03-15:
If you want to use rviz2 in the same container you need to use `dustynv/ros:humble-desktop-l4t-r35.2.1` as base image and make sure that the l4t version is the same for the image and your Jetson. Maybe you need a different version.
Note that the command I gave you will start the container detached. You could probably start rviz2 with `docker exec` or you start a new container for rviz2 or you can just run rviz2 on your host system if already installed. If you havn't set up a URDF file for the Cams make sure to use `Image` instead of `Camera` in rviz otherwise it won't show because it does not know the tf of teh cams.

anuj-neuralzome, 2024-03-22:
Hey @adriandavidauer ,thanks very much,your solution worked out for me! Getting data from 2 zedx cams simultaneously.

closed 2024-04-22 by adriandavidauer