stereolabs/zed-ros2-wrapper #212: Unable to build and download zed-ros2-interfaces

toastedbaguatte, 2024-03-28:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

"Currently trying to download this repo to run my zed camera. 

1) However, after running "git clone  --recursive https://github.com/stereolabs/zed-ros2-wrapper.git", this is the log file i receieve. 

cloning into 'zed-ros2-wrapper'...
remote: Enumerating objects: 5519, done.
remote: Counting objects: 100% (1217/1217), done.
remote: Compressing objects: 100% (335/335), done.
remote: Total 5519 (delta 958), reused 1017 (delta 877), pack-reused 4302
Receiving objects: 100% (5519/5519), 2.99 MiB | 10.48 MiB/s, done.
Resolving deltas: 100% (3546/3546), done.
Submodule 'zed-ros2-interfaces' (https://github.com/stereolabs/zed-ros2-interfaces.git) registered for path 'zed-ros2-interfaces'
Cloning into '/home/jadon/ros2grasp_ws/src/zed-ros2-wrapper/zed-ros2-interfaces'...
remote: Enumerating objects: 179, done.        
remote: Counting objects: 100% (27/27), done.        
remote: Compressing objects: 100% (22/22), done.        
remote: Total 179 (delta 11), reused 13 (delta 5), pack-reused 152        
Receiving objects: 100% (179/179), 9.82 MiB | 12.41 MiB/s, done.
Resolving deltas: 100% (86/86), done.
Submodule path 'zed-ros2-interfaces': checked out 'f7f90f11f818c3dda23219cd05d8c854a569fc85'

2) When running "rosdep install --from-paths src --ignore-src -r -y", I receive an error 
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
zed_wrapper: Cannot locate rosdep definition for [xacro]
zed_components: Cannot locate rosdep definition for [robot_localization]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully

3) When running "colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release --parallel-workers $(nproc)",
i receive an error showing this 

[image]

please advice on what to do next, thank you so much

### Steps to Reproduce

1.
2.
3.
...

### Expected Result

All libraries to be installed

### Actual Result

Starting >>> zed_interfaces
Starting >>> grasp_python_control
--- stderr: zed_interfaces                                         
CMake Error at /opt/ros/foxy/share/rosidl_adapter/cmake/rosidl_adapt_interfaces.cmake:60 (message):
  execute_process(/home/jadon/miniconda3/bin/python3 -m rosidl_adapter
  --package-name zed_interfaces --arguments-file
  /home/jadon/ros2grasp_ws/build/zed_interfaces/rosidl_adapter__arguments__zed_interfaces.json
  --output-dir
  /home/jadon/ros2grasp_ws/build/zed_interfaces/rosidl_adapter/zed_interfaces
  --output-file
  /home/jadon/ros2grasp_ws/build/zed_interfaces/rosidl_adapter/zed_interfaces.idls)
  returned error code 1:

  AttributeError processing template 'msg.idl.em'

  Traceback (most recent call last):

    File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/resource/__init__.py", line 51, in evaluate_template
      em.BUFFERED_OPT: True,
      ^^^^^^^^^^^^^^^

  AttributeError: module 'em' has no attribute 'BUFFERED_OPT'

  

  During handling of the above exception, another exception occurred:

  

  Traceback (most recent call last):

    File "<frozen runpy>", line 198, in _run_module_as_main
    File "<frozen runpy>", line 88, in _run_code
    File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/__main__.py", line 19, in <module>
      sys.exit(main())
               ^^^^^^
    File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/main.py", line 53, in main
      abs_idl_file = convert_to_idl(
                     ^^^^^^^^^^^^^^^
    File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/__init__.py", line 19, in convert_to_idl
      return convert_msg_to_idl(
             ^^^^^^^^^^^^^^^^^^^
    File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/msg/__init__.py", line 39, in convert_msg_to_idl
      expand_template('msg.idl.em', data, output_file, encoding='iso-8859-1')
    File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/resource/__init__.py", line 23, in expand_template
      content = evaluate_template(template_name, data)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/resource/__init__.py", line 69, in evaluate_template
      _interpreter.shutdown()
      ^^^^^^^^^^^^^^^^^^^^^

  AttributeError: 'NoneType' object has no attribute 'shutdown'

Call Stack (most recent call first):
  /opt/ros/foxy/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:130 (rosidl_adapt_interfaces)
  CMakeLists.txt:98 (rosidl_generate_interfaces)

---
Failed   <<< zed_interfaces [0.93s, exited with code 1]
Aborted  <<< grasp_python_control [1.20s]                     

Summary: 0 packages finished [1.57s]
  1 package failed: zed_interfaces
  1 package aborted: grasp_python_control
  1 package had stderr output: zed_interfaces
  3 packages not processed

### ZED Camera model

ZED

### Environment

```shell
GPU: Persistence-M, Quadro P4000
CUDA: 11.4
```

Myzhar, 2024-03-28:
Please check your ROS 2 installation because the `rosdep` command cannot find the required packages `xacro` and `robot_localization`.

```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
zed_wrapper: Cannot locate rosdep definition for [xacro]
zed_components: Cannot locate rosdep definition for [robot_localization]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
```

AllenIsaacJose, 2024-04-22:
> ### Preliminary Checks
> * [x]  This issue is not a duplicate. Before opening a new issue, please search existing issues.
> * [x]  This issue is not a question, feature request, or anything other than a bug report directly related to this project.
> 
> ### Description
> "Currently trying to download this repo to run my zed camera.
> 
> 1. However, after running "git clone  --recursive https://github.com/stereolabs/zed-ros2-wrapper.git", this is the log file i receieve.
> 
> cloning into 'zed-ros2-wrapper'... remote: Enumerating objects: 5519, done. remote: Counting objects: 100% (1217/1217), done. remote: Compressing objects: 100% (335/335), done. remote: Total 5519 (delta 958), reused 1017 (delta 877), pack-reused 4302 Receiving objects: 100% (5519/5519), 2.99 MiB | 10.48 MiB/s, done. Resolving deltas: 100% (3546/3546), done. Submodule 'zed-ros2-interfaces' (https://github.com/stereolabs/zed-ros2-interfaces.git) registered for path 'zed-ros2-interfaces' Cloning into '/home/jadon/ros2grasp_ws/src/zed-ros2-wrapper/zed-ros2-interfaces'... remote: Enumerating objects: 179, done. remote: Counting objects: 100% (27/27), done. remote: Compressing objects: 100% (22/22), done. remote: Total 179 (delta 11), reused 13 (delta 5), pack-reused 152 Receiving objects: 100% (179/179), 9.82 MiB | 12.41 MiB/s, done. Resolving deltas: 100% (86/86), done. Submodule path 'zed-ros2-interfaces': checked out 'f7f90f11f818c3dda23219cd05d8c854a569fc85'
> 
> 2. When running "rosdep install --from-paths src --ignore-src -r -y", I receive an error
>    ERROR: the following packages/stacks could not have their rosdep keys resolved
>    to system dependencies:
>    zed_wrapper: Cannot locate rosdep definition for [xacro]
>    zed_components: Cannot locate rosdep definition for [robot_localization]
>    Continuing to install resolvable dependencies...
>    #All required rosdeps installed successfully
> 3. When running "colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release --parallel-workers $(nproc)",
>    i receive an error showing this
> 
> ![image](https://private-user-images.githubusercontent.com/155293525/317664678-1942f871-7ff7-4d2c-9c48-b80b5d7d6e02.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM4MDQ1ODYsIm5iZiI6MTcxMzgwNDI4NiwicGF0aCI6Ii8xNTUyOTM1MjUvMzE3NjY0Njc4LTE5NDJmODcxLTdmZjctNGQyYy05YzQ4LWI4MGI1ZDdkNmUwMi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDIyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQyMlQxNjQ0NDZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jMjE3MmE4ODIyMDQ1MDBkOGM0YWZlYjQ4MDYzMjQyZTRlNDgyN2Y3N2RlMGQ1OGIzYjZmYjY1N2Q0YzQ0NjZiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.zuGQFlh1imbkbSq-Iu0gGJ3EMYWyFcJtNAouYIB9EZs)
> 
> please advice on what to do next, thank you so much
> 
> ### Steps to Reproduce
> 
> ...
> 
> ### Expected Result
> All libraries to be installed
> 
> ### Actual Result
> Starting >>> zed_interfaces Starting >>> grasp_python_control --- stderr: zed_interfaces CMake Error at /opt/ros/foxy/share/rosidl_adapter/cmake/rosidl_adapt_interfaces.cmake:60 (message): execute_process(/home/jadon/miniconda3/bin/python3 -m rosidl_adapter --package-name zed_interfaces --arguments-file /home/jadon/ros2grasp_ws/build/zed_interfaces/rosidl_adapter__arguments__zed_interfaces.json --output-dir /home/jadon/ros2grasp_ws/build/zed_interfaces/rosidl_adapter/zed_interfaces --output-file /home/jadon/ros2grasp_ws/build/zed_interfaces/rosidl_adapter/zed_interfaces.idls) returned error code 1:
> 
> AttributeError processing template 'msg.idl.em'
> 
> Traceback (most recent call last):
> 
> ```
> File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/resource/__init__.py", line 51, in evaluate_template
>   em.BUFFERED_OPT: True,
>   ^^^^^^^^^^^^^^^
> ```
> 
> AttributeError: module 'em' has no attribute 'BUFFERED_OPT'
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
> 
> ```
> File "<frozen runpy>", line 198, in _run_module_as_main
> File "<frozen runpy>", line 88, in _run_code
> File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/__main__.py", line 19, in <module>
>   sys.exit(main())
>            ^^^^^^
> File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/main.py", line 53, in main
>   abs_idl_file = convert_to_idl(
>                  ^^^^^^^^^^^^^^^
> File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/__init__.py", line 19, in convert_to_idl
>   return convert_msg_to_idl(
>          ^^^^^^^^^^^^^^^^^^^
> File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/msg/__init__.py", line 39, in convert_msg_to_idl
>   expand_template('msg.idl.em', data, output_file, encoding='iso-8859-1')
> File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/resource/__init__.py", line 23, in expand_template
>   content = evaluate_template(template_name, data)
>             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_adapter/resource/__init__.py", line 69, in evaluate_template
>   _interpreter.shutdown()
[... 6 lines cut ...]
> 
> Failed <<< zed_interfaces [0.93s, exited with code 1] Aborted <<< grasp_python_control [1.20s]
> 
[... cut ...]
> Ubuntu 20.04
> ROS2 Foxy
> ```
> 
> ### Anything else?
> _No response_

During `rosdep install`, all the dependencies are not installed. You can see it in your log. For example xacro and robot_localization in this case. Install these packages manually. I assume you have ros2 foxy.

`sudo apt-get install ros-foxy-xacro`
`sudo apt-get install ros-foxy-robot-localization`

After which perform `rosdep install` to again check for dependency issue and install them as above and then continue with the building.

closed 2024-05-28 by github-actions[bot]