stereolabs/zed-ros2-wrapper #213: Failed to find exported target names [zed_interfaces]

jprodriguezg, 2024-04-05:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hello there,

I am trying to compile the zed-ros2-wrapper inside a docker container that already has ROS2, the Zed SDK, and the additional dependencies. 

I am following the [instructions](https://github.com/stereolabs/zed-ros2-wrapper/tree/foxy-humble-v3.8?tab=readme-ov-file#installation) to install the wrapper, but I am having an issue by compiling `zed_components`. Each time I try to compile it (`colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release`), I get the expected result (BELOW).

The dependency issue regards to `zed_interfaces`.

Has someone experienced a similar issue? Any suggestion?
Thanks!

### Steps to Reproduce

`colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release`

### Expected Result

Package built successfully 

### Actual Result

```
--- stderr: zed_components                         
* Found ROS2 foxy
* ROS2 foxy is officially supported by this package.
CMake Error at /home/user/D4A/ras/ros2/sensors_ws/install/zed_interfaces/share/zed_interfaces/cmake/ament_cmake_export_targets-extras.cmake:18 (message):
  Failed to find exported target names in
  '/home/user/D4A/ras/ros2/sensors_ws/install/zed_interfaces/share/zed_interfaces/cmake/zed_interfaces__rosidl_generator_cExport.cmake'
Call Stack (most recent call first):
  /home/user/D4A/ras/ros2/sensors_ws/install/zed_interfaces/share/zed_interfaces/cmake/zed_interfacesConfig.cmake:41 (include)
  CMakeLists.txt:111 (find_package)

make: *** [Makefile:308: cmake_check_build_system] Error 1
---
Failed   <<< zed_components [0.78s, exited with code 2]

```

### ZED Camera model

ZED2i

### Environment

```shell
ZED SDK Version 3.8.2
zed-ros2-wrapper foxy-humble-v3.8 (I have also tried with foxy-humble-v3.8.2. without success).
ROS Foxy
```

### Anything else?

`zed_interfaces` compiles successfully. 
I have correctly sourced the ROS WS and checked that ROS detects the package. Besides, this is the output of ` rosdep install --from-paths src --ignore-src -r -y`

`#All required rosdeps installed successfully`

Myzhar, 2024-04-05:
@jprodriguezg you cannot use the master branch with ZED SDK v3.8.2.
You must use this release tag: https://github.com/stereolabs/zed-ros2-wrapper/releases/tag/foxy-humble-v3.8.2

jprodriguezg, 2024-04-05:
@Myzhar I am using the branch foxy-humble-v3.8 and I also tried foxy-humble-v3.8.2

Myzhar, 2024-04-05:
The tag does not include the submodules. You can check it, the folder `zed-ros2-interfaces` is surely empty.
You must fill it with this:
https://github.com/stereolabs/zed-ros2-interfaces/releases/tag/humble-v4.0.8
it should be compatible

jprodriguezg, 2024-04-05:
Thanks @Myzhar 
I just moved to branch foxy-humble-v3.8.2 of zed-ros2-wrapper, and to branch humble-v4.0.8 of submodule zed-ros2-interfaces. However, I still have the same issue.

```
--- stderr: zed_components                          
CMake Error at /home/user/D4A/ras/ros2/sensors_ws/install/zed_interfaces/share/zed_interfaces/cmake/ament_cmake_export_targets-extras.cmake:18 (message):
  Failed to find exported target names in
  '/home/user/D4A/ras/ros2/sensors_ws/install/zed_interfaces/share/zed_interfaces/cmake/zed_interfaces__rosidl_generator_cExport.cmake'
Call Stack (most recent call first):
  /home/user/D4A/ras/ros2/sensors_ws/install/zed_interfaces/share/zed_interfaces/cmake/zed_interfacesConfig.cmake:41 (include)
  CMakeLists.txt:110 (find_package)

make: *** [Makefile:308: cmake_check_build_system] Error 1
---
Failed   <<< zed_components [1.53s, exited with code 2]
```

Myzhar, 2024-04-08:
What do you get if you try to build the `zed_interfaces` package separately?
`$ colcon build --symlink-install --packages-select zed_interfaces`

jprodriguezg, 2024-04-08:
Here is the output:

```
Starting >>> zed_interfaces
Finished <<< zed_interfaces [2.37s]                     

Summary: 1 package finished [2.48s]
```

Myzhar, 2024-04-08:
The package is then correctly built. 
Can you now build all the other packages?

If that's not the case then check your environment settings.

jprodriguezg, 2024-04-08:
I get the same error (above) while trying to compile `zed_components`. 

```
--- stderr: zed_components                          
CMake Error at /home/user/D4A/ras/ros2/sensors_ws/install/zed_interfaces/share/zed_interfaces/cmake/ament_cmake_export_targets-extras.cmake:18 (message):
  Failed to find exported target names in
  '/home/user/D4A/ras/ros2/sensors_ws/install/zed_interfaces/share/zed_interfaces/cmake/zed_interfaces__rosidl_generator_cExport.cmake'
Call Stack (most recent call first):
  /home/user/D4A/ras/ros2/sensors_ws/install/zed_interfaces/share/zed_interfaces/cmake/zed_interfacesConfig.cmake:41 (include)
  CMakeLists.txt:110 (find_package)

make: *** [Makefile:308: cmake_check_build_system] Error 1
---
Failed   <<< zed_components [1.53s, exited with code 2]
```

I have sourced my ROS2 workspace and checked `$CMAKE_PREFIX_PATH`. Everything looks OK.  
Are there any other environmental settings that I should take into account?

Myzhar, 2024-04-08:
Can you check the content of the folders and subfolders in `/home/user/D4A/ras/ros2/sensors_ws/install/zed_interfaces`?
Are the searched files there?

jprodriguezg, 2024-04-08:
@Myzhar thanks for the feedback.
All of the files that were reported as having errors are located within the folder. I discovered that some of them are hyperlinks to a file in the `build` directory which also exists in its respective folder. Here is the content of  ` /home/user/D4A/ras/ros2/sensors_ws/install/zed_interfaces/share/zed_interfaces/cmake/`

```
cmake/zed_interfaces__rosidl_typesupport_cppExport-noconfig.cmake
cmake/zed_interfaces__rosidl_generator_cExport-noconfig.cmake
cmake/zed_interfacesConfig.cmake
cmake/ament_cmake_export_targets-extras.cmake
cmake/zed_interfaces__rosidl_typesupport_cExport.cmake
cmake/rosidl_cmake_export_typesupport_targets-extras.cmake
cmake/zed_interfaces__rosidl_typesupport_introspection_cppExport-noconfig.cmake
cmake/ament_cmake_export_libraries-extras.cmake
cmake/zed_interfaces__rosidl_typesupport_cExport-noconfig.cmake
cmake/ament_cmake_export_dependencies-extras.cmake
cmake/zed_interfaces__rosidl_generator_cppExport.cmake
cmake/zed_interfaces__rosidl_generator_cExport.cmake
cmake/zed_interfaces__rosidl_typesupport_introspection_cExport-noconfig.cmake
cmake/zed_interfaces__rosidl_typesupport_introspection_cExport.cmake
cmake/ament_cmake_export_include_directories-extras.cmake
cmake/zed_interfacesConfig-version.cmake
cmake/zed_interfaces__rosidl_typesupport_cppExport.cmake
cmake/rosidl_cmake_export_typesupport_libraries-extras.cmake
cmake/rosidl_cmake-extras.cmake
cmake/zed_interfaces__rosidl_typesupport_introspection_cppExport.cmake
```

closed 2024-05-15 by github-actions[bot]