stereolabs/zed-ros2-wrapper #214: ros2 launch zed_wrapper Failed

JiaqiangZhang, 2024-04-05:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I successfully installed SDK and built the zed ros2 wrapper, but when I run launch file I had following error. I do have foxy and galactic installed on my computer at the same time.

### Steps to Reproduce

1. source /opt/ros/foxy/setup.bash
2. source ./install/setup.bash
3. ros2 launch zed_wrapper zed_camera.launch.py camera_model:=zed2i
...

### Expected Result

run zed2i with launch file

### Actual Result

```
/opt/ros/foxy/bin/ros2:6: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  from pkg_resources import load_entry_point
[INFO] [launch]: All log files can be found below /home/ /.ros/log/2024-04-05-19-39-22-032167-
[INFO] [launch]: Default logging verbosity is set to INFO
[WARNING] [launch_ros.actions.node]: Parameter file path is not a file: .
[INFO] [robot_state_publisher-1]: process started with pid [127271]
[INFO] [zed_wrapper-2]: process started with pid [127273]
[robot_state_publisher-1] /opt/ros/foxy/lib/robot_state_publisher/robot_state_publisher: symbol lookup error: /opt/ros/galactic/lib/librmw_cyclonedds_cpp.so: undefined symbol: rmw_time_equal
[zed_wrapper-2] /home/FILES/Research/workspace/zed2i_ros2_ws/install/zed_wrapper/lib/zed_wrapper/zed_wrapper: symbol lookup error: /opt/ros/galactic/lib/librmw_cyclonedds_cpp.so: undefined symbol: rmw_time_equal
[ERROR] [robot_state_publisher-1]: process has died [pid 127271, exit code 127, cmd '/opt/ros/foxy/lib/robot_state_publisher/robot_state_publisher --ros-args -r __node:=zed_state_publisher -r __ns:=/zed --params-file /tmp/launch_params_fekorx92'].
[ERROR] [zed_wrapper-2]: process has died [pid 127273, exit code 127, cmd '/home/jqzhang/FILES/Research/workspace/zed2i_ros2_ws/install/zed_wrapper/lib/zed_wrapper/zed_wrapper --ros-args -r __node:=zed_node -r __ns:=/zed --params-file /home/FILES/Research/workspace/zed2i_ros2_ws/install/zed_wrapper/share/zed_wrapper/config/common.yaml --params-file /home/FILES/Research/workspace/zed2i_ros2_ws/install/zed_wrapper/share/zed_wrapper/config/zed2i.yaml --params-file /tmp/launch_params_lvymligq'].
```

### ZED Camera model

ZED2i

### Environment

```shell
OS: UBUNTU 20.04
CPU: AMD 5000
GPU: nvidia 3060
ZED SDK: 4.0.8
ROS2: FOXY
```

Myzhar, 2024-04-05:
Hi @JiaqiangZhang 
are you using Foxy or Galactic?

The error message shows that the linker is searching for libs in `/opt/ros/galactic/lib`:
`symbol lookup error: /opt/ros/galactic/lib/librmw_cyclonedds_cpp.so: undefined symbol: rmw_time_equal`

Please check your environment settings

JiaqiangZhang, 2024-04-05:
Hi, @Myzhar 
Thanks for the reply. I have both foxy and galactic installed on my computer. I did source foxy before ros2 launch.

JiaqiangZhang, 2024-04-08:
Hi, @Myzhar 
I think the reason that caused this error (`/opt/ros/galactic/lib:symbol lookup error: /opt/ros/galactic/lib/librmw_cyclonedds_cpp.so: undefined symbol: rmw_time_equal`) was that I didn't install cyclonedds on foxy. After I install cyclonedds, the error was gone (`sudo apt-get install -y ros-foxy-rmw-cyclonedds-cpp`).
But I have another error as follow:
```
[INFO] [launch]: All log files can be found below /home/.ros/log/2024-04-08-11-36-01-548913-*-15824
[INFO] [launch]: Default logging verbosity is set to INFO
[WARNING] [launch_ros.actions.node]: Parameter file path is not a file: .
[INFO] [robot_state_publisher-1]: process started with pid [15827]
[INFO] [zed_wrapper-2]: process started with pid [15829]
[robot_state_publisher-1] Link zed_camera_center had 2 children
[robot_state_publisher-1] Link zed_left_camera_frame had 1 children
[robot_state_publisher-1] Link zed_left_camera_optical_frame had 0 children
[robot_state_publisher-1] Link zed_right_camera_frame had 1 children
[robot_state_publisher-1] Link zed_right_camera_optical_frame had 0 children
[robot_state_publisher-1] [INFO] [1712565361.817728821] [zed.zed_state_publisher]: got segment zed_camera_center
[robot_state_publisher-1] [INFO] [1712565361.817794616] [zed.zed_state_publisher]: got segment zed_camera_link
[robot_state_publisher-1] [INFO] [1712565361.817809004] [zed.zed_state_publisher]: got segment zed_left_camera_frame
[robot_state_publisher-1] [INFO] [1712565361.817820528] [zed.zed_state_publisher]: got segment zed_left_camera_optical_frame
[robot_state_publisher-1] [INFO] [1712565361.817831843] [zed.zed_state_publisher]: got segment zed_right_camera_frame
[robot_state_publisher-1] [INFO] [1712565361.817842390] [zed.zed_state_publisher]: got segment zed_right_camera_optical_frame
[ERROR] [zed_wrapper-2]: process has died [pid 15829, exit code -11, cmd '/home/FILES/Research/workspace/zed2i_ros2_ws/install/zed_wrapper/lib/zed_wrapper/zed_wrapper --ros-args -r __node:=zed_node -r __ns:=/zed --params-file /home/FILES/Research/workspace/zed2i_ros2_ws/install/zed_wrapper/share/zed_wrapper/config/common.yaml --params-file /home/FILES/Research/workspace/zed2i_ros2_ws/install/zed_wrapper/share/zed_wrapper/config/zed2i.yaml --params-file /tmp/launch_params_xug9pm6b'].
```

Myzhar, 2024-04-08:
I think this crash happens because when building you are using Foxy, while Galactic binaries are retrieved when running the nodes.
I recommend you check the environment variables to find eventual conflicts.

JiaqiangZhang, 2024-04-08:
I checked the [issue](https://github.com/stereolabs/zed-ros-wrapper/issues/674), 
> The -11 code means `Invalid access to storage.`, check if the ROS log folder has not filled the TX2 system space.
Is this the problem?

Myzhar, 2024-04-08:
You can try to clean the log folder to see if it solves the problem

JiaqiangZhang, 2024-04-08:
> You can try to clean the log folder to see if it solves the problem

I tried clean the `.ros/log` folder with `rosclean purge`, I still get the error `[ERROR] [zed_wrapper-2]: process has died [pid 20339, exit code -11`.

After uninstalling galactic, the problem is gone, and I can run the zed2i with launch file.

closed 2024-04-08 by Myzhar

Harin99, 2025-05-07:
@JiaqiangZhang 
Hi, I'm also trying to use this wrapper.
However, I'm currently facing the issue described in this thread (https://github.com/stereolabs/zed-ros2-wrapper/issues/341). 
Would you mind sharing which version of the wrapper and interfaces (branch and tag) you used to make it work?
Thanks in advance!

My setting is in (https://github.com/stereolabs/zed-ros2-wrapper/issues/341)