stereolabs/zed-ros2-wrapper #215: zed_components zed_camera_component.cpp will not compile

DennisCButts, 2024-04-05:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I have ROS2 Humble on Ubuntu 22.04 and have installed the ZED_SDK 4.1. I have tried following the instructions you have listed in the README but have had no luck. 

### Steps to Reproduce

1. Setup ROS2 Humble environment
2. Setup ZED_SDK
3. Follow instructions on the README
4. There are no rosdep dependency issues

### Expected Result

zed_components should successfully build

### Actual Result

/home/roverteam/Desktop/roverSim/camera_proj/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:9561:5: error: template argument 1 is invalid
 9561 |     RCLCPP_WARN_STREAM(
      |     ^~~~~~~~~~~~~~~~~~
/home/roverteam/Desktop/roverSim/camera_proj/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:9561:5: error: ‘<declaration error>’ is not a template [-fpermissive]
 9561 |     RCLCPP_WARN_STREAM(
      |     ^~~~~~~~~~~~~~~~~~
/home/roverteam/Desktop/roverSim/camera_proj/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:9561:5: error: template argument 1 is invalid
 9561 |     RCLCPP_WARN_STREAM(
      |     ^~~~~~~~~~~~~~~~~~
/home/roverteam/Desktop/roverSim/camera_proj/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:9561:5: error: ‘<declaration error>’ is not a template [-fpermissive]
 9561 |     RCLCPP_WARN_STREAM(
      |     ^~~~~~~~~~~~~~~~~~
/home/roverteam/Desktop/roverSim/camera_proj/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:9561:5: error: template argument 1 is invalid
 9561 |     RCLCPP_WARN_STREAM(
      |     ^~~~~~~~~~~~~~~~~~
/home/roverteam/Desktop/roverSim/camera_proj/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:9564:26: error: ‘mLastClock’ was not declared in this scope
 9564 |         static_cast<int>(mLastClock.get_clock_type()));
      |                          ^~~~~~~~~~
/home/roverteam/Desktop/roverSim/camera_proj/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:9566:5: error: ‘mClockAvailable’ was not declared in this scope
 9566 |     mClockAvailable = false;
      |     ^~~~~~~~~~~~~~~
/home/roverteam/Desktop/roverSim/camera_proj/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp: At global scope:
/home/roverteam/Desktop/roverSim/camera_proj/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:9571:1: error: expected declaration before ‘}’ token
 9571 | }  // namespace stereolabs
      | ^
gmake[2]: *** [CMakeFiles/zed_camera_component.dir/build.make:104: CMakeFiles/zed_camera_component.dir/src/zed_camera/src/zed_camera_component.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:164: CMakeFiles/zed_camera_component.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< zed_components [22.6s, exited with code 2]
                                 

There is a lot more, but I thought this was probably enough to understand the problem. 

### ZED Camera model

ZED2

### Environment

```shell
OS: Ubuntu 22.04
ROS2: Humble
ZED SDK: Version 4.1
```

### Anything else?

I have tried modifying the cpp program and using earlier branches, but I cannot figure out the issue. I was able to get it to work with  Jetson Nano running on ROS2_Eloquent using an earlier branch, so I think the issue is more likely with my setup.

Myzhar, 2024-04-05:
The ZED ROS2 Wrapper repository does not yet support the new ZED SDK v4.1.
We plan to release the new version at the beginning of the next week.
Meanwhile, you can use the ZED SDK v4.0.8:
https://www.stereolabs.com/developers/release/4.0

closed 2024-04-05 by Myzhar

gauti1311, 2024-04-15:
Hello, When are you releasing the support for ZED SDK v4.1 ? 
thanks