stereolabs/zed-ros2-wrapper #221: Failed to build using JetPack 6.0.0

JulioPlaced, 2024-04-18:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hello,
`zed_components` fails to build on a Jetson AGX Orin with Jetpack 6.0.0. A bunch of errors appear, most of them related to `sl`.

### Steps to Reproduce

colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release

### Expected Result

ROS wrapper built.

### Actual Result

Compilation errors ( i can attach the full log if useful).

Some of the errors related to `sl`:
- zed_components/src/zed_camera/include/zed_camera_component.hpp:302:76: error: ‘QUALITY’ is not a member of ‘sl::POSITIONAL_TRACKING_MODE’
302 | POSITIONAL_TRACKING_MODE mPosTrkMode = sl::POSITIONAL_TRACKING_MODE::QUALITY;

- zed_components/src/zed_camera/src/zed_camera_component.cpp:1331:49: error: ‘QUALITY’ is not a member of ‘sl::POSITIONAL_TRACKING_MODE’
1331 |     mPosTrkMode = sl::POSITIONAL_TRACKING_MODE::QUALITY;

- zed_components/src/zed_camera/src/zed_camera_component.cpp:4585:8: error: ‘struct sl::BodyTrackingParameters’ has no member named ‘image_sync’
4585 |   bt_p.image_sync = true;

### ZED Camera model

ZED Mini

### Environment

```shell
OS: Linux
CPU: ARM
GPU: Nvidia Jetson AGX Orin
ZED SDK version: 4.1
ROS Humble
```

Myzhar, 2024-04-18:
ZED SDK v4.1 is not yet supported.
A new version of the wrapper will be released in the next few days

giovanidiniz, 2024-04-21:
any updates on the 4.1 compatibility release?

Myzhar, 2024-04-22:
We are currently testing it. The release is expected to happen soon.

Yyote, 2024-04-29:
@JulioPlaced, got the wrapper to work with zed2i and ZedSDK 4.0 for Jetpack 5.0 without AI modules. Stereo depth and odometry work fine.

Myzhar, 2024-04-29:
> any updates on the 4.1 compatibility release?
Released on Friday, April 26th.

closed 2024-06-05 by github-actions[bot]