stereolabs/zed-ros2-wrapper #225: Failed   <<< zed_components [30.9s, exited with code 2]

aliby94, 2024-04-28:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

after running 
` colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release --parallel-workers $(nproc)`
receiving 
```
Failed   <<< zed_components [31.9s, exited with code 2]

Summary: 1 package finished [34.6s]
  1 package failed: zed_components
  1 package had stderr output: zed_components
  2 packages not processed
```
with multiple errors
```
 error: use of deleted function ‘std::atomic<bool>::atomic(const std::atomic<bool>&)’
  420 |   std::atomic<bool> mStreamingServerRequired = false;
error: use of deleted function ‘std::atomic<bool>::atomic(const std::atomic<bool>&)’
  722 |   std::atomic<bool> mStreamingServerRunning = false;
error: ‘using PublisherOptions = struct rclcpp::PublisherOptionsWithAllocator<std::allocator<void> >’ {aka ‘struct rclcpp::PublisherOptionsWithAllocator<std::allocator<void> >’} has no member named ‘qos_overriding_options’
  130 |   mPubOpt.qos_overriding_options =
      |           ^~~~~~~~~~~~~~~~~~~~~~
/home/ali/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:131:13: error: ‘rclcpp::QosOverridingOptions’ has not been declared
  131 |     rclcpp::QosOverridingOptions::with_default_policies();
      |             ^~~~~~~~~~~~~~~~~~~~
/home/ali/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:132:11: error: ‘using SubscriptionOptions = struct rclcpp::SubscriptionOptionsWithAllocator<std::allocator<void> >’ {aka ‘struct rclcpp::SubscriptionOptionsWithAllocator<std::allocator<void> >’} has no member named ‘qos_overriding_options’
  132 |   mSubOpt.qos_overriding_options =
      |           ^~~~~~~~~~~~~~~~~~~~~~
/home/ali/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:133:13: error: ‘rclcpp::QosOverridingOptions’ has not been declared
  133 |     rclcpp::QosOverridingOptions::with_default_policies();
      |             ^~~~~~~~~~~~~~~~~~~~

```

### Steps to Reproduce

1.
2.
3.
...

### Expected Result

successfully installing the package

### Actual Result

the full code after running the build

```
colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release --parallel-workers $(nproc)
Starting >>> zed_interfaces
Finished <<< zed_interfaces [2.29s]                     
Starting >>> zed_components
[Processing: zed_components]                             
--- stderr: zed_components                               
In file included from /home/ali/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:15:
/home/ali/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/include/zed_camera_component.hpp:420:48: error: use of deleted function ‘std::atomic<bool>::atomic(const std::atomic<bool>&)’
  420 |   std::atomic<bool> mStreamingServerRequired = false;
      |                                                ^~~~~
In file included from /home/ali/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/include/zed_camera_component.hpp:18,
                 from /home/ali/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:15:
/usr/include/c++/9/atomic:73:5: note: declared here
   73 |     atomic(const atomic&) = delete;
      |     ^~~~~~
/usr/include/c++/9/atomic:77:15: note:   after user-defined conversion: ‘constexpr std::atomic<bool>::atomic(bool)’
   77 |     constexpr atomic(bool __i) noexcept : _M_base(__i) { }
      |               ^~~~~~
In file included from /home/ali/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:15:
/home/ali/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/include/zed_camera_component.hpp:722:47: error: use of deleted function ‘std::atomic<bool>::atomic(const std::atomic<bool>&)’
  722 |   std::atomic<bool> mStreamingServerRunning = false;
      |                                               ^~~~~
In file included from /home/ali/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/include/zed_camera_component.hpp:18,
                 from /home/ali/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:15:
/usr/include/c++/9/atomic:73:5: note: declared here
   73 |     atomic(const atomic&) = delete;
      |     ^~~~~~
/usr/include/c++/9/atomic:77:15: note:   after user-defined conversion: ‘constexpr std::atomic<bool>::atomic(bool)’
   77 |     constexpr atomic(bool __i) noexcept : _M_base(__i) { }
      |               ^~~~~~
/home/ali/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp: In constructor ‘stereolabs::ZedCamera::ZedCamera(const rclcpp::NodeOptions&)’:
/home/ali/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:130:11: error: ‘using PublisherOptions = struct rclcpp::PublisherOptionsWithAllocator<std::allocator<void> >’ {aka ‘struct rclcpp::PublisherOptionsWithAllocator<std::allocator<void> >’} has no member named ‘qos_overriding_options’
  130 |   mPubOpt.qos_overriding_options =
      |           ^~~~~~~~~~~~~~~~~~~~~~
/home/ali/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:131:13: error: ‘rclcpp::QosOverridingOptions’ has not been declared
  131 |     rclcpp::QosOverridingOptions::with_default_policies();
      |             ^~~~~~~~~~~~~~~~~~~~
/home/ali/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:132:11: error: ‘using SubscriptionOptions = struct rclcpp::SubscriptionOptionsWithAllocator<std::allocator<void> >’ {aka ‘struct rclcpp::SubscriptionOptionsWithAllocator<std::allocator<void> >’} has no member named ‘qos_overriding_options’
  132 |   mSubOpt.qos_overriding_options =
      |           ^~~~~~~~~~~~~~~~~~~~~~
/home/ali/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:133:13: error: ‘rclcpp::QosOverridingOptions’ has not been declared
  133 |     rclcpp::QosOverridingOptions::with_default_policies();
      |             ^~~~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/zed_camera_component.dir/build.make:102: CMakeFiles/zed_camera_component.dir/src/zed_camera/src/zed_camera_component.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:79: CMakeFiles/zed_camera_component.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< zed_components [31.9s, exited with code 2]

Summary: 1 package finished [34.6s]
  1 package failed: zed_components
  1 package had stderr output: zed_components
  2 packages not processed

```

### ZED Camera model

ZED

### Environment

```shell
ubuntu 20.04
Jetson orin dev. kit
ZED SDK 4.1
ros2 foxy
```

MrstupidJ, 2024-04-30:
Hi I got the same issue with you. But now I think they have just pushed the fix for this yesterday. Just update the local repository to the latest (see the bottom of ReadMe) should do the job.

Myzhar, 2024-04-30:
Please update to the latest version of the master branch. The problem with Foxy has been fixed in https://github.com/stereolabs/zed-ros2-wrapper/commit/7adee58b445953f9af1ddfa67de1587ace8563f0

Please follow [this guide](https://github.com/stereolabs/zed-ros2-wrapper#update-the-local-repository) to update.

closed 2024-04-30 by Myzhar