stereolabs/zed-ros2-wrapper #226: Colcon build failed - zed_components

shivaldube, 2024-04-28:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Cloned zed-ros2-wrapper in ros2 workspace and tried colcon build as per the steps in repository read me file but the output returned zed_components colcon build error

### Steps to Reproduce

1. Installed ZED SDK v4.0.8 on Ubuntu 22.04 LTS
2. Cloned zed-ros2-wrapper in source folder at ros2 workspace.
3. in ros2 root directory, colcon build
...

### Expected Result

colcon build successful. 4 packages finished

### Actual Result

shivaldubey@shivaldubey-Precision-7680:~/ros2_ws$ colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release --parallel-workers $(nproc)
Starting >>> zed_interfaces
Finished <<< zed_interfaces [3.98s]                     
Starting >>> zed_components
--- stderr: zed_components                               
In file included from /home/shivaldubey/ros2_ws/src/zed-ros2-wrapper/zed_components/src/tools/include/gnss_replay.hpp:5,
                 from /home/shivaldubey/ros2_ws/src/zed-ros2-wrapper/zed_components/src/tools/src/gnss_replay.cpp:1:
/usr/local/zed/include/sl/Fusion.hpp:68:64: note: ‘#pragma message: ~ FUSION SDK is distributed in Early Access ~’
   68 | #pragma message("~ FUSION SDK is distributed in Early Access ~")
      |                                                                ^
In file included from /home/shivaldubey/ros2_ws/src/zed-ros2-wrapper/zed_components/src/tools/include/gnss_replay.hpp:5,
                 from /home/shivaldubey/ros2_ws/src/zed-ros2-wrapper/zed_components/src/tools/include/sl_tools.hpp:26,
                 from /home/shivaldubey/ros2_ws/src/zed-ros2-wrapper/zed_components/src/tools/src/sl_tools.cpp:25:
/usr/local/zed/include/sl/Fusion.hpp:68:64: note: ‘#pragma message: ~ FUSION SDK is distributed in Early Access ~’
   68 | #pragma message("~ FUSION SDK is distributed in Early Access ~")
      |                                                                ^
In file included from /home/shivaldubey/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/include/zed_camera_component.hpp:20,
                 from /home/shivaldubey/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:15:
/usr/local/zed/include/sl/Fusion.hpp:68:64: note: ‘#pragma message: ~ FUSION SDK is distributed in Early Access ~’
   68 | #pragma message("~ FUSION SDK is distributed in Early Access ~")
      |                                                                ^
/home/shivaldubey/ros2_ws/src/zed-ros2-wrapper/zed_components/src/tools/src/gnss_replay.cpp: In member function ‘bool sl_tools::GNSSReplay::initialize()’:
/home/shivaldubey/ros2_ws/src/zed-ros2-wrapper/zed_components/src/tools/src/gnss_replay.cpp:26:37: error: ‘using element_type = class sl::Camera’ {aka ‘class sl::Camera’} has no member named ‘getSVODataKeys’
   26 |   auto svo_custom_data_keys = _zed->getSVODataKeys();
      |                                     ^~~~~~~~~~~~~~
/home/shivaldubey/ros2_ws/src/zed-ros2-wrapper/zed_components/src/tools/src/gnss_replay.cpp:40:31: error: ‘SVOData’ is not a member of ‘sl’
   40 |   std::map<sl::Timestamp, sl::SVOData> data;
      |                               ^~~~~~~
/home/shivaldubey/ros2_ws/src/zed-ros2-wrapper/zed_components/src/tools/src/gnss_replay.cpp:40:38: error: template argument 2 is invalid
   40 |   std::map<sl::Timestamp, sl::SVOData> data;
      |                                      ^
/home/shivaldubey/ros2_ws/src/zed-ros2-wrapper/zed_components/src/tools/src/gnss_replay.cpp:40:38: error: template argument 4 is invalid
/home/shivaldubey/ros2_ws/src/zed-ros2-wrapper/zed_components/src/tools/src/gnss_replay.cpp:41:23: error: ‘using element_type = class sl::Camera’ {aka ‘class sl::Camera’} has no member named ‘retrieveSVOData’
   41 |   auto status = _zed->retrieveSVOData(gnss_key, data);  // Get ALL
      |                       ^~~~~~~~~~~~~~~
/home/shivaldubey/ros2_ws/src/zed-ros2-wrapper/zed_components/src/tools/src/gnss_replay.cpp:98:20: error: ‘begin’ was not declared in this scope
   98 |   for (auto & it : data) {
      |                    ^~~~
/home/shivaldubey/ros2_ws/src/zed-ros2-wrapper/zed_components/src/tools/src/gnss_replay.cpp:98:20: note: suggested alternatives:
In file included from /home/shivaldubey/ros2_ws/src/zed-ros2-wrapper/zed_components/src/tools/include/json.hpp:181,
                 from /home/shivaldubey/ros2_ws/src/zed-ros2-wrapper/zed_components/src/tools/include/gnss_replay.hpp:7,
                 from /home/shivaldubey/ros2_ws/src/zed-ros2-wrapper/zed_components/src/tools/src/gnss_replay.cpp:1:
/usr/include/c++/11/valarray:1228:5: note:   ‘std::begin’
 1228 |     begin(const valarray<_Tp>& __va) noexcept
      |     ^~~~~
In file included from /usr/include/c++/11/filesystem:46,
                 from /home/shivaldubey/ros2_ws/src/zed-ros2-wrapper/zed_components/src/tools/include/json.hpp:4587,
                 from /home/shivaldubey/ros2_ws/src/zed-ros2-wrapper/zed_components/src/tools/include/gnss_replay.hpp:7,
[... 329 lines cut ...]
/home/shivaldubey/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:10703:9: error: ‘sl::GNSS_FUSION_STATUS’ has not been declared
10703 |     sl::GNSS_FUSION_STATUS::OK)
      |         ^~~~~~~~~~~~~~~~~~
gmake[2]: *** [CMakeFiles/zed_camera_component.dir/build.make:118: CMakeFiles/zed_camera_component.dir/src/zed_camera/src/zed_camera_component.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:164: CMakeFiles/zed_camera_component.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< zed_components [10.6s, exited with code 2]

Summary: 1 package finished [14.8s]
  1 package failed: zed_components
  1 package had stderr output: zed_components
  2 packages not processed

### ZED Camera model

ZED

### Environment

```shell
OS: Ubuntu 22.04 LTS
ZED SDK Version: 4.0.8
ROS2 Humble
```

shivaldube, 2024-04-29:
Just to make sure: I tried reinstalled ROS2, ZED SDK 4.0.8 on Ubuntu 22 and zed-ros2-wrapper and again I am getting the same error log. Any update will be appreciated.

Myzhar, 2024-04-29:
> ZED SDK 4.0.8

The `master` branch is now compatible with the latest [ZED SDK v4.1](https://www.stereolabs.com/developers/release#82af3640d775).
I recommend updating the ZED SDK.

If you still want to use the ZED SDK v4.0.8 you can retrieve the [last tagged release of the Wrapper](https://github.com/stereolabs/zed-ros2-wrapper/releases/tag/humble-v4.0.8)

shivaldube, 2024-04-29:
Thanks it worked with Humble release.

closed 2024-04-29 by shivaldube