stereolabs/zed-ros2-wrapper #228: Low pointcloud frequency with SDK 4.1

TAlscher, 2024-05-06:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hi,
first of all. Sorry if I made any mistakes in this report. I tend to not open issues due to the possibility of errors in my setup. But I wanted to give this a try since I'm not able to fix it.
I was using the Zed SDK 4.0 before and reached my roughly 10Hz. After the Update to 4.1 I was getting only ~2Hz. At first I thought that the Jetson we are using is just too slow, but even in PERFORMANCE mode with 30-50% GPU usage I'm still getting only 2Hz

depth:
            depth_mode: "PERFORMANCE" # Matches the ZED SDK setting: 'NONE', 'PERFORMANCE', 'QUALITY', 'ULTRA', 'NEURAL', 'NEURAL_PLUS' - Note: if 'NONE' all the modules that requires depth extraction are disabled by default (Pos. Tracking, Obj. Detection, Mapping, ...)
            depth_stabilization: 1 # Forces positional tracking to start if major than 0 - Range: [0,100]
            openni_depth_mode: false # 'false': 32bit float [meters], 'true': 16bit unsigned int [millimeters]
            point_cloud_freq: 10.0 # [DYNAMIC] - frequency of the pointcloud publishing (equal or less to `grab_frame_rate` value)
            depth_confidence: 50 # [DYNAMIC] removes points on edges to avoid ’linked’ objects.
            depth_texture_conf: 100 # [DYNAMIC] removes points from images’ uniform areas.
            remove_saturated_areas: true # [DYNAMIC]

### Steps to Reproduce

1.ros2 launch zed_wrapper zed_camera.launch.py \
    camera_model:=zed2i \
    publish_map_tf:=false \
    publish_tf:=false \
    sim_mode:=false \
    xacro_path:=/zed_setup/zed_descr.urdf.xacro \
    config_path:=/zed_setup/common.yaml 
2. ros2 topic hz 
3.
...

### Expected Result

expected roughly 10Hz for the point cloud

### Actual Result

got ~2Hz
average rate: 2.303
        min: 0.415s max: 0.467s std dev: 0.00710s window: 85
average rate: 2.301
        min: 0.415s max: 0.467s std dev: 0.00726s window: 88
average rate: 2.302
        min: 0.415s max: 0.467s std dev: 0.00723s window: 91

### ZED Camera model

ZED2i

### Environment

```shell
-Zed2i
- jetson orin nano 8GB
- OS: Jetpack 5.1.2-b104
- Docker
- Current master
- Zed SDK 4.1.0
```

### Anything else?

settings:
# config/common_yaml
# Common parameters to Stereolabs ZED and ZED mini cameras

---
/**:
[... 148 lines cut ...]

        advanced: # WARNING: do not modify unless you are confident of what you are doing
            # Reference documentation: https://man7.org/linux/man-pages/man7/sched.7.html
            thread_sched_policy: "SCHED_BATCH" # 'SCHED_OTHER', 'SCHED_BATCH', 'SCHED_FIFO', 'SCHED_RR' - NOTE: 'SCHED_FIFO' and 'SCHED_RR' require 'sudo'
            thread_grab_priority: 50 # ONLY with 'SCHED_FIFO' and 'SCHED_RR' - [1 (LOW) z-> 99 (HIGH)] - NOTE: 'sudo' required
            thread_sensor_priority: 70 # ONLY with 'SCHED_FIFO' and 'SCHED_RR' - [1 (LOW) z-> 99 (HIGH)] - NOTE: 'sudo' required
            thread_pointcloud_priority: 60 # ONLY with 'SCHED_FIFO' and 'SCHED_RR' - [1 (LOW) z-> 99 (HIGH)] - NOTE: 'sudo' required

        debug:
            sdk_verbose: 1 # Set the verbose level of the ZED SDK
            debug_common: false
            debug_sim: false
            debug_video_depth: false
            debug_camera_controls: false
            debug_point_cloud: false
            debug_positional_tracking: false
            debug_gnss: false
            debug_sensors: false
            debug_mapping: false
            debug_terrain_mapping: false
            debug_object_detection: false
            debug_body_tracking: false
            debug_roi: false
            debug_streaming: false
            debug_advanced: false

Myzhar, 2024-05-06:
I noticed that you have a lot of customization. 
Please post me a screenshot of the Runtime Monitor diagnostic from `rqt` and the TF tree.

Also post a log of the node while running.

TAlscher, 2024-05-06:
> Why did you disable Positional Tracking? ` pos_tracking_enabled: false # True to enable positional tracking from start`

I wanted to be 100% sure that my Pointcloud issue is not caused by any performance issue. So I disabled everything that's not needed for the point cloud.

TAlscher, 2024-05-06:
> I noticed that you have a lot of customization. Please post me a screenshot of the Runtime Monitor diagnostic from `rqt` and the TF tree.
> 
> Also post a log of the node while running.

sorry, but I don't know how to open the Runtime Monitor diagnostic from rqt in ros2. I think I need some guidance for that. 

[frames_2024-05-06_14.20.16.pdf](https://github.com/stereolabs/zed-ros2-wrapper/files/15221891/frames_2024-05-06_14.20.16.pdf)
[image]
[image]

log:
[INFO] [launch]: All log files can be found below /root/.ros/log/2024-05-06-15-07-20-847803-ubuntu-262139
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [262145]
[INFO] [zed_wrapper-2]: process started with pid [262147]
[robot_state_publisher-1] [INFO] [1715000841.407798127] [zed.zed_state_publisher]: got segment robot/base_link
[robot_state_publisher-1] [INFO] [1715000841.407994326] [zed.zed_state_publisher]: got segment zed_camera_center
[robot_state_publisher-1] [INFO] [1715000841.408019415] [zed.zed_state_publisher]: got segment zed_camera_link
[robot_state_publisher-1] [INFO] [1715000841.408031735] [zed.zed_state_publisher]: got segment zed_left_camera_frame
[robot_state_publisher-1] [INFO] [1715000841.408042680] [zed.zed_state_publisher]: got segment zed_left_camera_optical_frame
[robot_state_publisher-1] [INFO] [1715000841.408052888] [zed.zed_state_publisher]: got segment zed_right_camera_frame
[robot_state_publisher-1] [INFO] [1715000841.408063448] [zed.zed_state_publisher]: got segment zed_right_camera_optical_frame
[zed_wrapper-2] [INFO] [1715000841.531961017] [zed.zed_node]: ********************************
[zed_wrapper-2] [INFO] [1715000841.532192002] [zed.zed_node]:       ZED Camera Component 
[zed_wrapper-2] [INFO] [1715000841.532223075] [zed.zed_node]: ********************************
[zed_wrapper-2] [INFO] [1715000841.532241828] [zed.zed_node]:  * namespace: /zed
[zed_wrapper-2] [INFO] [1715000841.532260260] [zed.zed_node]:  * node name: zed_node
[zed_wrapper-2] [INFO] [1715000841.532292101] [zed.zed_node]: ********************************
[zed_wrapper-2] [INFO] [1715000841.532371976] [zed.zed_node]: *** DEBUG parameters ***
[zed_wrapper-2] [INFO] [1715000841.532637682] [zed.zed_node]:  * SDK Verbose: 1
[zed_wrapper-2] [INFO] [1715000841.532827832] [zed.zed_node]:  * Debug Common: FALSE
[zed_wrapper-2] [INFO] [1715000841.533003551] [zed.zed_node]:  * Debug Simulation: FALSE
[zed_wrapper-2] [INFO] [1715000841.533127459] [zed.zed_node]:  * Debug Video/Depth: FALSE
[zed_wrapper-2] [INFO] [1715000841.533307049] [zed.zed_node]:  * Debug Control settings: FALSE
[zed_wrapper-2] [INFO] [1715000841.533564115] [zed.zed_node]:  * Debug Point Cloud: FALSE
[zed_wrapper-2] [INFO] [1715000841.533761018] [zed.zed_node]:  * Debug GNSS: FALSE
[zed_wrapper-2] [INFO] [1715000841.533972001] [zed.zed_node]:  * Debug Positional Tracking: FALSE
[zed_wrapper-2] [INFO] [1715000841.534174728] [zed.zed_node]:  * Debug sensors: FALSE
[zed_wrapper-2] [INFO] [1715000841.534391216] [zed.zed_node]:  * Debug Mapping: FALSE
[zed_wrapper-2] [INFO] [1715000841.534620376] [zed.zed_node]:  * Debug Object Detection: FALSE
[zed_wrapper-2] [INFO] [1715000841.534971493] [zed.zed_node]:  * Debug Body Tracking: FALSE
[zed_wrapper-2] [INFO] [1715000841.535207309] [zed.zed_node]:  * Debug Streaming: FALSE
[zed_wrapper-2] [INFO] [1715000841.535453846] [zed.zed_node]:  * Debug ROI: FALSE
[zed_wrapper-2] [INFO] [1715000841.535708639] [zed.zed_node]:  * Debug Advanced: FALSE
[zed_wrapper-2] [INFO] [1715000841.535986761] [zed.zed_node]: *** GENERAL parameters ***
[zed_wrapper-2] [INFO] [1715000841.536614143] [zed.zed_node]:  * Camera model: zed2i - ZED 2i
[zed_wrapper-2] [INFO] [1715000841.536807654] [zed.zed_node]:  * Camera name: zed
[zed_wrapper-2] [INFO] [1715000841.537097169] [zed.zed_node]:  * Camera SN: 0
[zed_wrapper-2] [INFO] [1715000841.537345273] [zed.zed_node]:  * Camera timeout [sec]: 5
[zed_wrapper-2] [INFO] [1715000841.537588354] [zed.zed_node]:  * Camera reconnection temptatives: 5
[zed_wrapper-2] [INFO] [1715000841.537828555] [zed.zed_node]:  * Camera framerate: 30
[zed_wrapper-2] [INFO] [1715000841.538025394] [zed.zed_node]:  * GPU ID: -1
[zed_wrapper-2] [INFO] [1715000841.538208248] [zed.zed_node]:  * Camera resolution: HD720
[zed_wrapper-2] [INFO] [1715000841.538385887] [zed.zed_node]:  * Publishing resolution: NATIVE
[zed_wrapper-2] [INFO] [1715000841.538552933] [zed.zed_node]:  * OpenCV custom calibration: 
[zed_wrapper-2] [INFO] [1715000841.538724523] [zed.zed_node]:  * Camera self calibration: TRUE
[zed_wrapper-2] [INFO] [1715000841.538898065] [zed.zed_node]:  * Camera flip: FALSE
[zed_wrapper-2] [INFO] [1715000841.539097304] [zed.zed_node]:  * [DYN] Publish framerate [Hz]:  30
[zed_wrapper-2] [INFO] [1715000841.539128249] [zed.zed_node]: *** VIDEO parameters ***
[zed_wrapper-2] [INFO] [1715000841.539368514] [zed.zed_node]:  * [DYN] Brightness: 4
[zed_wrapper-2] [INFO] [1715000841.539546408] [zed.zed_node]:  * [DYN] Contrast: 4
[zed_wrapper-2] [INFO] [1715000841.539780720] [zed.zed_node]:  * [DYN] Hue: 0
[zed_wrapper-2] [INFO] [1715000841.539959575] [zed.zed_node]:  * [DYN] Saturation: 4
[zed_wrapper-2] [INFO] [1715000841.540200223] [zed.zed_node]:  * [DYN] Sharpness: 4
[zed_wrapper-2] [INFO] [1715000841.540363845] [zed.zed_node]:  * [DYN] Gamma: 8
[zed_wrapper-2] [INFO] [1715000841.540558604] [zed.zed_node]:  * [DYN] Auto Exposure/Gain: TRUE
[... cut ...]
[zed_wrapper-2] [INFO] [1715000845.682299831] [zed.zed_node]: Initial ZED left camera pose (ZED pos. tracking): 
[zed_wrapper-2] [INFO] [1715000845.682332216] [zed.zed_node]:  * T: [-0.00272973,0.06,1.41782]
[zed_wrapper-2] [INFO] [1715000845.682361177] [zed.zed_node]:  * Q: [0,0.216277,0,0.976332]
[zed_wrapper-2] [INFO] [1715003984.510968408] [zed.zed_node]: *** Odometry reset for LOOP CLOSURE event ***
[zed_wrapper-2] [INFO] [1715004148.412838775] [zed.zed_node]: *** Odometry reset for LOOP CLOSURE event ***

TAlscher, 2024-05-06:
I really dont know how to open the Runtime Monitor. I Installed "ros-humble-rqt-runtime-monitor", but I dont find any window in rqt that looks like yours. Sorry :( 

tf tree is attached as a full pdf above. Here is the screenshot of it.
[image]

TAlscher, 2024-05-06:
[image]

TAlscher, 2024-05-06:
Im very sorry. seems like the issue is on my side. thanks for showing me the rqt runtime monitor.
Looks like a dds issue in my setup

closed 2024-05-06 by Myzhar