stereolabs/zed-ros2-wrapper #231: Issue with zed_components build in ROS 2 workspace

manaswakchaure, 2024-05-24:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I encountered an error while attempting to build the zed_components package in my ROS 2 workspace using the colcon build command. Below is the error message I received:

_[2.282s] ERROR:colcon.colcon_core.shell:Exception in shell extension 'sh': Expected ['.', '/home/msw_legion/ros2_ws/build/zed_components/colcon_command_prefix_build.sh', '&&', 'env'] to pass: /bin/sh: 1: eval: Syntax error: Unterminated quoted string_

### Steps to Reproduce

Navigate to the ROS 2 workspace directory.
Run the following command to build the workspace:
_colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release --parallel-workers $(nproc)_

### Expected Result

I expected the zed_components package to build successfully without encountering any errors.

### Actual Result

The build process failed with the error mentioned above.

### ZED Camera model

ZED Mini

### Environment

```shell
ROS Distribution: Humble
ROS Version: 2
Python Version: 3
ZED SDK 4.1
CUDA 12.1
```

### Anything else?

Detail error- 

Finished <<< examples_rclcpp_wait_set [1.67s]
--- stderr: examples_rclpy_minimal_action_server                                                             
/usr/lib/python3/dist-packages/setuptools/command/easy_install.py:158: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< examples_rclpy_minimal_action_server [1.97s]
--- stderr: examples_rclpy_minimal_subscriber                                                             
/usr/lib/python3/dist-packages/setuptools/command/easy_install.py:158: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< examples_rclpy_minimal_subscriber [1.97s]
--- stderr: launch_testing_examples
/usr/lib/python3/dist-packages/setuptools/command/easy_install.py:158: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< launch_testing_examples [1.99s]
Finished <<< zed_interfaces [2.04s]
Starting >>> zed_components
--- stderr: examples_rclpy_executors
/usr/lib/python3/dist-packages/setuptools/command/easy_install.py:158: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< examples_rclpy_executors [2.03s]
--- stderr: examples_rclpy_guard_conditions
/usr/lib/python3/dist-packages/setuptools/command/easy_install.py:158: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< examples_rclpy_guard_conditions [2.03s]
[2.281s] ERROR:colcon.colcon_core.shell:Exception in shell extension 'sh': Expected ['.', '/home/msw_legion/ros2_ws/build/zed_components/colcon_command_prefix_build.sh', '&&', 'env'] to pass: /bin/sh: 1: eval: Syntax error: Unterminated quoted string

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/colcon_core/shell/__init__.py", line 306, in get_command_environment
    return await extension.generate_command_environment(
  File "/usr/lib/python3/dist-packages/colcon_core/shell/sh.py", line 154, in generate_command_environment
    env = await get_environment_variables(cmd, cwd=str(build_base))
  File "/usr/lib/python3/dist-packages/colcon_core/shell/__init__.py", line 342, in get_environment_variables
    output = await check_output(cmd, cwd=cwd, shell=shell)
  File "/usr/lib/python3/dist-packages/colcon_core/subprocess.py", line 129, in check_output
    assert not rc, f'Expected {args} to pass: {stderr_data}'
AssertionError: Expected ['.', '/home/msw_legion/ros2_ws/build/zed_components/colcon_command_prefix_build.sh', '&&', 'env'] to pass: /bin/sh: 1: eval: Syntax error: Unterminated quoted string

[2.282s] ERROR:colcon.colcon_cmake.task.cmake.build:Could not find a shell extension for the command environment
Failed   <<< zed_components [0.02s, exited with code 1]
Aborted  <<< examples_rclpy_minimal_service [2.02s]
Aborted  <<< examples_rclpy_minimal_action_client [2.04s]
Aborted  <<< examples_rclpy_minimal_publisher [2.03s]
Aborted  <<< examples_rclpy_minimal_client [2.04s]
Aborted  <<< examples_rclpy_pointcloud_publisher [2.12s]                       

Summary: 18 packages finished [2.35s]
  1 package failed: zed_components
  5 packages aborted: examples_rclpy_minimal_action_client examples_rclpy_minimal_client examples_rclpy_minimal_publisher examples_rclpy_minimal_service examples_rclpy_pointcloud_publisher
  10 packages had stderr output: examples_rclpy_executors examples_rclpy_guard_conditions examples_rclpy_minimal_action_client examples_rclpy_minimal_action_server examples_rclpy_minimal_client examples_rclpy_minimal_publisher examples_rclpy_minimal_service examples_rclpy_minimal_subscriber examples_rclpy_pointcloud_publisher launch_testing_examples
  2 packages not processed

closed 2024-05-28 by manaswakchaure

camron-BHF, 2024-10-16:
@manaswakchaure  What was the fix to this error? I'm trying to build on my Jetson Orin AGX (Jetpack 6) and I'm getting an error on zed_components with messages saying about CUDA but I have the correct version installed (12.2). Thanks!

Myzhar, 2024-10-17:
Have you installed the Jetpack development packages?
`sudo apt install nvidia-jetpack nvidia-jetpack-dev`