stereolabs/zed-ros2-wrapper #234: ZED X configuration, missing parameter

pawelirh, 2024-06-21:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

When launching the node with `zed_camera.launch.py` and your predefined config for the ZED X camera, initialization fails. The camera component tries to create a subscription but the topic name is neither defined with the `mapping.clicked_point_topic` param in the config file, nor the default value is provided. To be precise, refer to [this line](https://github.com/stereolabs/zed-ros2-wrapper/blob/69373c855475c7309cf4a2099946631fe79ad926/zed_components/src/zed_camera/src/zed_camera_component.cpp#L3752)

My camera model is ZEDX (can't select this model from your drop-down list)

### Steps to Reproduce

1. Launch the camera node with `zedx.yaml`
2. Verify if the subscription was created, omitted, or failed.

### Expected Result

Driver initializes with `zedx.yaml` config file. 

### Actual Result

Driver fails initialization with `zedx.yaml` config file.

### ZED Camera model

ZED

### Environment

```shell
OS: Ubuntu 20.04
Computer: ZED Box Orin NX 16GB / 1TB GMSL@ (RTK & antenna)
ROS: Humble
```

Myzhar, 2024-06-21:
The parameter is defined in `common.yaml`:
https://github.com/stereolabs/zed-ros2-wrapper/blob/master/zed_wrapper/config/common.yaml#L107

pawelirh, 2024-06-24:
Yes, that's right, and driver with `common.yaml` works fine.

The problem is, that it doesn't work while using `zedx.yaml` config file as explained in the issue description. I assume the goal behind creating this config file was to provide param values dedicated to the ZEDX camera. Because of that, it seems to me natural to launch ZEDX with `zedx.yaml`.

pawelirh, 2024-06-24:
Unfortunately, I have no longer access to the ZED X camera so, can't test the proposed solution. Anyway, thanks a lot for your quick assistance, I'll keep that in mind while playing with your camera next time! 📷

closed 2024-06-24 by pawelirh