stereolabs/zed-ros2-wrapper #240: How to use zed2i on a computer without GPU?

zylNote, 2024-07-12:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I am currently using zed2i. It has the functions I need. But I encountered a problem. Zed2i ROS2 requires a GPU, but I don't have one on my computer. What should I do?

At present, I need to read the left and right eye images of the camera, as well as the IMU count. If you treat it as an ordinary camera, it seems that it cannot read IMU data. IMU data is very important. Do I have any other options?

### Steps to Reproduce

There will be errors

### Expected Result

I want to use zed2i on a computer without GPU

### Actual Result

There will be errors

### ZED Camera model

ZED2i

### Environment

```shell
Ubuntu22.04
```

### Anything else?

No

Myzhar, 2024-07-12:
You can only use [this repository](https://github.com/stereolabs/zed-open-capture) without a GPU, but there's no ROS 2 Driver available.

closed 2024-07-12 by Myzhar

zylNote, 2024-07-13:
Thank you very much, I have decided to add a GPU to use the Zed camera!

zylNote, 2024-07-13:
> You can only use [this repository](https://github.com/stereolabs/zed-open-capture) without a GPU, but there's no ROS 2 Driver available.

Thank you very much, I have decided to add a GPU to use the Zed camera!