stereolabs/zed-ros2-wrapper #249: Unable to get a constant publishing rate on the integrated IMU.

nicolaslauzon, 2024-08-26:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I wish to have a constant publishing rate for the IMU on my zedx camera. However it is not. It seems to be better with higher `sensors_pub_rate` but it is never constant. Here is a figure of the different settings I tried compared to a reference imu in yellow.
[image]

I played with these parameters when experimenting on getting a constant rate.
```yaml
 sensors:
        sensors_image_sync: false # Synchronize Sensors messages with latest published video/depth message
        sensors_pub_rate: 200.0 # frequency of publishing of sensors data. MAX: 400. - MIN: grab rate
        qos_history: 1 # '1': KEEP_LAST - '2': KEEP_ALL
        qos_depth: 1 # Queue size if using KEEP_LAST
        qos_reliability: 2 # '1': RELIABLE - '2': BEST_EFFORT -
        qos_durability: 2 # '1': TRANSIENT_LOCAL - '2': VOLATILE
```

### Steps to Reproduce

# Steps
1. I ran the zed_node with the following launchfile and config. I verified and my parameters are getting applied properly.
2. I recorded the data in a rosbag and analysed the delta between two imu messages using the timestamp in their header using a python script.

# I used this snipped to analyse my data
```py
def plot_imu(color, filename, label):
    df = pd.read_csv(filename)

    t = df["timestamp"]

    delta = []
    for i in range(1, len(t)-1):
        delta.append((t[i] - t[i-1])/ 1.0e9)

    plt.scatter(np.arange(0, len(delta)), delta, label=label, c=color)

```
# I used this launchfile
```py
import os

from launch import LaunchDescription
from launch.actions import IncludeLaunchDescription
from launch.launch_description_sources import PythonLaunchDescriptionSource
from ament_index_python.packages import get_package_share_directory

from launch_ros.actions import ComposableNodeContainer
from launch_ros.descriptions import ComposableNode
from launch_ros.actions import Node

def generate_launch_description():

    # ZED config
    zed_config_path = os.path.join(
                get_package_share_directory('package'), 
                'config', 
                'config.yaml'
            )
    
    # ZED Wrapper node
    zed_wrapper_node = Node(
        package='zed_wrapper',
        namespace="",
        executable='zed_wrapper',
        name="zed_node",
        output='screen',
[... 280 lines cut ...]
[zed_wrapper-1] [DEBUG] [1724178672.122002556] [zed_node]: SENSOR LAST PERIOD: 0.00528216 sec @189.316 Hz
[zed_wrapper-1] [DEBUG] [1724178672.122034140] [zed_node]: IMU MEAN freq: 1.67793
[zed_wrapper-1] [DEBUG] [1724178672.124409850] [zed_node]: Sensors callback: counting subscribers
[zed_wrapper-1] [DEBUG] [1724178672.124497307] [zed_node]: No new sensors data
[zed_wrapper-1] [DEBUG] [1724178672.126965145] [zed_node]: Sensors callback: counting subscribers
[zed_wrapper-1] [DEBUG] [1724178672.127037146] [zed_node]: No new sensors data
[zed_wrapper-1] [DEBUG] [1724178672.129582969] [zed_node]: Sensors callback: counting subscribers
[zed_wrapper-1] [DEBUG] [1724178672.129719547] [zed_node]: No new sensors data
[zed_wrapper-1] [DEBUG] [1724178672.132140601] [zed_node]: Sensors callback: counting subscribers
[zed_wrapper-1] [DEBUG] [1724178672.132246394] [zed_node]: SENSOR LAST PERIOD: 0.00487018 sec @205.331 Hz

```

### ZED Camera model

ZED

### Environment

```shell
Nvidia Jetson AGX Orin
ZED SDK : Humble-v4.0.8
ROS2 Foxy
Jetpack 5.1.3
```

boxanm, 2024-09-26:
Hey, can we please have a response on this issue? Is the uncertainty on the rate data rate an internal characteristic of the camera, or is it related to this driver?

Myzhar, 2024-09-26:
This will be fixed at the SDK level with the release v4.2

boxanm, 2024-09-26:
Nice! Do you have a release date for the new SDK version?

Myzhar, 2024-09-26:
It's imminent

closed 2024-11-01 by github-actions[bot]