stereolabs/zed-ros2-wrapper #251: Launch wrapper with Sim Mode occurs error: It's not possible to enable the required Positional Tracking module

cocobird231, 2024-09-03:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Trying to launch wrapper with simulation mode, but it occurs an error: **It's not possible to enable the required Positional Tracking module**. With some trials, I found that the error occurs when the `imu_fusion` set to `true`. If I set it to `false`, the error doesn't occur.

### Steps to Reproduce

1. Start Isaac Sim simulation.
2. Launch wrapper: `ros2 launch zed_wrapper zed_camera.launch.py camera_model:=zedx sim_mode:=true`
3. Error occurs: **It's not possible to enable the required Positional Tracking module**.
4. Stop wrapper, set `imu_fusion` to `false` under `config/common.yaml` file.
5. Launch wrapper: `ros2 launch zed_wrapper zed_camera.launch.py camera_model:=zedx sim_mode:=true`
6. Wrapper works without error.

### Expected Result

Wrapper should work without error when `sim_mode` set to `true` without changing `imu_fusion` parameter.

### Actual Result

FATAL ERROR: It's not possible to enable the required Positional Tracking module.

### ZED Camera model

ZED

### Environment

```shell
OS: Ubuntu 22.04
CPU: 12th Gen Intel(R) Core(TM) i9-12900E
GPU: NVIDIA Corporation GA106 [RTX A2000 12GB] (rev a1)
GPU Driver Version: 550.90.07
CUDA Version: 12.4
ZED SDK Version: v4.1.2
ROS: ROS2 Humble

Isaac SIM Version: 2023.1.1
Isaac Sim ZED Camera Extension v1.1.0
```

### Anything else?

Log output when error occurs

```bash
[INFO] [launch]: All log files can be found below /home/mirdc/.ros/log/2024-09-03-16-39-45-902898-mirdc-3d-29342
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [29344]
[INFO] [zed_wrapper-2]: process started with pid [29346]
[robot_state_publisher-1] [INFO] [1725352786.024594177] [zed.zed_state_publisher]: got segment zed_camera_center
[robot_state_publisher-1] [INFO] [1725352786.024651105] [zed.zed_state_publisher]: got segment zed_camera_link
[robot_state_publisher-1] [INFO] [1725352786.024656558] [zed.zed_state_publisher]: got segment zed_left_camera_frame
[robot_state_publisher-1] [INFO] [1725352786.024659692] [zed.zed_state_publisher]: got segment zed_left_camera_optical_frame
[robot_state_publisher-1] [INFO] [1725352786.024662473] [zed.zed_state_publisher]: got segment zed_right_camera_frame
[robot_state_publisher-1] [INFO] [1725352786.024665188] [zed.zed_state_publisher]: got segment zed_right_camera_optical_frame
[zed_wrapper-2] [INFO] [1725352786.038222792] [zed.zed_node]: ********************************
[zed_wrapper-2] [INFO] [1725352786.038281621] [zed.zed_node]:       ZED Camera Component 
[zed_wrapper-2] [INFO] [1725352786.038287135] [zed.zed_node]: ********************************
[zed_wrapper-2] [INFO] [1725352786.038291421] [zed.zed_node]:  * namespace: /zed
[zed_wrapper-2] [INFO] [1725352786.038295236] [zed.zed_node]:  * node name: zed_node
[zed_wrapper-2] [INFO] [1725352786.038298074] [zed.zed_node]: ********************************
[zed_wrapper-2] [INFO] [1725352786.088539359] [zed.zed_node]: *** DEBUG parameters ***
[zed_wrapper-2] [INFO] [1725352786.088685320] [zed.zed_node]:  * SDK Verbose: 1
[zed_wrapper-2] [INFO] [1725352786.088724099] [zed.zed_node]:  * Debug Common: FALSE
[zed_wrapper-2] [INFO] [1725352786.088757316] [zed.zed_node]:  * Debug Simulation: FALSE
[... 257 lines cut ...]
[zed_wrapper-2] [INFO] [1725352787.639540175] [zed.zed_node]:  Static transform Camera Center to Base [zed_camera_center -> zed_camera_link]
[zed_wrapper-2] [INFO] [1725352787.639552551] [zed.zed_node]:   * Translation: {0.000,0.000,-0.016}
[zed_wrapper-2] [INFO] [1725352787.639570004] [zed.zed_node]:   * Rotation: {0.000,-0.000,0.000}
[zed_wrapper-2] [INFO] [1725352787.640656929] [zed.zed_node]: Initial ZED left camera pose (ZED pos. tracking): 
[zed_wrapper-2] [INFO] [1725352787.640683394] [zed.zed_node]:  * T: [-0.01,0.06,0.016]
[zed_wrapper-2] [INFO] [1725352787.640702335] [zed.zed_node]:  * Q: [0,0,0,1]
[zed_wrapper-2] [2024-09-03 08:39:47 UTC][ZED][WARNING] Positional Tracking GEN2 with IMU fusion enabled requires high frequency sensors data (available with Streaming version 2, with ZED SDK >=4.1). IMU Fusion will be disabled, performance will be degraded.
[zed_wrapper-2] [WARN] [1725352788.148846624] [zed.zed_node]: Pos. Tracking not started: SENSORS DATA REQUIRED FOR THIS MODULE
[zed_wrapper-2] [INFO] [1725352788.182228014] [zed.zed_node]: *** Starting Positional Tracking ***
[zed_wrapper-2] [INFO] [1725352788.182263133] [zed.zed_node]:  * Waiting for valid static transformations...
[zed_wrapper-2] [INFO] [1725352788.183335947] [zed.zed_node]: Initial ZED left camera pose (ZED pos. tracking): 
[zed_wrapper-2] [INFO] [1725352788.183354934] [zed.zed_node]:  * T: [-0.01,0.06,0.016]
[zed_wrapper-2] [INFO] [1725352788.183369711] [zed.zed_node]:  * Q: [0,0,0,1]
[zed_wrapper-2] [2024-09-03 08:39:48 UTC][ZED][WARNING] Positional Tracking GEN2 with IMU fusion enabled requires high frequency sensors data (available with Streaming version 2, with ZED SDK >=4.1). IMU Fusion will be disabled, performance will be degraded.
[zed_wrapper-2] [WARN] [1725352788.658049115] [zed.zed_node]: Pos. Tracking not started: SENSORS DATA REQUIRED FOR THIS MODULE
[zed_wrapper-2] [INFO] [1725352788.659302610] [zed.zed_node]: *** Starting Positional Tracking ***
[zed_wrapper-2] [INFO] [1725352788.659318402] [zed.zed_node]:  * Waiting for valid static transformations...
[zed_wrapper-2] [INFO] [1725352788.660378596] [zed.zed_node]: Initial ZED left camera pose (ZED pos. tracking): 
[zed_wrapper-2] [INFO] [1725352788.660382786] [zed.zed_node]:  * T: [-0.01,0.06,0.016]
[zed_wrapper-2] [INFO] [1725352788.660391890] [zed.zed_node]:  * Q: [0,0,0,1]
[zed_wrapper-2] [2024-09-03 08:39:48 UTC][ZED][WARNING] Positional Tracking GEN2 with IMU fusion enabled requires high frequency sensors data (available with Streaming version 2, with ZED SDK >=4.1). IMU Fusion will be disabled, performance will be degraded.
[zed_wrapper-2] [WARN] [1725352789.124746789] [zed.zed_node]: Pos. Tracking not started: SENSORS DATA REQUIRED FOR THIS MODULE
[zed_wrapper-2] [FATAL] [1725352789.124791410] [zed.zed_node]: It's not possible to enable the required Positional Tracking module.
[ERROR] [zed_wrapper-2]: process has died [pid 29346, exit code 1, cmd '/home/mirdc/ros2_ws/install/zed_wrapper/lib/zed_wrapper/zed_wrapper --ros-args -r __node:=zed_node -r __ns:=/zed --params-file /home/mirdc/ros2_ws/install/zed_wrapper/share/zed_wrapper/config/common.yaml --params-file /home/mirdc/ros2_ws/install/zed_wrapper/share/zed_wrapper/config/zedx.yaml --params-file /tmp/launch_params_jsi69bs5'].
```

Myzhar, 2024-09-03:
Hi @cocobird231 
What simulation engine are you using?

nirty-fai, 2024-09-03:
I had filed a similar issue earlier and possibly a duplicate - https://github.com/stereolabs/zed-ros2-wrapper/issues/247

Myzhar, 2024-09-03:
If the simulator does not provide IMU data it's not possible to use GEN_2 positional tracking.
What simulator are you using?

cocobird231, 2024-09-04:
I'm using `Isaac Sim 2023.1.1` with `ZED Camera extension v1.1.0`, and stream the ZED with `ZED camera streamer` provided by `ZED Camera extension`.

I had try default the `common.yaml` and set `pos_tracking_mode` to `GEN_1` and it works, but the topic `/zed/zed_node/imu/data` and `/zed/zed_node/imu/data_raw` not publishing. I'm not sure the IMU data support sim mode or not.

Myzhar, 2024-09-04:
That's expected. It's not a ROS 2 Wrapper issue, but a limitation of the ZED Camera Extension for ISAAC Sim that does not yet publish simulated IMU data.

Myzhar, 2024-09-04:
I recommend you open a Feature Request issue to the ISAAC Sim repository:
https://github.com/stereolabs/zed-isaac-sim/issues

closed 2024-09-04 by Myzhar

TannerGilbert, 2025-02-17:
@Myzhar I would suggest that information about needing to change the tracking method and to disable IMU fusion should be added to [Using ZED with ROS 2 and Isaac Sim](https://www.stereolabs.com/docs/isaac-sim/ros2_integration).

TannerGilbert, 2025-02-17:
Hello @Myzhar thanks for the fast reply I'm using humble-v4.1.4 as the simulation crashes for me when I use 4.2.5 as mentioned in https://github.com/stereolabs/zed-isaac-sim/issues/17 and I wasn't able to find any legacy releases for 4.2.1-4.2.4.

With 4.1.4 it's working for me even though I'm not able to use multiple cameras in simulation at the same time currently (similar to https://community.stereolabs.com/t/zed-robot-integration-and-visualising-using-the-camera-in-isaac-sim/7818).

Error when I set two different serial numbers:
```
[Error] [sl.sensor.camera.ogn.nodes.SlCameraStreamer] Failed to initialize the ZED SDK streamer with serial number 40976320.
```

but this might be better handled in a separate issue or thread.