stereolabs/zed-ros2-wrapper #252: Jetson Orin Nano Dockerfile Fails Build -- Missing point_cloud_transport

javieryu, 2024-09-04:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

When running the docker build process on my Jetson Orin Nano the docker build fails to complete the `colcon build` step in the docker file because `zed _components` is missing the `point_cloud_transport` package.

### Steps to Reproduce

1. `git clone --recursive https://github.com/stereolabs/zed-ros2-wrapper.git`
2. `cd zed-ros2-wrapper/docker`
3. `./ros_entrypoint_jetson.sh./jetson_build_dockerfile_from_sdk_and_l4T_version.sh l4t-r35.4.1 zedsdk4.1.2`

...

### Expected Result

Docker image finishes building.

### Actual Result

Error snippet:
```
...
373.9 CMake Error at CMakeLists.txt:142 (find_package):
373.9   By not providing "Findpoint_cloud_transport.cmake" in CMAKE_MODULE_PATH
373.9   this project has asked CMake to find a package configuration file provided
373.9   by "point_cloud_transport", but CMake did not find one.
373.9
373.9   Could not find a package configuration file provided by
373.9   "point_cloud_transport" with any of the following names:
373.9
373.9     point_cloud_transportConfig.cmake
373.9     point_cloud_transport-config.cmake
373.9
373.9   Add the installation prefix of "point_cloud_transport" to CMAKE_PREFIX_PATH
373.9   or set "point_cloud_transport_DIR" to a directory containing one of the
373.9   above files.  If "point_cloud_transport" provides a separate development
373.9   package or SDK, be sure it has been installed.
373.9
373.9
373.9 -- Configuring incomplete, errors occurred!
374.1 --- stderr: zed_components
374.1 CMake Warning (dev) at /usr/local/zed/zed-config.cmake:72 (find_package):
374.1   Policy CMP0146 is not set: The FindCUDA module is removed.  Run "cmake
374.1   --help-policy CMP0146" for policy details.  Use the cmake_policy command to
374.1   set the policy and suppress this warning.
374.1
374.1 Call Stack (most recent call first):
374.1   CMakeLists.txt:81 (find_package)
374.1 This warning is for project developers.  Use -Wno-dev to suppress it.
374.1
374.1 CMake Warning (dev) at CMakeLists.txt:88 (find_package):
374.1   Policy CMP0146 is not set: The FindCUDA module is removed.  Run "cmake
374.1   --help-policy CMP0146" for policy details.  Use the cmake_policy command to
374.1   set the policy and suppress this warning.
374.1
374.1 This warning is for project developers.  Use -Wno-dev to suppress it.
374.1
374.1 CMake Error at CMakeLists.txt:142 (find_package):
374.1   By not providing "Findpoint_cloud_transport.cmake" in CMAKE_MODULE_PATH
374.1   this project has asked CMake to find a package configuration file provided
374.1   by "point_cloud_transport", but CMake did not find one.
374.1
374.1   Could not find a package configuration file provided by
374.1   "point_cloud_transport" with any of the following names:
374.1
374.1     point_cloud_transportConfig.cmake
[... 92 lines cut ...]
#15 37.50 Skip end-of-life distro "indigo"
#15 37.50 Add distro "iron"
#15 37.50 Skip end-of-life distro "jade"
#15 37.50 Add distro "jazzy"
#15 37.50 Skip end-of-life distro "kinetic"
#15 37.50 Skip end-of-life distro "lunar"
#15 37.50 Skip end-of-life distro "melodic"
#15 37.50 Add distro "noetic"
#15 37.50 Add distro "rolling"
#15 37.50 updated cache in /root/.ros/rosdep/sources.cache
#15 37.50 Warning: running 'rosdep update' as root is not recommended.
#15 37.50   You should run 'sudo rosdep fix-permissions' and invoke 'rosdep update' again without sudo.
#15 43.91 Reading package lists...
#15 45.36 Building dependency tree...
#15 45.78 Reading state information...
#15 46.43 The following packages were automatically installed and are no longer required:
#15 46.43   libgtk2.0-0 libgtk2.0-common
#15 46.43 Use 'apt autoremove' to remove them.
#15 46.43 The following additional packages will be installed:
#15 46.44   clang-format-10 libclang-cpp10 libllvm10
#15 46.50 The following NEW packages will be installed:
#15 46.50   clang-format clang-format-10 libclang-cpp10 libllvm10
#15 46.81 0 upgraded, 4 newly installed, 0 to remove and 174 not upgraded.
...
```

javieryu, 2024-09-04:
Partial solution found on the ZED Community Forums!

https://community.stereolabs.com/t/zed-ros-2-wrapper-docker-build-fails/6756/5

This fix also worked for me and I can confirm that everything seems to work properly with my ZEDM afterwards. I'm not experiencing the topic issues that the OP for the forum post had, and that seems to be a separate issue.

I'll leave this open for now because it would seem that this indicates the current `master` branch is broken for the docker install.

Copy of my working `Dockerfile.l4t-humble` for posterity:
```
ARG L4T_VERSION=l4t-r35.4.1
ARG IMAGE_NAME=dustynv/ros:humble-ros-base-${L4T_VERSION}

FROM ${IMAGE_NAME}

ARG ZED_SDK_MAJOR=4
ARG ZED_SDK_MINOR=1
ARG ZED_SDK_PATCH=0
ARG JETPACK_MAJOR=5
ARG JETPACK_MINOR=0
ARG L4T_MAJOR=35
ARG L4T_MINOR=1

ARG ROS2_DIST=humble       # ROS2 distribution

# ZED ROS2 Wrapper dependencies version
ARG XACRO_VERSION=2.0.8
ARG DIAGNOSTICS_VERSION=3.0.0
ARG AMENT_LINT_VERSION=0.12.4
ARG GEOGRAPHIC_INFO_VERSION=1.0.4
ARG ROBOT_LOCALIZATION_VERSION=3.4.2

ENV DEBIAN_FRONTEND noninteractive

# ZED SDK link
ENV ZED_SDK_URL="https://stereolabs.sfo2.cdn.digitaloceanspaces.com/zedsdk/${ZED_SDK_MAJOR}.${ZED_SDK_MINOR}/ZED_SDK_Tegra_L4T${L4T_MAJOR}.${L4T_MINOR}_v${ZED_SDK_MAJOR}.${ZED_SDK_MINOR}.${ZED_SDK_PATCH}.zstd.run"

# Check that this SDK exists
RUN if [ "$(curl -I "${ZED_SDK_URL}" -o /dev/null -s -w '%{http_code}\n' | head -n 1)" = "200" ]; then \
        echo "The URL points to something."; \
    else \
        echo "The URL does not point to a .run file or the file does not exist."; \
        exit 1; \
    fi

# Disable apt-get warnings
RUN apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 42D5A192B819C5DA || true && \
  apt-get update || true && apt-get install -y --no-install-recommends apt-utils dialog && \
  rm -rf /var/lib/apt/lists/*

ENV TZ=Europe/Paris

RUN ln -snf /usr/share/zoneinfo/$TZ /etc/localtime && echo $TZ > /etc/timezone && \ 
  apt-get update && \
  apt-get install --yes lsb-release wget less udev sudo build-essential cmake python3 python3-dev python3-pip python3-wheel git jq libpq-dev zstd usbutils && \    
  rm -rf /var/lib/apt/lists/*

RUN echo "# R${L4T_MAJOR} (release), REVISION: ${L4T_MINOR}" > /etc/nv_tegra_release && \
  apt-get update -y || true && \
  apt-get install -y --no-install-recommends zstd wget less cmake curl gnupg2 \
  build-essential python3 python3-pip python3-dev python3-setuptools libusb-1.0-0-dev -y && \
  pip install protobuf && \
  wget -q --no-check-certificate -O ZED_SDK_Linux_JP.run \
  ${ZED_SDK_URL} && \
  chmod +x ZED_SDK_Linux_JP.run ; ./ZED_SDK_Linux_JP.run silent skip_tools && \
  rm -rf /usr/local/zed/resources/* && \
  rm -rf ZED_SDK_Linux_JP.run && \
  rm -rf /var/lib/apt/lists/*

# Install the ZED ROS2 Wrapper
ENV ROS_DISTRO ${ROS2_DIST}

# Copy the sources in the Docker image
WORKDIR /root/ros2_ws/src
COPY tmp_sources/ ./

# Install missing dependencies
WORKDIR /root/ros2_ws/src
RUN wget https://github.com/ros/xacro/archive/refs/tags/${XACRO_VERSION}.tar.gz -O - | tar -xvz && mv xacro-${XACRO_VERSION} xacro && \
  wget https://github.com/ros/diagnostics/archive/refs/tags/${DIAGNOSTICS_VERSION}.tar.gz -O - | tar -xvz && mv diagnostics-${DIAGNOSTICS_VERSION} diagnostics && \
  wget https://github.com/ament/ament_lint/archive/refs/tags/${AMENT_LINT_VERSION}.tar.gz -O - | tar -xvz && mv ament_lint-${AMENT_LINT_VERSION} ament-lint && \
  wget https://github.com/cra-ros-pkg/robot_localization/archive/refs/tags/${ROBOT_LOCALIZATION_VERSION}.tar.gz -O - | tar -xvz && mv robot_localization-${ROBOT_LOCALIZATION_VERSION} robot-localization && \
  wget https://github.com/ros-geographic-info/geographic_info/archive/refs/tags/${GEOGRAPHIC_INFO_VERSION}.tar.gz -O - | tar -xvz && mv geographic_info-${GEOGRAPHIC_INFO_VERSION} geographic-info && \
  cp -r geographic-info/geographic_msgs/ . && \
  rm -rf geographic-info && \
  git clone https://github.com/ros-drivers/nmea_msgs.git --branch ros2 && \  
  git clone https://github.com/ros/angles.git --branch humble-devel && \
  git clone https://github.com/ros2/rcpputils.git --branch humble && \
  git clone https://github.com/ros-perception/point_cloud_transport.git --branch humble

# Check that all the dependencies are satisfied
WORKDIR /root/ros2_ws
RUN apt-get update -y || true && rosdep update && \
  rosdep install --from-paths src --ignore-src -r -y && \
  rm -rf /var/lib/apt/lists/*

# Install cython
RUN python3 -m pip install --upgrade cython

# Build the dependencies and the ZED ROS2 Wrapper
RUN /bin/bash -c "source /opt/ros/$ROS_DISTRO/install/setup.bash && \
  colcon build --parallel-workers $(nproc) --symlink-install --packages-select rcpputils point_cloud_transport \
  --event-handlers console_direct+ --base-paths src \
  --cmake-args ' -DCMAKE_BUILD_TYPE=Release' \
  ' -DCMAKE_LIBRARY_PATH=/usr/local/cuda/lib64/stubs' \
  ' -DCMAKE_CXX_FLAGS="-Wl,--allow-shlib-undefined"' \
  ' --no-warn-unused-cli' "

RUN /bin/bash -c "source /opt/ros/$ROS_DISTRO/install/setup.bash && source install/local_setup.bash && \
  colcon build --parallel-workers $(nproc) --symlink-install \
  --event-handlers console_direct+ --base-paths src \
  --cmake-args ' -DCMAKE_BUILD_TYPE=Release' \
  ' -DCMAKE_LIBRARY_PATH=/usr/local/cuda/lib64/stubs' \
  ' -DCMAKE_CXX_FLAGS="-Wl,--allow-shlib-undefined"' \
  ' --no-warn-unused-cli' "

WORKDIR /root/ros2_ws

# Set the default DDS middleware to FastRTPS for improved big data transmission
ENV RMW_IMPLEMENTATION=rmw_fastrtps_cpp

# Setup environment variables 
COPY ros_entrypoint_jetson.sh /sbin/ros_entrypoint.sh
RUN sudo chmod 755 /sbin/ros_entrypoint.sh

ENTRYPOINT ["/sbin/ros_entrypoint.sh"]
CMD ["bash"]

```

juliangaal, 2024-09-11:
Any idea why this multi-step build is necessary, even though the dependencies are stated correctly?

https://github.com/stereolabs/zed-ros2-wrapper/blob/baad42b042370732f5e8d3639753c561db7c1e30/zed_components/package.xml#L30

Myzhar, 2024-09-11:
I'm sorry, I cannot understand your question.

juliangaal, 2024-09-11:
In the patch that @javieryu posted, the build is separated into two steps.

First, `rcpputils` and `point_cloud_transport` are built *separately*

```bash
# Build the dependencies and the ZED ROS2 Wrapper
RUN /bin/bash -c "source /opt/ros/$ROS_DISTRO/install/setup.bash && \
  colcon build --parallel-workers $(nproc) --symlink-install --packages-select rcpputils point_cloud_transport \
  --event-handlers console_direct+ --base-paths src \
  --cmake-args ' -DCMAKE_BUILD_TYPE=Release' \
  ' -DCMAKE_LIBRARY_PATH=/usr/local/cuda/lib64/stubs' \
  ' -DCMAKE_CXX_FLAGS="-Wl,--allow-shlib-undefined"' \
  ' --no-warn-unused-cli' "
```

Then, the rest is built

```bash
RUN /bin/bash -c "source /opt/ros/$ROS_DISTRO/install/setup.bash && source install/local_setup.bash && \
  colcon build --parallel-workers $(nproc) --symlink-install \
  --event-handlers console_direct+ --base-paths src \
  --cmake-args ' -DCMAKE_BUILD_TYPE=Release' \
  ' -DCMAKE_LIBRARY_PATH=/usr/local/cuda/lib64/stubs' \
  ' -DCMAKE_CXX_FLAGS="-Wl,--allow-shlib-undefined"' \
  ' --no-warn-unused-cli' "
```

To my knowledge, this two-step process shoudn't be needed, because the dependencies are correctly stated, as alluded to in 

> Any idea why this multi-step build is necessary, even though the dependencies are stated correctly?
> 
> https://github.com/stereolabs/zed-ros2-wrapper/blob/baad42b042370732f5e8d3639753c561db7c1e30/zed_components/package.xml#L30
> 
> and
> 
> https://github.com/stereolabs/zed-ros2-wrapper/blob/baad42b042370732f5e8d3639753c561db7c1e30/zed_ros2/package.xml#L15

Do you understand my confusion?

Myzhar, 2024-09-11:
@javieryu can you justify your choice?

javieryu, 2024-09-11:
> Any idea why this multi-step build is necessary, even though the dependencies are stated correctly? [...]

This is a great question, as I noted in my post it seems like some of the dependencies in the `zed_components` `package.xml` are not being installed during the `rosdep` dependency install. When I posted this I didn't have much time to look into the source of this error, but I would think that this should be investigated so that the missing dependencies don't need to be installed from source. Clearly because this error has come up for multiple people the dependencies are not completely installed.

> @javieryu can you justify your choice?

Per the post on the ZED community forums "I then also had to first build rcpputils and point_cloud_transport, source the local workspace and then build the rest of the packages. Otherwise it wouldn’t find the point_cloud_transport package". I also was not sure why the original solution used separate colcon builds, but it may be related to the order in which colcon builds packages.

The truth is that I have ICRA deadlines this week, and didn't have time to investigate a more elegant patch. Hence no PR etc.

closed 2024-10-17 by github-actions[bot]