stereolabs/zed-ros2-wrapper #254: [docker] Update stereolabs digital ocean cdn with links for ZED SDK 4.1.4/`l4t-v36.3.0`

juliangaal, 2024-09-12:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

Even though the base image `dustynv/ros:humble-ros-base-l4t-r36.3.0` is not available yet, the jetson docker image build script fails when attempting to download.

Running

```
./jetson_build_dockerfile_from_sdk_and_l4T_version.sh l4t-r36.3.0 zedsdk4.1.4 # or zedsdk4.1.3, same result
```

fails because [https://stereolabs.sfo2.cdn.digitaloceanspaces.com/zedsdk/4.1/ZED_SDK_Tegra_L4T36.3_v4.1.3.zstd.run](https://stereolabs.sfo2.cdn.digitaloceanspaces.com/zedsdk/4.1/ZED_SDK_Tegra_L4T36.3_v4.1.3.zstd.run) does not exist.

### Use-Case

Building jetson docker images for jp6 and `l4t-r36.3.0`

Myzhar, 2024-09-12:
Hi @juliangaal 
Why do you say that https://stereolabs.sfo2.cdn.digitaloceanspaces.com/zedsdk/4.1/ZED_SDK_Tegra_L4T36.3_v4.1.3.zstd.run does not exist? The link is valid.

The link https://stereolabs.sfo2.cdn.digitaloceanspaces.com/zedsdk/4.1/ZED_SDK_Tegra_L4T36.3_v4.1.4.zstd.run is valid too.

juliangaal, 2024-09-12:
huh, seems to be a network issue on my side then! Can't reach the links since lunch. Sorry about that

closed 2024-09-12 by juliangaal